openshwprojects/OpenBK7231T_App #585: We need better rounding for sprintf

openshwprojects, 2023-01-05:
Our lightweight sprintf replacement needs better rounding.

I got such report:

> The 2 decimal points seems to be working except for the energy. It rounded 0.8299999... to 0.82 but the value should be 0.83. Seems to truncate and not round.

The syntax used for rounding is:

```
	poststr(request, "<tr><td>");
			hprintf255(request, "Temperature Channel %i value %.2f C<br>", i, fValue);
			poststr(request, "</td></tr>");
```
can anyone look into that or should I investigate it myself? @iprak  @btsimonh  @valeklubomir

iprak, 2023-01-07:
Is the actual printing now happening via nanoprintf ?

openshwprojects, 2023-01-07:
Yes in nanoprintf.

So either fix nanoprintf or createt a dedicated other hacky function for proper rounding.

I'd prefer nanoprintf fix

iprak, 2023-01-07:
This seems to be the current implementation in nanoprintf (in  readme). There are multiple issues logged in that project for enhancing the floating/precision aspect.