openshwprojects/OpenBK7231T_App #587: Add defaults for Arlec GLD130HA RGBWW light

ByteMangler, 2023-01-06:
Arlec GLD130HA lightbulb 230V B22 base.
1050 Lumen
BK7231T
uses SM2135. P24 clk, P26 dat
current default map swaps green and blue channels.
SM2135_map 1,2,0,4,3
fixes it. 
Tested with OpenBK7231T_App 1.15.286

Side effect of putting the map as a startup cmd is that the remap happens after the initial color is set.  
Eg: if the default was green, then the actual color is blue until the brightness/color is changed.

openshwprojects, 2023-01-08:
Do we have a teardown for this device?
[image]
The search yields nothing.
If you have that device, can you submit a one?
https://openbekeniot.github.io/webapp/devicesList.html

ByteMangler, 2023-01-08:
[arlec_gld130ha.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/10368210/arlec_gld130ha.zip)
Could not find how to upload files to the link you provided, so here's some pictures.

Arlec GLD130HA. Got from Bunnings (home depot like hardware store) in Australia. The 3 pack part# GLD130P3HA for AUD $20.

Opening the device is much like the Feit OM60/RGB/CA/AG/3 and similar lightbulbs. 
The 'bulb' is glued in with silicone and can be removed with careful use of a thin spudger and some force.
the led board can be removed after removing the silicone. It is a press fit, so a little leverage with a small screwdriver can release it. 

The main electronics can be removed once you remove the push-in power contacts on the base. The board pulls straight out.

It is necessary to desolder the WB2L board in order to access the pads to uart program it.

openshwprojects, 2023-01-08:
The procedure is simple, register here and post in IoT devices, write information about flashing, pictures from inside and also information where to buy and also copy some screenshots (reupload to our  hosting) from the seller website:
https://www.elektroda.com/rtvforum/forum507.html

ByteMangler, 2023-01-08:
I don't wish to register to that site. I try to keep my internet footprint deliberately small.

openshwprojects, 2023-01-08:
No problem, can you provide me some more information about your device so I can submit it myself?

openshwprojects, 2023-01-08:
Maybe a link to where to where to buy?

openshwprojects, 2023-01-08:
Ok, I will post the information that you provided and update the database. Thanks!
If you have any other devices not listed, and don't' want to register on our base site, then just please send photos/information to me and I will do all that is required.

openshwprojects, 2023-01-08:
@ByteMangler 
"Side effect of putting the map as a startup cmd is that the remap happens after the initial color is set.
Eg: if the default was green, then the actual color is blue until the brightness/color is changed."
there is also "early.bat" support for LittleFS, which executes file earlier than autoexec.bat is ran

ByteMangler, 2023-01-08:
Thanks. I had purchased these lights a few months ago as they were cheap. Once I cracked them open I found they used the BK7231 rather than the esp8266/esp8586 of the earlier products. Then that lead me to your project. Maybe they were cheap as they were running them out??

My recent work with the temp/humidity power switch caused me to update the firmware on the lightbulbs and revisit the color mapping for the SM2135 chip and thus the post. I expect there are similar if not identical products with different branding in other markets so the mapping probably needs to be applied at an earlier stage than what the startup command can provide.

openshwprojects, 2023-01-08:
Thank you, added:
[image]

closed 2023-01-08 by openshwprojects