openshwprojects/OpenBK7231T_App #588: Spectrum Smart E27 1500 lm 13 W takes long to connect to WiFi - BK7231T

Kosygor, 2023-01-06:
**Describe the bug**
Hello,
Device is slowly reconnecting to WiFi after swtching it on by Zigbee switch. 
It takes about 3-5 min to reconnect.
When I pulled BK outside and connected it to serial terminal it reconnects to WiFi in acceptable 12s. 
WiFi Signal should not be an issue since router is 3 m and 1 25cm wall from both of them. 
I have two Spectrum Smart E27 1500 lm 13 W with 
Is thare any way to save log when BK is inside the bulb ?
Can reason of this can be that I have Loaded configuration for BK7231N instead of T  ? It seems like this is only responsible for Pinout mapping.
I just assume that W2BL is BK7231T because i was not able to read chip marks after removing putty.
**Firmware:**
- Version 1.15.263
- Device Spectrum Smart E27 1500 lm 13 W with 
- Chip/model: BK7231T
- Device config
- Flag 10 and 12 are setted up plus Ping Watchdog,  configured as [E27 A60-Bulb LED RGBCCT 5PWMs (Tuya)](https://www.elektroda.com/rtvforum/topic3920286.html) Only two pins (2 and 3) had to be switched to correct control

**To Reproduce**
Well apparently no other devices have problems. 

**Screenshots**
[image]
Ripped one pad so had to solder wires directly to one of bulbs chips. 
[image]

**Additional context**
Log - not sure where it starts and i do not want to cut it.:

```
Info:MAIN:[SAFE] Time 5, idle 171361/s, free 98320, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/38
Info:MAIN:no flash configuration, use default
Info:MAIN:set ip info: 192.168.4.1,255.255.255.0,192.168.4.1
Info:MAIN:ssid:KuchniaOpenBK7231T  key: mode:0
[bk]tx_txdesc_flush
me_set_ps_disable:840 0 0 0 572573 792676
------beacon_int_set:100 TU
set_active param 0
[msg]APM_STOP_CFM
update_ongoing_1_bcn_update
update_ongoing_1_bcn_update
enter low level!
mac 10:d5:61:7d:7e:3f
leave low level!

configuring interface uap (with Static IP)WARN: TCPIP mutex is NOT locked (1) caller 515AF

sending broadcast_deauth:5
Info:MAIN:[SAFE] Time 6, idle 169847/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 7, idle 172206/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 8, idle 172690/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 9, idle 172694/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
do td cur_t:292--last:idx:13,t:283 -- new:idx:14,t:295 
--0xc:08, shift_b:0, shift_g:0, X:2
Info:MAIN:[SAFE] Time 10, idle 345135/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:GEN:dhcp=0 ip=0.0.0.0 gate=0.0.0.0 mask=0.0.0.0 mac=00:00:00:00:00:00 
Info:GEN:sta: 0, softap: 1, b/g/n
Info:GEN:softap:ssid=KuchniaOpenBK7231T,channel=1,dhcp=1,cipher_type:OPEN
Info:GEN:ip=192.168.4.1,gate=192.168.4.1,mask=255.255.255.0,dns=192.168.4.1
Info:MAIN:[SAFE] Time 11, idle 166448/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 12, idle 171629/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 13, idle 172611/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 14, idle 172700/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 15, idle 169868/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 16, idle 172559/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 17, idle 172651/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 18, idle 172625/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
do td cur_t:303--last:idx:14,t:295 -- new:idx:15,t:307 
--0xc:08, shift_b:0, shift_g:0, X:1
Info:MAIN:[SAFE] Time 19, idle 171909/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
Info:MAIN:[SAFE] Time 20, idle 253325/s, free 90600, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 5/38
[... 428 lines cut ...]
Info:MAIN:Time 39, idle 246862/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 40, idle 245060/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:GEN:dhcp=0 ip=192.168.0.245 gate=192.168.0.1 mask=255.255.255.0 mac=10:d5:61:7d:7e:3e 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-72,ssid=PiachWoczy,bssid=9c:9d:7e:75:19:90 ,channel=11,cipher_type:MIXED
Info:MAIN:Time 41, idle 239605/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 42, idle 263812/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 43, idle 244714/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
do td cur_t:351--last:idx:18,t:343 -- new:idx:19,t:355 
--0xc:08, shift_b:0, shift_g:0, X:0
Info:MAIN:Time 44, idle 244645/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 45, idle 246571/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 46, idle 247175/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 47, idle 256001/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 48, idle 247242/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 49, idle 245330/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 50, idle 243906/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:GEN:dhcp=0 ip=192.168.0.245 gate=192.168.0.1 mask=255.255.255.0 mac=10:d5:61:7d:7e:3e 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-67,ssid=PiachWoczy,bssid=9c:9d:7e:75:19:90 ,channel=11,cipher_type:MIXED
Info:MAIN:Time 51, idle 241255/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 52, idle 246699/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 53, idle 245079/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
Info:MAIN:Time 54, idle 492075/s, free 87600, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/38
```

openshwprojects, 2023-01-06:
Dziękuję, thank you, for the report.

So basically this one single device has problems with WiFi and others work?

Would you be able to submit a full 2MB flash dump of that device for analysis? 

I suspect that it may be possible to fix by clearing the BK7231 RF section, but it's a risky operation, don't try doing it yourself, I will give specific instructions when ready.

Btw, @btsimonh , thoughts on that?

Kosygor, 2023-01-07:
Well I have to admit that yesterday update of BK7231 GUI Flash Tool let me read that without an isse.
Before I had to use python uartprogram and there was no go without using CEN or RST, on controary what manuals said (pretty usefull to have reminder now in Flash Tool while getting serial). 
I am not sure if FW dump containts any credentials (probably not but ... ) so I just have sent it directly to Your email. 
I am really proud to see Elektroda`s (so I assume Polish) fighting with China`s pursuance to wrap everyone in their Tuya world ;)

openshwprojects, 2023-01-08:
Obviously FW dump contains SSID and password, it has too, as it's the full memory of the device.

We're from Elektroda and you can contribute as well by creating a teardown or a review for us:
https://openbekeniot.github.io/webapp/devicesList.html
As long as you are from within Poland, you can also get a free SD card for submitting a teardown (status of shipping outside of PL is not clear yet): https://www.elektroda.pl/rtvforum/topic3550170.html

Your case seems very rare for me and I really flashed lots of devices. Currently I can only suggest you one risky operation:
[image]
**Restore RF Config**

You can also download it first, but the whole operation is up to you, I am not sure yet if it can help or worsen the issue.

Try that now, but only if you are willing to take small risk.

The another way to fix it might be to just copy whole flash dump from another BK and just change mac, again, don't try it yourself yet because there is a risk of overwriting the bootloader.

Kosygor, 2023-01-08:
Ok I got it, thanks for clarification about FW. 

I can share teardown of course, but shouldn`t I wait for it to be working like it should  ?
Yep I am from Poland I can make PL and EN teardown`s ( but I am laking writng skill in both - so do not expect much).

About my problem I think it is worth to notice that I have counted it took about 20 min for device to connect and this is exactly what `Online for` filed said in http://IP/index Webui. So device seems to start without an issue.

Sure I have tried it, whole HA journey is trial and error. After resetting Restore RF Config It did not make much change it took 5 min to reconnect. It seems like these re-connection times are random 5-20 min. Only diffrence that I have noticed is that after this operation device sends less data trough mqtt. I have sent You before and after RF blocks.

One Idea that I have is because of that particular bulb RF antenna is on the level of aluminium plate with LED`s so I think that maybe rising whole WB2l can help. Is it worth giving a try ? 
I may fail because I have damaged few pads on first and second bulb but I will try to make one for teardown from both of them.