openshwprojects/OpenBK7231T_App #593: Relay does not toggle when Button and / or LED are on the same channel

kinsi55, 2023-01-07:
**Describe the bug**
I'm encountering a weird issue right now where the relay in my plug will only toggle on if it is on its own seperate channel - If I try to put the Button and / or LED on the same channel, the button does work, the LED does turn on, but the Relay does not.

**Firmware:**
- Version 1.15.294
- Aubess 16A Plug with BL0937
- Chip/model: BK7231N

The Pins are configured correctly as all functions work (BL0937 is read correctly, LED toggles on when pressing the button, Relay turns on when its on its own channel that doesnt share the button / LED)

openshwprojects, 2023-01-08:
Sorry, omg, fixed. Thank you for reporting. That was BAD. I will add a self test for that.

openshwprojects, 2023-01-08:
@iprak  https://github.com/openshwprojects/OpenBK7231T_App/commit/8de7957f7d5b9414e626fc6b4f56e6a159bee1f2 please be careful more next time, I will add a self-test so it doesn't happen, but basically that condition:
[image]
broke setting further channels due to calls being in the same loop:
[image]

kinsi55, 2023-01-08:
Confirmed fixed, thanks so much!

closed 2023-01-08 by kinsi55

iprak, 2023-01-08:
Thanks. My intention was to quit if DHT channel was found since that would be shared with anything else but I messed up the condition. Can you change it to quit in that case?