openshwprojects/OpenBK7231T_App #595: MQTT Discovery - Discovery Home Assistant Channels TuyaMCU

ipn00tic, 2023-01-08:
I have several double MOES switches, they appear in the identities but they don't appear as devices making integrations difficult.
Mosquitto MQTT only adds in device format through automatic discovery but in my case I cannot activate automatic discovery in openbeken firmware because none has been configured: "No relay, PWM, binary sensor or power driver running.MQTT discovery queued."

Switch configuration is done only by TuyaMCU driver and not by PINs configuration:
"
startDriver TuyaMCU
setChannelType 1 toggle
linkTuyaMCUOutputToChannel 1 1 1

setChannelType 2 toggle
linkTuyaMCUOutputToChannel 2 1 2
"

In this way, openbeken does not automatically disover to the home assistant.

I think it's just necessary to adjust the firmware?

openshwprojects, 2023-01-08:
Yes, it's on TODO list. It shouldn't be too hard, considering that our Simulator on Windows supports MQTT, so it's possible to develop stuff VERY QUICKLY without having OBK device....
@iprak , would you like to look into that? I can help you with getting simulator running

ipn00tic, 2023-01-08:
Basically what is needed is to launch the channels configured in autoexec.bat in MQTT discovery:
setChannelType 1 toggle
linkTuyaMCUOutputToChannel 1 1 1
setChannelType 2 toggle
linkTuyaMCUOutputToChannel 2 1 2

But the firmware only launches in MQTT Discovery contents configured by PIN's

MatthewReed303, 2023-01-09:
@ipn00tic see this post here where we have discussed a possible solution for this, there is also a link to an HA Forum post discussing it and a manual way to add discovery into HA using the built in script function 

https://www.elektroda.com/rtvforum/topic3946427.html