openshwprojects/OpenBK7231T_App #600: Compiling Windows OpenBK simulator on Linux

Xinayder, 2023-01-10:
I've been seeing progress being made on the Windows port of OpenBK which will allow speedy testing and simulating of the firmware.

I checked the code and it's currently based off some Windows libraries, which would make it a bit troublesome to work on Linux or macOS natively. I'm not an expert in C++ or socket programming but I think the Windows-only calls can be substituted when dealing with Linux or macOS, which would make the simulator run on those OSes.

For that, I'd like to know how is it possible to compile the simulator right now; most of the libraries required are already present on Linux or are easy enough to install.

It seems like the only way to compile the simulator right now is by using Visual Studio. Is there a way to compile it via Make? As I mentioned, I'm not that savvy with C++ and that includes Make, so if there are any tips or suggestions that can guide me to create a build target for Windows, it would be appreciated; so I can mess around with it and try to get a version of the simulator running on Linux.

openshwprojects, 2023-01-10:
Hello, thank you for your interest in Simulator (CC: @iprak  might be interested in this).
Currently, the only Windows-only part of Simulator is this menu bar:
[image]
I was unable to do it without WINAPI, at least I was unable to do it quickly.
Everything else should be portable. Sockets are portable. The time function can be also easily substituted on other platforms.

The simulator is almost fully functional, the only thing left to resolve is a MQTT connect lag that happens because of a blocking TCP connect call. 

I have compiled Simulator only in MSVC2017 so far. I do not know much about Make, it's not within my skills set yet, but it obviously should be very possible.

I will happily assist you with running/testing/building simulator and I have time for that, but you'll have to start figuring out Make stuff from scratch (same as me).

I am planning to upload simulator dependencies to separate repo soon.

Xinayder, 2023-01-11:
What I want to know firstly is, is the simulator (the screenshot you showed above) a separate part from the OpenBK port for Windows? That is, is it possible for me to just simulate the MCU running OpenBK without having to run the simulator?

If it's possible, it would be easier to start by porting the socket stack then porting the simulator. I did a quick search and turns out mimicking this menu bar outside Windows is tricky because it might depend on whether you want to use GTK or Qt or something else for your GUI. I might be wrong though, I'm not that familiar with low level programming using libs like SDL2 or OpenGL.

openshwprojects, 2023-01-11:
Windows port is written in such a way that it should be very simple to run it without OpenGL and SDL stuff. It's not in the code yet, but a simple #ifdef  trickery should do that. I can try adding such a feature for you, if you want to play with Windows. Tell me if you interested, I might do that tomorrow, and also upload all used libraries and prepared simulation files.

Xinayder, 2023-01-11:
Where would those `#ifdefs` be added? I would want to test adding them, but would you add them everywhere where SDL is utilized? So is the "network stack" (the one that runs OpenBK) already separated from the schematic designer, or are you proposing to separate it?

I'll check later with VS in a Windows machine to compile and check which files are being compiled to try to figure out how to make a Make target for the simulator.

openshwprojects, 2023-01-11:
Everything is separate.
Everything.
win_main.c
cut off calls to:
- SIM_RunWindow
- SIM_CreateWindow
- and that's all what I can think off now.
Of course, you also then have to don't include the cpp files in the build, but I don't remember any other links between them

If you do following steps, you will skip:
- opengl
- sdl
- winapi
but you will still need:
- winsock (99% compatible with linux sockets, just add #ifdefs for linux)
- timeGetTime port because its windows only
That's all. win_main.c could be renamed to sim_main.c easily.

Xinayder, 2023-01-11:
So, I tried to mess around with it today and reached some pretty good results so far, I guess.

I took an example Makefile for C/C++ projects and worked my way around that file. Most of the times I would run the target and fix the errors that popped up. I removed some resources that were included in the build because they were causing errors and my goal was to get an executable to compile, at least.

It didn't work, the Makefile manages to compile the `.c` files to `.c.d` (which I suspect are dependency files?) but when the linker tries to create the object files, it fails with `undefined reference`:

```
/usr/bin/ld: build/src/logging/logging.c.o: warning: relocation against `g_StartupDelayOver' in read-only section `.text'
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/12.2.0/../../../../lib/Scrt1.o: in function `_start':
/build/glibc/src/glibc/csu/../sysdeps/x86_64/start.S:103: undefined reference to `main'
/usr/bin/ld: build/src/bitmessage/bitmessage_read.c.o: in function `MSG_ReadString':
/home/alex/Git/OpenBK7231T_App/src/bitmessage/bitmessage_read.c:38: undefined reference to `strcpy_safe'
/usr/bin/ld: build/src/cmnds/cmd_eventHandlers.c.o: in function `EVENT_ParseEventName':
/home/alex/Git/OpenBK7231T_App/src/cmnds/cmd_eventHandlers.c:126: undefined reference to `wal_strnicmp'
/usr/bin/ld: build/src/cmnds/cmd_tokenizer.c.o: in function `Tokenizer_IsArgInteger':
/home/alex/Git/OpenBK7231T_App/src/cmnds/cmd_tokenizer.c:40: undefined reference to `strIsInteger'
/usr/bin/ld: build/src/cmnds/cmd_tokenizer.c.o: in function `Tokenizer_GetArg':
/home/alex/Git/OpenBK7231T_App/src/cmnds/cmd_tokenizer.c:55: undefined reference to `CHANNEL_Get'
/usr/bin/ld: build/src/cmnds/cmd_tokenizer.c.o: in function `Tokenizer_TokenizeString':
/home/alex/Git/OpenBK7231T_App/src/cmnds/cmd_tokenizer.c:163: undefined reference to `strcpy_safe'
/usr/bin/ld: build/src/cmnds/cmd_main.c.o: in function `CMD_ScheduleHADiscovery':
/home/alex/Git/OpenBK7231T_App/src/cmnds/cmd_main.c:58: undefined reference to `Main_ScheduleHomeAssistantDiscovery'
/usr/bin/ld: build/src/cmnds/cmd_main.c.o: in function `CMD_Flags':
/home/alex/Git/OpenBK7231T_App/src/cmnds/cmd_main.c:83: undefined reference to `CFG_SetFlags'
/usr/bin/ld: build/src/cmnds/cmd_main.c.o: in function `CMD_HTTPOTA':
/home/alex/Git/OpenBK7231T_App/src/cmnds/cmd_main.c:91: undefined reference to `OTA_RequestDownloadFromHTTP'
/usr/bin/ld: build/src/cmnds/cmd_main.c.o: in function `CMD_Restart':
/home/alex/Git/OpenBK7231T_App/src/cmnds/cmd_main.c:124: undefined reference to `RESET_ScheduleModuleReset'
```

The main problem I had to deal with were sockets. I managed to port some code but I cannot test if it works yet because, well, the Makefile still doesn't produce a successful build :stuck_out_tongue: 

I also removed any calls to the simulator window and frame as that part isn't required now. One step at a time.

I'll be publishing my results so far in my fork under a separate branch.

openshwprojects, 2023-01-12:
That's great, let me see... but before that, I will just note that I don't have much experience with Make. I had a large C++ multiplatform project with sockets and on Windows I used to compile it with MSVC 2008 and on Linux I used Code Blocks - it worked really well. I just had to maintain two separate IDE projects at once.

Ok so... you could commit your changes now so we don't stray too far from each other branches but:
[image]
Here please use a #ifdef

This looks like just a test code? Also you don't need  close socket if you had a socket function fail... there is no socket to close.
[image]