openshwprojects/OpenBK7231T_App #601: ToggleChannelOnToggle huge delay

Suxsem, 2023-01-10:
Dear all,

it seems that ToggleChannelOnToggle add an huge delay between the switch being switched and the relay being activated/deactivated, please see the below video.

[image]

Everything is fine if the PIN is configured as Button, but obviously I cannot use a two positons switch with the Button configuration.

I also tried with the "6 - [BTN] Instant touch reaction instead of waiting for release (aka SetOption 13)" without any luck.

Any chance we can tune the delay time value?
Thank you!!!

openshwprojects, 2023-01-10:
@Suxsem  I can add api to tune value and/or I can just make it  lower but I am worried about debouncing of contacts. Would you be able to test and tell me if it works ok?

Suxsem, 2023-01-10:
@openshwprojects sure! Anyway, why waiting before activate the switch? If you are worried about bouncing I would ignore any change after the initial switch.
Maybe you are worried about brief voltage drop in the mains?

Thank you!!

Suxsem, 2023-01-11:
hi @openshwprojects , any news on this one? Thanks!

openshwprojects, 2023-01-12:
Hey @Suxsem , sorry for the delay, I had to find a two state switch for testing that. I have one now, so I will reply later today, remind me if I forget

openshwprojects, 2023-01-12:
Can you check now?

Suxsem, 2023-01-12:
@openshwprojects awesome! it works flawlessly! I will monitor the behavior in the next days and open a new ticket if anything seems strange, but for now I don't see any issue.
Thank you again for your quick support, BR!