openshwprojects/OpenBK7231T_App #602: Feature Request: Counter (flow sensor)

NielsPiersma, 2023-01-11:
I want to request a Counter / Impulse sensor as an Input pin.

We have water meters (analog) in the Netherlands; they have a spinning dial.
Each spin in a specific amount of water is consumed.
[image]

Using Tasmota, we can connect the sensor (with an optocoupler) and count each turn successful.

However, as the OpenBeken devices are much smaller, and we only need one pin, I want to use an OpenBeken.

So we would like a Counter presented in MQTT as a sensor and, optionally, the ability to flash a LED (on a free pin) once the counter increases.

[image]

Niels

openshwprojects, 2023-01-12:
Hello, would you like to count on falling edge or on raising edge? Or should it be configurable?

NielsPiersma, 2023-01-12:
For our purpose it is not relevant. But future implementations we would apricate configurable.

NIels

openshwprojects, 2023-01-12:
Maybe Counter_r and Counter_f? but it sounds more confusing than Counter,,,

NielsPiersma, 2023-01-12:
I am happy with either of them. But when looking at naming the "_n" is the inverted value/state.
For example LED vs LED_n 

So for standardization purposes I mentioned Counter_n

And Tasmota also uses "Counter_n"

[image]

Niels

NielsPiersma, 2023-01-12:
And our optocouplers :)
[image]

ByteMangler, 2023-01-28:
I can't see those optocouplers doing much - you have a common power rail so there's no isolation. There's also no input protection. You'd be better off using transzorbs rather than optocouplers as they will give you a degree of protection. If you want to use optocouplers, then use a separate isolated power supply for the sensor side. If you have long wires from the sensor you might want to add some protection against lightning. A lightning strike only needs to be close to cause significant damage. if it hits you, then the you've got other issues!