openshwprojects/OpenBK7231T_App #603: BL602 build issue

forty76, 2023-01-12:
I try build OpenBK7231T_App 1.15.310 for BL602 on Ubuntu 22.04.1
My step

mkdir /home/MyUsuer/BL602
cd BL602
git clone https://github.com/openshwprojects/OpenBL602.git
go to OpenBL602/customer_app/bl602_sharedApp/bl602_sharedApp
mkdir shared
cd shared
git clone https://github.com/openshwprojects/OpenBK7231T_App.git
cd OpenBK7231T_App
make

Can anyone describe step by step to compile?
What compiler do you use for ubuntu ? Can I use some IDE ? (visual studio core like for esp32 or eclipse)

normanr, 2023-01-27:
`genromap` runs `make` with `CONFIG_CHIP_NAME` and `CONFIG_LINK_ROM` set, maybe try that?

You should also review https://pine64.github.io/bl602-docs/Quickstart_Guide/Linux/Quickstart_Linux_ubuntu.html

forty76, 2023-03-02:
Mistake is create OpenBK7231T_App folder. I solved copy all file and folder from OpenBK7231T_App folder in shared folder

closed 2023-03-02 by forty76