openshwprojects/OpenBK7231T_App #605: Incorrect dimmer level for 15%

ElclarkKuhu, 2023-01-14:
**Describe the bug**
Device sets dimmer level to 14% instead of 15% when attempting to set to 15% via Web UI or MQTT

**Firmware:**
- Version 1.15.318
- Device: Rabit 14W RGBCW Bulb (BP5758D)
- Chip/model: BK7231N
- Device config: 

    Flags: 2, 4, 10, 12, 18, 25    
    PIN 24 - BP5758D_CLK
    PIN 26 - BP5758D_DAT

**To Reproduce**
Steps to reproduce the behavior:
  1. Set Brightness to 15
  2. Check Brightness

**Additional context**
I want to gradually turn on the light over 5 minutes, but I have discovered that OpenBeken does not currently support MQTT transitions. As a workaround, I have implemented an automation loop using Home Assistant that increases the brightness by 1% every 3 seconds. However, I am encountering an issue where the brightness becomes stuck at 14%.

openshwprojects, 2023-01-29:
Hello, can you check if latest commit fixes it for you?
https://github.com/openshwprojects/OpenBK7231T_App/commit/01c5f82c80bf64fc11fb226f9856d7d7bf992d20

ElclarkKuhu, 2023-01-29:
Yes, the latest commit fixed the issue. Thank you!

closed 2023-01-29 by ElclarkKuhu