openshwprojects/OpenBK7231T_App #606: Feature Request: Native HomeAssistant API support

agittins, 2023-01-14:
So far I have been able to integrate all of my devices directly with HomeAssistant, so I don't run an MQTT server at all. While this isn't for everyone, I personally find it easier to not have to run yet another service, so native HA integration is super-handy for me and would make using these with homeassistant a much smoother process.

Would it be possible to implement the API that esphome uses, so that homeassistant could automatically discover, set up and use openBK devices without having to use mqtt in between? It has been designed to be a good fit for resource-constrained wifi devices and I think it has the features that would cover most of what's needed for these devices.

Essentially the device answers requests on a tcp port, and the HA instance can request details, subscribe for notifications and issue commands - with optional authentication and without having additional protocol layers to deal with.

The API is documented at <https://github.com/esphome/esphome/blob/dev/esphome/components/api/api.proto> with more info from the esphome side at <https://esphome.io/components/api.html> and a python client implementation at <https://github.com/esphome/aioesphomeapi>.

I'm currently testing obk on some Aldi CASALife 706391 downlights, which I have configured with
P7: PWM_n=0
P8: PWM=1
With `Flag 8 - [LED] Alternate CW light mode (first PWM for warm/cold slider, second for brightness)` set.

Build on Nov 14 2022 04:22:54 version 1.15.33
Short name: obk120B4E19, Chipset BK7231T

openshwprojects, 2023-01-14:
Hello, that's a great idea, thank you for posting. My only question would be how much of that API would be required to run the basic devices? It seems like a pretty advanced API and I would be worried that it would require us to write support for dozens of packet types in order to be able to just toggle a relay.

Is there any more docs on that topic, especially on Home Assistant side of things?

Would you be able to tell which packets from links would be required for your use case?

agittins, 2023-01-14:
Great questions! I didn't look deeply into it, but was expecting that the API probably grew gradually as esphome evolved so I expected (naively, perhaps!) that there would be a minimal part to implement, then grow it as more devices came onboard.

For my use case which right now is just one type of downlight, with two controls: brightness and colourtemp, I'd expect it would need (each one has a well-defined request and response variant, it seems):

Hello
Connect
Disconnect
Ping
DeviceInfo (lots of esphome-specific stuff, but probably can be fudged)

That much seems to be the base requirements, then for each sensor/switch (entity) the device has, it needs to support responding to:

ListEntities
SubscribeStates

For my light, I think everything it needs would be covered by a single "light" entity, per <https://github.com/esphome/esphome/blob/dev/esphome/components/api/api.proto#L396>:

for ListEntitiesLightResponse we'd report ColorMode support for ON_OFF, BRIGHTNESS, and I guess COLD_WARM_WHITE (but not certain - looks like you can offer multiple ways to control temp).

LightStateResponse to send the current state of the light (on/off, brightness, color_mode, cold_white/warm_white)

LightCommandRequest to receive setting to apply to the light.

I think that's all we'd have to implement - all the "homeassistant" endpoints are for actually implementing HA services and stuff as customised things inside of the device, which we wouldn't need, it's just the hello/[dis]connect/ping/list/state/command stuff, and only for the entity types (in this case, just "light") we want to implement.

You can see the implementation of the state, info and command messages in the `#ifdef USE_LIGHT` block at <https://github.com/esphome/esphome/blob/dev/esphome/components/api/api_connection.cpp#L309> and it looks like the setup of the socket and listener is covered at <https://github.com/esphome/esphome/blob/dev/esphome/components/api/api_server.cpp#L24>

I don't know if that gives some helpful pointers or not - my C/C++ knowledge is pretty poor so not knowing what I don't know, it's hard for me to identify what's useful vs noise! :-)

Hmmm speaking of... it's a "protobuf" aka Protocol Buffers implementation, for which there is (presumably helpful?) info at <https://protobuf.dev/> - which includes some code-generation tools. I'm guessing that the idea is you can take the protocol definition from the esphome project, and have the tool template out some code for your desired implementation. That might offer a base model without having a lot of confusing stuff from the esphome-specific parts baked into it.

I wonder if it would be worth pinging the folks at esphome, as they might be interested in helping to integrate things? I think the core esphome folk are directly employed by nabucasa (homeassistant) so I expect they'd be open to collaboration and integration.

openshwprojects, 2023-01-14:
I think I'll look at it myself first. It might not be that hard.

agittins, 2023-01-15:
Oh, it just occurred to me that we would probably also need to send an mDNS broadcast, in order for the discovery process to kick off. It looks like this is done inside the ifdef API blocks at <https://github.com/esphome/esphome/blob/dev/esphome/components/mdns/mdns_component.cpp>

If it helps, when I run on my server `avahi-browse -t -r -a -p 2>/dev/null | grep esp` to find what esp boards are sending out via mdns, I get entries like:

```
+;br-lan;IPv6;camfront;_esphomelib._tcp;local
+;br-lan;IPv4;pluge2a9;_esphomelib._tcp;local
+;br-lan;IPv4;hubcentral;_esphomelib._tcp;local
```
and
```
=;br-lan;IPv4;lounge_remote;_esphomelib._tcp;local;lounge_remote.local;192.168.xx.xx;6053;"version=1.16.2" "address=lounge_remote.local" "mac=xxxxxxxxxxxx"
=;br-lan;IPv4;plug5120;_esphomelib._tcp;local;plug5120.local;192.168.xx.xx;6053;"version=2022.7.0-dev" "mac=xxxxxxxxxxxx" "platform=ESP8266" "board=esp01_1m"
```
(I've redacted mac and ip addresses, obviously).

My guess is that it would probably be enough to advertise just the `_esphomelib._tcp` service and that might be all it needs for HA to then make contact. If you're looking for HA's side of the implementation it appears to be at <https://github.com/home-assistant/core/tree/dev/homeassistant/components/esphome> but I don't know how much help that might be initially. I think the config_flow might have the core bits of it.

If I get some time I might try capturing the network traffic that happens when an esp node comes up, does discovery and gets connected to by HA, if that would be helpful.

openshwprojects, 2023-01-15:
I will look into it once our Windows Simulator is released and functional. We will also increase compatibility with Tasmota as well.

agittins, 2023-01-15:
Awesome. FWIW I just discovered the libretuya-esphome project as well - I lean more toward the esphome way of doing things rather than the tasmota path, so I will probably spend a bit of time looking into that project as well.

Y'all are doing amazing work!

iprak, 2023-01-16:
I had looked into it because a good number of my esp devices are on espHome. Then I realized that a fair number of end points would need to be implemented to get something working.

I don't think there is any discovery involved, right?  espHome device gets added via its integration which uses the ip address of the device.

agittins, 2023-01-16:
@iprak the discovery process uses mDNS. The esphome device broadcasts its _esphome._tcp service, which HA sees, and then presents the user the option to add the device to the system. So the initial discovery is via mdns (but you can also add devices manually by IP), then HA talks to the esp device directly via tcp (the esp device runs a protobuf service, and HA connects to it as a client - mostly as described above, except anywhere I've misunderstood! 🤣 ).

iprak, 2023-01-16:
Thank you for the clarification. I figured out my issue. I was seeing too many random devices so turned off zeroconf in HomeAssistant.

btsimonh, 2023-01-16:
Ref mdns, look at the SSDP implementation...  it may be we can expand on that.

agittins, 2023-01-21:
So.. I've had success with the libretuya-esphome project, so I'll be using that for all my bk devices from here I think, as it just fits well with how I have my setup configured and the way I like to structure things. For anyone who likes the tasmota-like style of doing things this project is and will continue to be the way to go, I am sure.

(this is like writing a break-up letter!) so I won't have a use/need for the HA api in this project personally, so I'll leave it up to you and others if you want to keep this issue open as a feature request, or decide to close it and re-open if some interest reignites at a later stage. Thanks for the awesome work you've done in opening these devices up for everyone!

closed 2023-01-21 by openshwprojects