openshwprojects/OpenBK7231T_App #610: REQUEST New driver support - SM2235EGH

EnginDzhemil, 2023-01-15:
Hello,

I have started migrating my tuya lights to openbeken using tuyacloudcutter.
Most of them are BP1658CJ or PWM based.
But I couldn't get couple of them to work so I opened one and found a SM2235EGH chip inside.
Is it possible to support it?

Thank you for the great work you are doing on openbeken.

openshwprojects, 2023-01-15:
Hello, would you have time to check builds if we were to attempt supporting it?

EnginDzhemil, 2023-01-15:
> Hello, would you have time to check builds if we were to attempt supporting it?

Yes, I will do my best to test.

openshwprojects, 2023-01-15:
Which pins are used for I2C in this case?

openshwprojects, 2023-01-15:
Can you check? Do you know how to check?
https://github.com/openshwprojects/OpenBK7231T_App/commit/bd84eb4878170c83a7743d205dc747bb9bd1160f
1. set pins
2.  startDriver SM2235
3 SM2235_RGBCW FF00000000

EnginDzhemil, 2023-01-16:
How can I check which pins are used?
I will have time tonight to test.

korexeu, 2023-01-16:
Hi, I have also RGBW light with SM2235 driver.
P24 is data PIN
P26 is CLK PIN
(Kanlux 33643 S GU10 RGBCCT Bulb)
from MSB to LSB
Blue Red Green WW CW
I uploaded - Build on Jan 16 2023 13:08:55 version 1.15.336
I can turn on bulb from Command Tool but from main menu I can't control bulb in right way. (WW is swapped with CW).
Increasling Brightness slider sometimes give more light, and next increase give lower LED power.

EnginDzhemil, 2023-01-16:
My bulbs are from Moes with BK7231N.
I tried several pin combinations but couldn't make it work.
Looks like I will have to sacrifice one and take it apart to check with multimeter where the pins are connected.
And I will make some photos for the database.

openshwprojects, 2023-01-16:
Please do a bulb teardown and check connections with multimeter.
Please submit your bulbs here, it will be very helpful for us and future users:
https://openbekeniot.github.io/webapp/devicesList.html
@korexeu the channel order can be simply modified with the mapping command:
[image]

What else is going on wrong with the driver? Can you be a bit more specific?

I will try to look at the code again tomorrow. For now, I've set default brightness values and always use 5 channels mode.

It could be improved a lot soon, I guess.

"Increasling Brightness slider sometimes give more light, and next increase give lower LED power."
This is strange but maybe first it would be better to test with console command: SM2235_RGBCW FF00000000

korexeu, 2023-01-17:
Thanks for help. After SM2235_Map comment control coloures is good.
For my bulb map is SM2235_Map 1 0 2 3 4
When I used commands mentioned by you above, its give effects which you said.
But when I increasing slider point by point on main page, it gives me below effects:
max brightness, low brightness, max brightness, medium brightness, etc...

BR
korexeu

openshwprojects, 2023-01-17:
@korexeu  I think I know why, I will submit an update tomorrow

openshwprojects, 2023-01-17:
@korexeu I didn't want to wait so I submitted quickly a supposed fix. Please tell me how it behaves now

korexeu, 2023-01-18:
> @korexeu I didn't want to wait so I submitted quickly a supposed fix. Please tell me how it behaves now

Now its working well. Thanks for your help.

openshwprojects, 2023-01-18:
@korexeu  wait wait but we didn't add a current adjustment yet? You can change the current value to make LED brighter. I've added default setting only.

korexeu, 2023-01-18:
@openshwprojects, what do you mean ?
When I using commands for start sm2235 driver and pin mapping, I can control all parameters (colors, temperature and brightness) of my bulb.

openshwprojects, 2023-01-19:
Of course, brightness will work, but there is also a current setting, which allows you to change the total current for LEDs, which can make them even brighter or less brighter.There are different current values used in the Tuya devices, so you might one day realize that now obk leds are darker than with original firmware... that's because they may be using higher current setting

So I will be making current setting adjustable as well soon.

It's good to note that on the other hand, if you set a higher current setting than manufacturer planned, you might overheat and damage your LEDs.

EnginDzhemil, 2023-02-17:
Hi, I finally got the time and found the pins. Everything works great. I will contribute photos of the internals on the forum. Thank you very much! I am closing the issue.

closed 2023-02-17 by EnginDzhemil