openshwprojects/OpenBK7231T_App #611: Rocker switch sync

DiegoJp, 2023-01-15:
Hello, I have installed OpenBeken in a Aubess MINI Wifi switch like this:
[image]

I loaded the config from the webapp like this: 
[image]

However is missing a functionality from the Smartlife/Tuya app in which the button status (S1 and S2 connection) is synced to the channel status. It's called like 'switch type setting' and the options as follows:

[image]

- rocker switch - flip to change state (this is the default behaviour in Tuya fw and OpenBeken fw)
      - On physical button flip to **on**, it will toggle channel (if it's on will turn to off and viceversa)
      - On physical button flip to **off**, it will toggle channel (if it's on will turn to off and viceversa)
- rocker switch - state synchronous (this functionality is missing)
      - On physical button flip to **on** will turn on channel, if channel it's already on **it will remain on.** 
      - On physical button flip to **off** will turn off channel, if channel it's already off **it will remain off.**

This funcionality is useful for physical switches that are labelled as "On / Off" and it's more intuitive than the default behaviour.
I hope this is understandable. I have not tried to make any changes in Config module as it seems that are not well documented in the github wiki.

Keep the good work, many thanks in advance!

openshwprojects, 2023-01-15:
Hello, I think you are looking for "ToggleChannelOnToggle" pin role?

DiegoJp, 2023-01-15:
> Hello, I think you are looking for "ToggleChannelOnToggle" pin role?
> 
> Or just for dInput?
> 
> state synchronous is dInput, digital input, copies values to channel. Albo exists in dInput_n form, an inversed one

Yeah, that wast it! Set the corresponding pin to Dinput_n and now works as intended!
[image]

Many thanks

closed 2023-01-15 by DiegoJp