openshwprojects/OpenBK7231T_App #613: Bad Home Assistant Discovery Queue Message

hftsai256, 2023-01-16:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version 1.15.323
- Device? DayBetter 800 Lumen 120V 9W RGB+2700-6500K Bulbs
- Chip/model: BK7231N/CB2L/BP5758D
- Device Flags: 0, 2, 10, 27, 28
- Startup Command: `backlog BP5758D_Map 2 1 0 3 4; led_enableAll 1; led_temperature 312; BP5758D_Current 30`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Config/Home Assistant Configuration'
2. Click on 'Start Home Assistant Discovery'
3. See error in logs

**Screenshots**
[image]

**Additional context**
This function was working properly on FW version 1.15.279. Seems like a pointer/stack memory corruption to me.

openshwprojects, 2023-01-16:
Can you please narrow more between release versions when it broke?

iprak, 2023-01-16:
Once I got the  Win32 app configured, investigating this was very easy. It was one of my merges where a line was lost. I just pushed out a PR for this.

openshwprojects, 2023-01-16:
Sorry, I was faster, I think....
Btw please also see this @iprak : 
https://github.com/openshwprojects/OpenBK7231T_App/commit/18fae1fe0e6b23d3c2389be9f8531b894752b114
the const string should be const and you also HAVE TO set pointer to NULL after freeing (otherwise it points at memory area even after it was fried)

openshwprojects, 2023-01-16:
@iprak  you're gonna like this:
https://github.com/openshwprojects/OpenBK7231T_App/commit/2932c718b7265c42a47157e95cd5d0bef6836fcf
This is automatic testing of HA discovery so issues like this one will never happen again.

Please tell me if you have Win32 version running and would like to fill those functions:
[image]
I need to know because otherwise I'll do it myself.

iprak, 2023-01-16:
> Sorry, I was faster, I think.... Btw please also see this @iprak : [18fae1f](https://github.com/openshwprojects/OpenBK7231T_App/commit/18fae1fe0e6b23d3c2389be9f8531b894752b114) the const string should be const and you also HAVE TO set pointer to NULL after freeing (otherwise it points at memory area even after it was fried)

Yes resetting it to NULL would definitely prevent accidental misuse. 

Sorry, I don't follow why the return value should be `const`. Should `dev_info` be defined as const too ?

iprak, 2023-01-16:
> @iprak you're gonna like this: [2932c71](https://github.com/openshwprojects/OpenBK7231T_App/commit/2932c718b7265c42a47157e95cd5d0bef6836fcf) This is automatic testing of HA discovery so issues like this one will never happen again. https://github.com/openshwprojects/OpenBK7231T_App/blob/2932c718b7265c42a47157e95cd5d0bef6836fcf/src/selftest/selftest_hass_discovery.c Please tell me if you have Win32 version running and would like to fill those functions: [image] I need to know because otherwise I'll do it myself.

I am having a hard time getting Win32/Simulator to build/run. It is not clear how it should be build/used successfully. My main dev platform also is VSCode.

openshwprojects, 2023-01-17:
@iprak  I will try to add libs to another repo and prepare a very easy build mechanism for people soon

iprak, 2023-01-17:
This is my first hand experince of the Simulator/Win32 project. I feel that it needs a fair amount of clarification for usage/build and maybe is something for advanced users.

Usage:
* What is it? I was under the impression that Simulator was a different app and didn't know that Win32 project launched the simulator.
* What does simulator do? Upon successful build, you see a layout of some components. Then what?
* I did not realize that one could navigate to http://127.0.0.1
* It spits out a lots of error messages/states into the console and that is where the every_second message also dumps making identifying issue difficult.

Build:
* The Win32 project is based on VS2017 which is like 5-6 years old now and targets platform 10.0.17763.0 I(which is October 2018 Update, version 1809). The projects compiles on latest components/VS. Please consider updating it. The alternative is to install vs_buildtools and the required target platform.
* As identified, many libraries are needed to be downloaded. The `nativefiledialog` library itself needs to be build. We should include script/instructions to download them. We discussed that their licenses are slightly different and this project has no license (#191).
* A sln should be included, any change results in a prompt by VS to save the auto generated sln. So one can be included, right?
* First compile fails because the default build configuration is Win32 and one needs to pick "Debug Win32". I am pretty sure "Debug Win32" can be defaulted.
* Even if compilation is successful, the run will fail because dependent assemblies need to be manually copied into "Debug Win32" folder.