openshwprojects/OpenBK7231T_App #615: BK7231T Bug  button switching

az-src, 2023-01-17:
Hello. The error of the buttons operation migrated from the original SDK. Or rather, it doesn't work correctly. The relay switches on a falling edge (finger release). Very uncomfortable and uncomfortable. Help make buttons work on rising edge(finger touch)

openshwprojects, 2023-01-17:
Hello, you are most likely looking for a "Instant Touch Reaction" flag which is available in flags menu.

az-src, 2023-01-17:
I'm not sure, but I'll check now, thanks

az-src, 2023-01-17:
Thanks a lot !!! I didn't know where to look. Everything works great!!!

openshwprojects, 2023-01-20:
By the way, it's not a bug, it's a feature.

Without delayed click event, the "double click" would fire both a click and dbl click.

closed 2023-01-20 by openshwprojects