openshwprojects/OpenBK7231T_App #617: Stability issues [BK7321T Hangs around 3.03 AM]

NielsPiersma, 2023-01-19:
I've 40 BK7321T in the office. Roughly 16 are set up for ceiling lights, six are set up for switching desks, and the others are lying in our meeting room, dormant but powered on and connected.

Almost every night at 3.03 AM, about 3 to 6 plugs go offline (except the dormant ones in the meeting room). Disconnecting and reconnecting the plug resolves the issue. But I need help finding an actual root cause.

1) The devices are randomly halting, but only the ones connected and have a load. The devices I am not using actively, stay connected.
2) It happens almost always at 3.03 (it could be 3.00 AM, but the reports show 3.03)
3) they are connected to different Unify APs
4) I've updated the firmware daily in the last week, but no improvement. 

For now, I'll schedule a reboot of the device at 02:00 pm and see if that mitigates the issue, but better would be if the device doesn't stop,

NIels

openshwprojects, 2023-01-19:
Hey, what kind of device, maybe it has a low quality power supply and you need a [b]PowerSave[/b] command in short startup command?
If not, you can try checking the release that was tested many times and is deemed as very stable:
[image]

Build on Nov 25 2022 10:36:03 version 1.15.77

NielsPiersma, 2023-01-19:
Thanks,
I'll try the power save first. As Build on Nov 25 2022 10:36:03 version 1.15.77 is lacking a lot of features I need for management that won't be a feasible solution.

NIels

openshwprojects, 2023-01-19:
It would be also way easier if you manage to narrow down a commit which breaks things for you. 
I also remember that there was some small stability issue caused by logging, I am not sure when it was, few weeks ago? @btsimonh solved that. Or it was a month ago...

Which version do you have @NielsPiersma  ?

NielsPiersma, 2023-01-23:
Guys (and girls),
I likely found the root cause. Our Unify network controller is executing channel optimization precisely at 03.00 am.
As it is too big of a coincidence that some BK devices lose their connections and don't reconnect at 03.04 am.

I'll disable nightly optimization and see if they stay connected.

Niels

[image]
AP excluded from Channel Optimization.

openshwprojects, 2023-01-25:
It sounds like a non-OBK issue. Did turning off optimizations help?

NielsPiersma, 2023-01-25:
@openshwprojects, yes, turning it of resolved the issue. I won't necessarily say it is not an OBK issue. Other connected devices don't have issues with Channel Optimization and reconnect fine after selecting a new channel with less interference.

Wifi Optimization is used by multiple vendors, such as Cisco, Aruba, Unify, and Axiros. 

For us, disabling this feature works fine. Many home users will not have this issue anyway.

I would recommend having a look, but no rush at this moment.

Niels

openshwprojects, 2023-01-25:
@NielsPiersma  ok ok, I still have your Counter request on TODO list as well. I was busy with WEMO and BL602 OTA and ioBroker fixes (now we are REALLY Tasmota compatible, very high compatibility level), but you probably saw it in commit log anyway, right? Very busy days, but progress is extremely quick.

Being able to run firmware on Windows (with MQTT) is a game changer.

jf-wm, 2023-01-27:
I happened to see NielsPiersma's solution to his problem, as I also have an OBK device that crashes at night at irregular intervals. For me, too, this problem is apparently triggered by the Wifi channel optimization of my router (FritzBox). However, a Tasmota device (ESP8266EX) right next to the OBK is not affected by Wifi channel optimization, it has never crashed.

NielsPiersma, 2023-01-27:
@jf-wm , 
As far as I can diagnose at this moment it is the optimizing process causing the obk getting confused and not reconnecting. In my setup esp82xx are not affected. 

Thanks for confirming this. 

Niels

openshwprojects, 2023-01-28:
@NielsPiersma what about using Ping Watchdog with scriptable event to force device reboot?