openshwprojects/OpenBK7231T_App #623: Issues with Flag 12, Flag 18 and homeassistant with Bl602 bulbs

zain1144, 2023-01-21:
Hello, 

I have a bulbs that is using BL602 with an led driver BP5758D

The first issue is when enabling Flag 12 the bulb turn on all LEDs on the board on Power on, when this happens I have to adjust the brightness and the color temperature slides to make the bulb work normally again.
https://i.ibb.co/yPV0WjG/20230121-152453.jpg

Also, if I enable Flag 12 and Flag 18 the bulb turn on all LEDs on the board but with a pulse and doesn't stop until I disable flag 18

https://www.youtube.com/watch?v=4dLWyafSxXA

Second issue is the bulb on Power On doesn't report the led state to homeassistant. Since I can't set up Flag 12 I put this command on startup `backlog led_dimmer 100; led_temperature 500; led_enableAll 1;BP5758D_Current 50;BP5758D_Map 2 1 0 3 4` . When the bulb turn on homeassistant is showing OFF and it doesn't report the state until I change brightness or color either in homeassistant or in openbeken page

I have a second bulb that uses BL602 & SM2135 and it does the same behavior as BL602 & BP5758D

I also have a bulb with BK7231N, this one https://www.elektroda.com/rtvforum/viewtopic.php?p=20396948 and no issues with this one at all, Flag 12 & 18 and homeassistant all working perfectly.

openshwprojects, 2023-01-21:
Hello @zain1144 , the BL602-specific issues are harder to diagnose than a generic ones, but I will try to think about something...
First of all, do you happen to know if the problem also occurs with the simple PWM-based LEDs on BL602 platform?

" homeassistant is showing OFF and it doesn't report the state until I change brightness or color either in homeassistant or in openbeken page"
do you have those?
[image]

zain1144, 2023-01-21:
Unfortunately I don't have BL602 PWM-based LED.
Regarding Flag 10 it was on by default, I enabled Flag 2 but it didn't solve the issue.

zain1144, 2023-01-21:
Another thing I noticed 
For BL602 bulb when I do restart and open the log I see these MQTT messages 

Info:MQTT:Publishing val Ikea_Lamp to Ikea_Lamp/host retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -13 result 0
Info:MAIN:Time 13, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val Build on Jan 18 2023 12:04:57 version BL602_Test to Ikea_Lamp/build retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -12 result 0
Info:MAIN:Time 14, idle 0/s, free 99864, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/21 
Info:MQTT:Publishing val 7CB94C1EBA13 to Ikea_Lamp/mac retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -11 result 0
Info:MAIN:Time 15, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val 2 to Ikea_Lamp/sockets retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -8 result 0
Info:MAIN:Time 16, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val -1 to Ikea_Lamp/rssi retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -7 result 0
Info:MAIN:Time 17, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val 17 to Ikea_Lamp/uptime retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -6 result 0
Info:MAIN:Time 18, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val 102400 to Ikea_Lamp/freeheap retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -5 result 0
Info:MAIN:Time 19, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val 192.168.0.6 to Ikea_Lamp/ip retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -4 result 0

For BK7231N bulb this the MQTT at the start up

Info:MQTT:Publishing val *******  to Ikea_Lamp/mac retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -11 result 0
Info:MAIN:Time 32, idle 177031/s, free 75392, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val 2 to Ikea_Lamp/sockets retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -8 result 0
Info:MAIN:Time 33, idle 188871/s, free 75392, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val -63 to Ikea_Lamp/rssi retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -7 result 0
Info:MAIN:Time 34, idle 186368/s, free 75392, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val 34 to Ikea_Lamp/uptime retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -6 result 0
Info:MAIN:Time 35, idle 183683/s, free 74944, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MQTT:Publishing val 75392 to Ikea_Lamp/freeheap retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -5 result 0
Info:MAIN:Time 36, idle 193816/s, free 75160, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val 192.168.0.196 to Ikea_Lamp/ip retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -4 result 0
Info:MAIN:Time 37, idle 183403/s, free 75392, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val 1 to Ikea_Lamp/led_enableAll/get retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -3 result 0
Info:MAIN:Time 38, idle 184522/s, free 75152, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:Publishing val 500 to Ikea_Lamp/led_temperature/get retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -2 result 0
Info:MAIN:Time 39, idle 368498/s, free 75176, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MQTT:Publishing val 100 to Ikea_Lamp/led_dimmer/get retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -1 result 0

I seems that the BK7231N send the LED status where as BL602 doesn't, did I understand this right ?

openshwprojects, 2023-01-21:
What happens in log after that part:
_Info:MAIN:Time 19, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21
Info:MQTT:Publishing val 192.168.0.6 to Ikea_Lamp/ip retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -4 result 0_
on BL602?
The item -3 is led_enableAll, is it even attempted to be published?

zain1144, 2023-01-21:
Nothing happens, it stopes publishing

Info:MAIN:Time 3039, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 3040, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 3041, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val 2 to Ikea_Lamp/sockets retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -8 result 0
Info:MAIN:Time 3042, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val -1 to Ikea_Lamp/rssi retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -7 result 0
Info:MAIN:Time 3043, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val 3043 to Ikea_Lamp/uptime retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -6 result 0
Info:MAIN:Time 3044, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val 102400 to Ikea_Lamp/freeheap retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -5 result 0
Info:MAIN:Time 3045, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:Publishing val 192.168.0.6 to Ikea_Lamp/ip retain=0
Info:MQTT:[g_bPublishAllStatesNow] item -4 result 0
Info:MAIN:Time 3046, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 3047, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 3048, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 3049, idle 0/s, free 96776, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/21 
Info:MAIN:Time 3050, idle 0/s, free 102400, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21

openshwprojects, 2023-01-22:
@zain1144  I really need to get such bulb...
So far, I added OTA for BL602 so we can develop faster. Now I want to solve your issues, but I am not sure how to approach them....

can you tell me do you have any MQTT errors here?
[image]

zain1144, 2023-01-22:
Thank you for the OTA update, I flashed the bulb to the latest firmware.

I don't see MQTT errors

[image]

hftsai256, 2023-04-02:
Hi all. I was about to create a new ticket but I then found this one resonance with my issues. My bulbs are [Daybetter 800 Lumens](https://www.elektroda.com/rtvforum/topic3925050.html#20227148) on BP5758D as well. I feel the root cause is that such devices don't broadcast critical information to proper MQTT topics on boot. For example:

OpenBeken would publish the following information on color temp change:
```
Info:MQTT:Publishing val 234 to obk177FF51C/led_temperature/get retain=0
```
and HASS would capture such events and update its database & UI:
```
Message 61 received on obk177FF51C/led_temperature/get at 11:56 PM:
265
```
Nevertheless, those boot messages (flag 10) and self-state broadcasting (flag 2) do not include interesting topics such as `led_dimmer`, `led_temperature`, `led_enableAll`, etc.
```
Info:MQTT:Publishing val 4 to obk177FF51C/sockets retain=0
Info:MQTT:Publishing val -28 to obk177FF51C/rssi retain=0
Info:MQTT:Publishing val 1108 to obk177FF51C/uptime retain=0
Info:MQTT:Publishing val 73008 to obk177FF51C/freeheap retain=0
Info:MQTT:Publishing val 10.0.0.121 to obk177FF51C/ip retain=0
```
Therefore, HASS has no idea about the status of those newcomers. I'm wondering if there is a command to manually publish that information when MQTT connects.

openshwprojects, 2023-04-02:
I haven't seen it yet, let me check

openshwprojects, 2023-04-02:
there is a bug for I2C LED drivers: 
[image]

openshwprojects, 2023-04-02:
I am sorry for late reply, but why don't you open a new issue when new issue arise?

Here's part 1 of fix:
https://github.com/openshwprojects/OpenBK7231T_App/commit/7714a0ab25b908a40e7410705c4d86fc27dcfa63
I am in the middle of fixing, but now I see there was a serious bug with "broadcast self state" and self state was skipping LED params for LEDs that has I2C LED driver.