openshwprojects/OpenBK7231T_App #624: Please give me I2C sensors/peripherals ideas to support. Or maybe other sensors as well..

openshwprojects, 2023-01-21:
We already have both hardware I2C and software I2C working, so I would like to add some more drivers for sensors.
Can you give me some more suggestions for popular sensors used with Arduino, preferably I2C (but not necessary)?

You can also give me generic sensors suggestions, we already have support for DHT11 family, etc.

TheMasterFX, 2023-01-24:
I like to use BME/BMP 280 for my weather station, actual prices for the modules are quite high due to chip shortage (normally they where like 2€)

syfq91, 2023-01-28:
+1 for BME280. Besides that I also have BH1750 light sensor & HTU21D temp/humidity sensor

Xinayder, 2023-01-30:
Dallas temperature sensors; BH1750; Bosch temperature/humidity sensors. AHT30 from Adafruit.

dilpish, 2024-12-01:
+BH1750 or TSL2561

Light sensors are really useful. 

and may be:
HC-SR04