openshwprojects/OpenBK7231T_App #627: Hue or Wemo Integration/Emulation

TheMasterFX, 2023-01-22:
Is there any chance to get Wemo/Hue Emulation to make the device directly usable from Alexa etc.?

openshwprojects, 2023-01-22:
Hello, yes, would you be able to help with testing that feature? I don't have Alexa.

TheMasterFX, 2023-01-22:
Of cause. I can test and flash every day. This might be a good starting point: [Tasmota Wemo Emulation](https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_21_wemo.ino)

openshwprojects, 2023-01-23:
is it possible to check it without having Alexa?

Or can you help with testing?

I added unfinished version and I need to know how it reacts. No switch control yet. I just want to know if msearch works can is received correctly.

Update to latest version.
https://github.com/openshwprojects/OpenBK7231T_App/releases

Do a command:
backlog startDriver WEMO; startDriver SSDP

What are the values here?
[image]
Are you able to get something in logs?

TheMasterFX, 2023-01-23:
Since Wemo is basicaly a Hue Emulation this might work: https://pypi.org/project/discoverhue/
I tried the new firmware and started device discovery at Alexa/Echo but now search was received
[image]

TheMasterFX, 2023-01-23:
I didn't got the discoverhue to work. I also tried many different go projects which doesn't work either.

But when I start the discovery on alexa I see some thing happen in the Log of the OpenBK:
```
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 3037, idle 242446/s, free 85672, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Debug:MAIN:PS: 0ms/1467998ms longests:0ms/0ms req:24ms/4294967276ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 120927 0 8 394 0
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:HTTP:DRV_SSDP_Service_Http
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 3038, idle 247680/s, free 85672, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Debug:MAIN:PS: 0ms/1458352ms longests:0ms/0ms req:24ms/4294967276ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 120259 0 0 526 0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 3039, idle 244466/s, free 85672, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Debug:MAIN:PS: 0ms/1579450ms longests:0ms/0ms req:24ms/4294967276ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 128079 0 0 12 0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 3040, idle 252040/s, free 85672, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Debug:MAIN:PS: 0ms/1523968ms longests:0ms/0ms req:24ms/4294967276ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 125789 0 0 31 0
Info:GEN:dhcp=0 ip=192.168.178.62 gate=192.168.178.1 mask=255.255.255.0 mac=a0:92:08:f8:62:1e 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-60,ssid=FRITZ!Box 7530 JK,bssid=1c:ed:6f:f0:8a:64 ,channel=9,cipher_type:CCMP
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:HTTP:DRV_SSDP_Shutdown
Info:MAIN:Drv SSDP has been stopped.
Info:HTTP:DRV_SSDP_Init
Info:HTTP:DRV_SSDP_CreateSocket_Receive: Socket created, waiting for packets
```

TheMasterFX, 2023-01-23:
Hi, I did activate loglevel 7 and filtered for HTTP & HTTP_CLIENT. Hope this Helps
```
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: space: 298 msg:244
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: 
NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:0acbc1e6-e0ec-4b62-1c33-86462710ec14::upnp:rootdevice

ExtraDebug:HTTP:DRV_SSDP_Send_Notify: sent message 244 bytes
ExtraDebug:HTTP:Received 244 bytes from 192.168.178.62
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:0acbc1e6-e0ec-4b62-1c33-86462710ec14::upnp:rootdevice

ExtraDebug:HTTP:NOTIFY from a peer device
ExtraDebug:HTTP:SSDP obk device still present 0x3eb2a8c0
ExtraDebug:HTTP:Received 94 bytes from 192.168.178.29
ExtraDebug:HTTP:data: M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: ssdp:all
MAN: "ssdp:discover"
MX: 3

ExtraDebug:HTTP:Is MSEARCH - responding
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
ExtraDebug:HTTP:Received 101 bytes from 192.168.178.29
ExtraDebug:HTTP:data: M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
ST: upnp:rootdevice
MAN: "ssdp:discover"
MX: 3

ExtraDebug:HTTP:Is MSEARCH - responding
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
ExtraDebug:HTTP:Received 107 bytes from 192.168.178.28
ExtraDebug:HTTP:data: M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:Belkin:device:**

ExtraDebug:HTTP:Is MSEARCH - responding
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
ExtraDebug:HTTP:Received 122 bytes from 192.168.178.28
ExtraDebug:HTTP:data: M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:schemas-upnp-org:device:basic:1

ExtraDebug:HTTP:Is MSEARCH - responding
Debug:HTTP:DRV_SSDP_Send_Advert_To: sent message
Debug:HTTP:DRV_SSDP_Service_Http
ExtraDebug:HTTP:Received 122 bytes from 192.168.178.28
ExtraDebug:HTTP:data: M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:schemas-upnp-org:device:basic:1

[... 282 lines cut ...]
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: 
NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:4916afd0-da69-82e8-215a-520425172d4e::upnp:rootdevice

ExtraDebug:HTTP:DRV_SSDP_Send_Notify: sent message 244 bytes
ExtraDebug:HTTP:Received 244 bytes from 192.168.178.62
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:4916afd0-da69-82e8-215a-520425172d4e::upnp:rootdevice

ExtraDebug:HTTP:NOTIFY from a peer device
ExtraDebug:HTTP:SSDP obk device still present 0x3eb2a8c0

```

openshwprojects, 2023-01-23:
It seems very helpful. Please check the updated version and see how it reacts.
https://github.com/openshwprojects/OpenBK7231T_App/commit/bfd9a4dbbd1eeda15c26ba74531a052d1c81ef6d
It will not allow control yet.

I need the log from latest version, especially for two things:
1. is my MSEARCH parsing now ok?
2. do you get that in log?
[image]

TheMasterFX, 2023-01-24:
Here you go:
[image]
And the Log (Level 7, All)
```
Debug:HTTP:DRV_SSDP_Service_Http
ExtraDebug:HTTP:Received 107 bytes from 192.168.178.28
ExtraDebug:HTTP:data: M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:Belkin:device:**

ExtraDebug:HTTP:Is MSEARCH - responding
ExtraDebug:HTTP:WEMO - sends reply urn:Belkin:device:**
ExtraDebug:HTTP:WEMO - Sending HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Fri, 15 Apr 2016 04:56:29 GMT
EXT:
LOCATION: http://192.168.178.62:80/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: b9200ebb-736d-4b93-bf03-835149d13983
SERVER: Unspecified, UPnP/1.0, Unspecified
ST: urn:Belkin:device:**
USN: uuid:Socket-1_0-20161208F8621E::urn:Belkin:device:**
X-User-Agent: redsonic

ExtraDebug:HTTP:Received 107 bytes from 192.168.178.28
ExtraDebug:HTTP:data: M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:Belkin:device:**

ExtraDebug:HTTP:Is MSEARCH - responding
ExtraDebug:HTTP:WEMO - sends reply urn:Belkin:device:**
ExtraDebug:HTTP:WEMO - Sending HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Fri, 15 Apr 2016 04:56:29 GMT
EXT:
LOCATION: http://192.168.178.62:80/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: b9200ebb-736d-4b93-bf03-835149d13983
SERVER: Unspecified, UPnP/1.0, Unspecified
ST: urn:Belkin:device:**
USN: uuid:Socket-1_0-20161208F8621E::urn:Belkin:device:**
X-User-Agent: redsonic

Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 164, idle 252256/s, free 76152, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38 
Debug:MAIN:PS: 0ms/1502074ms longests:0ms/0ms req:24ms/4294967280ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 122886 0 0 999 0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: space: 298 msg:244
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: 
NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:6b4a27a4-47b6-490e-5ea3-1a7c650f05f1::upnp:rootdevice

ExtraDebug:HTTP:DRV_SSDP_Send_Notify: sent message 244 bytes
Info:MAIN:Time 171, idle 241983/s, free 84976, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Debug:MAIN:PS: 0ms/1510844ms longests:0ms/0ms req:24ms/4294967280ms unkn HTTP Client
Debug:MAIN:PS: nosleep reasons 122130 0 0 350 0
ExtraDebug:HTTP:Received 244 bytes from 192.168.178.62
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:6b4a27a4-47b6-490e-5ea3-1a7c650f05f1::upnp:rootdevice

ExtraDebug:HTTP:NOTIFY from a peer device
ExtraDebug:HTTP:SSDP obk device still present 0x3eb2a8c0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 172, idle 233153/s, free 76152, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38 
Debug:MAIN:PS: 0ms/1394616ms longests:0ms/0ms req:24ms/4294967280ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 113869 0 0 517 0

```

openshwprojects, 2023-01-24:
Nothing more? No page visits by Alexa? Nothing if you try to see if Alexa sees device or can toggle it?

TheMasterFX, 2023-01-24:
No Alexa says no new devices found. My devices list only shows my known devices

openshwprojects, 2023-01-24:
Oh no! Wrong variable name! That's how doing things without testing ends, shame on me:
[image]
https://github.com/openshwprojects/OpenBK7231T_App/commit/0003de3970ee3bb4fdb271ce83470e1f34f63f9f
can you check now?

TheMasterFX, 2023-01-24:
Looks good:
[image]

Alexa sees now one new device but it doesn't switch on/off. But this is a huge progress i think

```

ExtraDebug:HTTP:Is MSEARCH - responding
ExtraDebug:HTTP:WEMO - sends reply upnp:rootdevice
ExtraDebug:HTTP:WEMO - Sending HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Fri, 15 Apr 2016 04:56:29 GMT
EXT:
LOCATION: http://192.168.178.62:80/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: b9200ebb-736d-4b93-bf03-835149d13983
SERVER: Unspecified, UPnP/1.0, Unspecified
ST: upnp:rootdevice
USN: uuid:Socket-1_0-20161208F8621E::upnp:rootdevice
X-User-Agent: redsonic

ExtraDebug:HTTP:Is MSEARCH - responding
ExtraDebug:HTTP:WEMO - sends reply upnp:rootdevice
ExtraDebug:HTTP:WEMO - Sending HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Fri, 15 Apr 2016 04:56:29 GMT
EXT:
LOCATION: http://192.168.178.62:80/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: b9200ebb-736d-4b93-bf03-835149d13983
SERVER: Unspecified, UPnP/1.0, Unspecified
ST: upnp:rootdevice
USN: uuid:Socket-1_0-20161208F8621E::upnp:rootdevice
X-User-Agent: redsonic

ExtraDebug:HTTP:Is MSEARCH - responding
ExtraDebug:HTTP:WEMO - sends reply urn:Belkin:device:**
ExtraDebug:HTTP:WEMO - Sending HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Fri, 15 Apr 2016 04:56:29 GMT
EXT:
LOCATION: http://192.168.178.62:80/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: b9200ebb-736d-4b93-bf03-835149d13983
SERVER: Unspecified, UPnP/1.0, Unspecified
ST: urn:Belkin:device:**
USN: uuid:Socket-1_0-20161208F8621E::urn:Belkin:device:**
X-User-Agent: redsonic

Info:HTTP:Wemo post event <?xml version="1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/"s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 79, idle 196258/s, free 66192, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 5/38 
Debug:MAIN:PS: 0ms/1007488ms longests:0ms/0ms req:24ms/4294967276ms unkn TCP_server
Debug:MAIN:PS: nosleep reasons 86169 0 0 294 0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MAIN:Time 80, idle 257084/s, free 86512, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Debug:MAIN:PS: 0ms/1469044ms longests:0ms/0ms req:24ms/4294967276ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 120896 0 0 413 0
Info:GEN:dhcp=0 ip=192.168.178.62 gate=192.168.178.1 mask=255.255.255.0 mac=a0:92:08:f8:62:1e 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-61,ssid=FRITZ!Box 7530 JK,bssid=1c:ed:6f:f0:8a:64 ,channel=9,cipher_type:CCMP
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: space: 298 msg:244
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: 
NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:64d9bc2b-aa76-bf5c-919d-d77a13acab24::upnp:rootdevice

ExtraDebug:HTTP:DRV_SSDP_Send_Notify: sent message 244 bytes
Info:MAIN:Time 82, idle 242256/s, free 86200, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Debug:MAIN:PS: 0ms/1385972ms longests:0ms/0ms req:24ms/4294967276ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 116479 0 0 586 0
ExtraDebug:HTTP:Received 244 bytes from 192.168.178.62
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:64d9bc2b-aa76-bf5c-919d-d77a13acab24::upnp:rootdevice

ExtraDebug:HTTP:DRV_SSDP_Send_Notify: space: 298 msg:244
ExtraDebug:HTTP:DRV_SSDP_Send_Notify: 
NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:64d9bc2b-aa76-bf5c-919d-d77a13acab24::upnp:rootdevice

ExtraDebug:HTTP:DRV_SSDP_Send_Notify: sent message 244 bytes
Info:MAIN:Time 142, idle 261623/s, free 84936, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/38 
Debug:MAIN:PS: 0ms/1489172ms longests:0ms/0ms req:24ms/4294967276ms unkn timer_thd
Debug:MAIN:PS: nosleep reasons 123809 0 0 707 0
ExtraDebug:HTTP:Received 244 bytes from 192.168.178.62
ExtraDebug:HTTP:data: NOTIFY * HTTP/1.1
SERVER: OpenBk
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.178.62:80/ssdp.xml
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:64d9bc2b-aa76-bf5c-919d-d77a13acab24::upnp:rootdevice
```

openshwprojects, 2023-01-24:
There is no on/off code yet. I will add it tomorrow morning. See you in about 12 hours, I just need some sleep.

Btw, once the Wemo driver works, are you able to test the Hue driver as well?

openshwprojects, 2023-01-25:
@TheMasterFX I've added event handling, can you check?

By the way, if it works, would you be kind enough to do some kind of short presentation of this feature, post on our forum with title something like "OpenBeken Alexa Wemo setup guide - pair BK7231 directly with Alexa", put a device configuration, some screenshots from obk and the alexa stuff, maybe even a video how it works? 
https://www.elektroda.com/rtvforum/forum390.html
It would be very helpful and I don't have alexa to do it myself.
If you do some detailed presentation, maybe you get even try to get into this programme: https://www.elektroda.com/rtvforum/topic3950844.html

TheMasterFX, 2023-01-25:
Hi, sorry I just came home from work.
Switching seems to work. In my case it switches Channel 1. How can I autoload the WEMO and SSDP driver?
[image]

openshwprojects, 2023-01-25:
Short startup command or autoexec.bat in LittleFS.

So basic version works. What do we do now? A multi channel version or something? A hue integration?

TheMasterFX, 2023-01-25:
Yesmultichannel would be great. As you can see in my setup my device has two channels. In Tasmota Multichannel is implemented as Hue Bridge I think with different names (maybe configurable like in Tasmota via MAX_FRIENDLYNAMES). Maybe Wemo (single Device => Switches all channels on / off) and Multi-Device via different names. Maybe a Checkbox for the behavior in the Configuration menu!?

sromano1992, 2023-02-12:
Hi all,
I flashed last firmware (1.15.439) on my Aubess power monitor switch device because of my need to discover it via Alexa app. I was able to discover it as one single switch (see attached screenshot) and I can send on/off commands via Alexa app.

I configured openbk firmware to have 4 toggle (see attached image) and I would like to see every toggle as Alexa device, like I did with Tasmota firmware.

Is there a way to do that?

Thanks in advance, 
Simone
[image]
[image]

openshwprojects, 2023-02-22:
Hey @sromano1992 , are you the same person who asked on Elektroda?
@TheMasterFX , do you have time for more testing?

I want to add support for multi-relay integration and will need testers,

openshwprojects, 2023-02-22:
The thing is, I may need more than one test. I don't have Alexa. I will try to add multi switch support (Hue style) but I am not sure how to test that without Alexa. Or is there a mobile app for that as well?

or will Alexa app work without having a physical alexa device?

TheMasterFX, 2023-02-22:
> @TheMasterFX , do you have time for more testing?
> 
> I want to add support for multi-relay integration and will need testers,
I'm ready for testing anytime

Bamber1972, 2024-10-29:
Hi,

I have a Calex 4 way socket that I've flashed OBK onto but can't fully control with Alexa due to each of the 4 sockets being un-addressable.  Hue integration would be ideal to solve this issue, so I'm available to test and help out if needed?

Thanks,

Gary.

Bamber1972, 2024-11-11:
Hello again,

Is this sort of thing not a priority for this project?  I've only got one device with OBK installed on it, but it's one that really needs proper Hue integration due to there being 4 independant switched outlets that i can currently only control over the web interface.

Any information would be helpful!

Thanks,

Gary.