openshwprojects/OpenBK7231T_App #631: Interlock / Exclusive relay config

Condorello, 2023-01-22:
Hello, i flashed OpenBK on a 3 gang inwall relay which i use for control a fan. It's really basic, 3 gang and each of that control turn on a specific speed. Each relay must be exclusive, so if i turn on relay 1 (speed 1) it must be the only one turned on, otherwise the fan stays off. So it could be usefull to implement an auto-exclusive relay group inside wich  when one will became ON the others in the group must goes OFF automatically. I think tasmota already has this (Rules?).
Thanks

openshwprojects, 2023-01-22:
Wouldn't it just be enough to set one role to Rel and second to Rel_n ?

Condorello, 2023-01-22:
Thank Openshwproject for your quick reply. I tried but it doesn't turn off the first relay (speed1). It works as opposite than regular relay but not according to others relay. I think we need a rules section like tasmota :(

openshwprojects, 2023-01-22:
We have scripting system, even more advanced than Tasmota, but at the moment I am not sure what kind of behaviour you need.

If you set Rel_n and Rel to one channel, they always will work in opposite fashion, when one is on, second will be off.

Can you describe again what kind of behaviour you need?

Condorello, 2023-01-23:
I've a 3 channel in-wall relay, used for command a Fan which has 3 speed:
RELAY 1 ON -> LOW SPEED
RELAY 2 ON -> MEDIUM SPEED
RELAY 3 ON -> HIGH SPEED
ALL OFF -> OFF

It's need that just one Relay is ON at the same time, otherwise it does not work. So for medium speed ex:
RELAY 1 -> OFF
RELAY 2 -> ON
RELAY 3 -> OFF

so i need a rule that turn off all the other two relay when the one is ON (but keeping the possibility to set all OFF, otherwise the fan is always ON at some speed).
The very best is to set all as a FAN device, were a speed of 0 percentage correspond to OFF-OFF-OFF, a speed 1-33 to ON-OFF-OFF, a seed of 34-77 to OFF-ON-OFF and a speed between 78-100 to OFF-OFF-ON... but I don't know how to make a rule as this directly in Beken, currently i was using a function in HomeAssistant for control this behavoiur.

openshwprojects, 2023-01-23:
Sounds very easy to do? But if that's a common thing, then maybe a dedicated driver would be better?

Put in autoexec.bat:

addChangeHandler Channel0 == 1 backlog setChannel 1 0; setChannel 2 0
addChangeHandler Channel1 == 1 backlog setChannel 0 0; setChannel 2 0
addChangeHandler Channel1 == 2 backlog setChannel 0 0; setChannel 1 0

The speed thing... it also should be possible, I can try to script something for you as well for this, if you need this

openshwprojects, 2023-01-24:
ok regarding this:

> The very best is to set all as a FAN device, were a speed of 0 percentage correspond to OFF-OFF-OFF, a speed 1-33 to ON-OFF-OFF, a seed of 34-77 to OFF-ON-OFF and a speed between 78-100 to [...]

I've added a tiny command to do this, more coming soon, but range mapping is ready:
https://github.com/openshwprojects/OpenBK7231T_App/commit/a61405cfc9975d2652d92e51c177982f0f32db4e

Condorello, 2023-01-24:
Hi, thank for you kind support. The mutual exclusion [https://github.com/openshwprojects/OpenBK7231T_App/commit/03cb05c04f0a37d0f4fab1967dbf622aca34eae6](url) works perfectly.
In the weekend i'll try the range mapping, thanks!

openshwprojects, 2023-01-24:
By the way. we don't have any simple relay-based fan controllers on our list yet:
https://openbekeniot.github.io/webapp/devicesList.html
It's the first time I see such device. Please consider making a short teardown or just a setup article and posting to our list, you can even get some free gadget this way (if you want).

openshwprojects, 2023-01-31:
@Condorello , any news on this?

Condorello, 2023-01-31:
This weekend i was sick :(
How can i run the script? In cmd-line webgui?
Thanks again!

openshwprojects, 2023-01-31:
Create autoexec.bat in littlefs, copy script there, save, and either run or reboot

Condorello, 2023-06-24:
Hi, i totally Lost this until now that Summer come back...
How can i tru the selftest template? I tried to Copy the contenti of the script in autoexec.bat but any sliler appare. I moved the relays to ch 10-11-12 as template. Thanks forcyour help

Condorello, 2024-07-18:
> I also added an example how to map 0-100% value from HA to relays - basically a ready setup for you? [81407e0](https://github.com/openshwprojects/OpenBK7231T_App/commit/81407e038ba5113480e38868d1f2e2d7c38e9365)

FULLY WORKING THANKS!!!!