openshwprojects/OpenBK7231T_App #634: Sonoff IObroker testers needed! Basic compatilibity seems to work!

openshwprojects, 2023-01-24:
It seems to be working, but can you guys help with testing?

See video here:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20402248#20402248

Hahnweiler, 2023-01-24:
There appears only one object at the sonoff adapter:
[image]

I miss the object POWER to swich On and OFF.

openshwprojects, 2023-01-24:
@Hahnweiler  very good, **the power is not pushed to repository yet, check back in few hours**. I will do a release then.

christi-ko, 2023-01-24:
Ready for testing, awaiting your release ...

openshwprojects, 2023-01-24:
@christi-ko @Hahnweiler It really need testing. I created two brokers, one on my PC and that one works and second on VM and that VM one has issues. Not sure what's wrong.
Usage:
1. enable here:
[image]
2. set MQTT data for iobroker
3. full reboot! Do restart, please
4. it should appear:
[image]
5. both ways communication works
https://imgur.com/a/SSim6Ii

Hahnweiler, 2023-01-24:
Works great now. I get some warnings but i can confirm that the communication works in both ways.

[image]
[image]

openshwprojects, 2023-01-24:
@Hahnweiler  I know nothing about that broker, but I will try to fix warnings.

Btw, is this supposed to work for bulbs as well?

I just installed it today for you guys to fix the support so I don't know yet, I will check later today.

It turned out that only TELE was missing, 99% was done already.

christi-ko, 2023-01-24:
Hi, for me the transmission is working fine and most variables can be seen now. However 2 questions left.  

1.) However i am using a Power Plug which should send voltage and powerconsumption too, as i configured BL0937 also. But no message sent for those. Do i miss something in the confirguration to send them also, or is this still a bug?

2.) The shown load average is always 99 and the performance has become less responsive i think, but this might be a wrong impression.

openshwprojects, 2023-01-24:
LoadAvg is a placeholder. It's not calculated yet.

I will investigate Power measurement tomorrow.

Please report any further issues.

openshwprojects, 2023-01-25:
Power TELE added 
[image]

christi-ko, 2023-01-25:
Just tested the newest version. This includes all energy values required. Thank you very much for this great tool, now i can really use my powerplugs 👍

Hahnweiler, 2023-01-25:
I can confirm the warnings:

[image]

openshwprojects, 2023-01-25:
Do anyone know whats the source of these warnings? I could fix them if I had any idea what they mean...

Now I will also include DHT results in TELE message. Temperature and humidity.

@SteffenB71 , hey, you have T34 chip device? We don't have such one on our list. Can you consider posting a teardown?
https://openbekeniot.github.io/webapp/devicesList.html
We can even send you a free gift (a SD card, a DIY hacker card, etc) for submitting one: 
https://www.elektroda.com/rtvforum/topic3950844.html

Hahnweiler, 2023-01-25:
It seems as if there come too many of the same messages.
Every 2 minutes i get 12 entries and then one warning:

[image]

Hahnweiler, 2023-01-25:
Whe i restart the Sonoff Instance:
[image]

And after some minutes only the messages you can see at my last posting

Hahnweiler, 2023-01-25:
i dont know if this ist normal or if this has someting to do with the debug messages at the iobroker:
[image]

SteffenB71, 2023-01-25:
@Hahnweiler
How did you manage that the messages only come to you every 120 seconds, to me they come every few seconds.

Is there somewhere that can be set like with Tasmota - telemetry
period
@openshwproject
I'll have a look with the teardown. The T34 behaves exactly like the N. But I'll test the new flash method first. The T34 are installed in the BSEED series.
Steffen

openshwprojects, 2023-01-25:
I can explain - currently it's hardcoded that SENSOR comes every 3 seconds and TELE comes every 120 or so (and also when changing relay states ofc).
Those delays will be configurable soon

christi-ko, 2023-01-25:
I have the similar failure message, but my log says something different
`Info:MQTT:Publishing val (216 bytes) to tele/DVES_BK7231_N1/SENSOR retain=0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MQTT:Publishing val 236.338577 to DVES_BK7231_N1/voltage/get retain=0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Info:MQTT:Publishing val 0.007794 to DVES_BK7231_N1/current/get retain=0
Info:MQTT:Publishing val (216 bytes) to tele/DVES_BK7231_N1/SENSOR retain=0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0
Debug:MQTT:MQTT deduper sent 0, culled duplicates 0, culled too fast 0`

openshwprojects, 2023-01-25:
Those obk logs are 100% normal.

christi-ko, 2023-01-26:
I am not that sure, but it looks like the iobroker is sending a message back to the device and cannot get a connection. However, i attached a log of the iobroker set to debug status "silly" for the mqtt plugin. Maybe it helps:

<html>
<body>

sonoff.0 | 2023-01-26 19:34:42.316 | warn | Client [DVES_BK7231_N1] Message 59 deleted after 11 retries
-- | -- | -- | --
sonoff.0 | 2023-01-26 19:34:42.316 | warn | Client [DVES_BK7231_N1] Message 60 deleted after 11 retries
sonoff.0 | 2023-01-26 19:34:42.316 | debug | Client [DVES_BK7231_N1] Resend message topic: DVES_BK7231_N1/voltage/get, payload: 227.636642
sonoff.0 | 2023-01-26 19:34:42.316 | debug | Client [DVES_BK7231_N1] Resend message topic: tele/DVES_BK7231_N1/SENSOR, payload: {"Time":"1970-01-01T00:00:00","ENERGY":{"Power": 0.000000,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.000000,"Voltage":227.636642,"Current":0.000000,"ConsumptionTotal":270.451538,"ConsumptionLastHour":0.000000}}
sonoff.0 | 2023-01-26 19:34:42.316 | debug | Client [DVES_BK7231_N1] Resend message topic: tele/DVES_BK7231_N1/SENSOR, payload: {"Time":"1970-01-01T00:00:00","ENERGY":{"Power": 0.000000,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.000000,"Voltage":228.622085,"Current":0.000000,"ConsumptionTotal":270.451538,"ConsumptionLastHour":0.000000}}
sonoff.0 | 2023-01-26 19:34:42.316 | debug | Client [DVES_BK7231_N1] Resend message topic: DVES_BK7231_N1/current/get, payload: 0.007840
sonoff.0 | 2023-01-26 19:34:42.315 | debug | Client [DVES_BK7231_N1] Resend message topic: tele/DVES_BK7231_N1/SENSOR, payload: {"Time":"1970-01-01T00:00:00","ENERGY":{"Power": 0.000000,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.000000,"Voltage":227.513458,"Current":0.000000,"ConsumptionTotal":270.451538,"ConsumptionLastHour":0.000000}}
sonoff.0 | 2023-01-26 19:34:42.315 | debug | Client [DVES_BK7231_N1] Resend message topic: DVES_BK7231_N1/voltage/get, payload: 231.728683
sonoff.0 | 2023-01-26 19:34:42.315 | debug | Client [DVES_BK7231_N1] Resend message topic: tele/DVES_BK7231_N1/SENSOR, payload: {"Time":"1970-01-01T00:00:00","ENERGY":{"Power": 0.000000,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.000000,"Voltage":228.745269,"Current":0.007794,"ConsumptionTotal":270.451538,"ConsumptionLastHour":0.000000}}
sonoff.0 | 2023-01-26 19:34:42.315 | debug | Client [DVES_BK7231_N1] Resend message topic: DVES_BK7231_N1/current/get, payload: 0.000000
sonoff.0 | 2023-01-26 19:34:42.315 | debug | Client [DVES_BK7231_N1] Resend message topic: tele/DVES_BK7231_N1/SENSOR, payload: {"Time":"1970-01-01T00:00:00","ENERGY":{"Power": 0.000000,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.000000,"Voltage":228.129364,"Current":0.000000,"ConsumptionTotal":270.451538,"ConsumptionLastHour":0.000000}}
sonoff.0 | 2023-01-26 19:34:42.315 | debug | Client [DVES_BK7231_N1] Resend message topic: DVES_BK7231_N1/voltage/get, payload: 228.842071
sonoff.0 | 2023-01-26 19:34:42.314 | debug | Client [DVES_BK7231_N1] Resend message topic: tele/DVES_BK7231_N1/SENSOR, payload: {"Time":"1970-01-01T00:00:00","ENERGY":{"Power": 0.000000,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.000000,"Voltage":229.484344,"Current":0.000000,"ConsumptionTotal":270.451538,"ConsumptionLastHour":0.000000}}
sonoff.0 | 2023-01-26 19:34:42.314 | debug | Client [DVES_BK7231_N1] Resend message topic: tele/DVES_BK7231_N1/SENSOR, payload: {"Time":"1970-01-01T00:00:00","ENERGY":{"Power": 0.000000,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.000000,"Voltage":228.252548,"Current":0.000000,"ConsumptionTotal":270.451538,"ConsumptionLastHour":0.000000}}
sonoff.0 | 2023-01-26 19:34:42.313 | debug | Client [DVES_BK7231_N1] Resend message topic: DVES_BK7231_N1/voltage/get, payload: 229.331848
sonoff.0 | 2023-01-26 19:34:40.422 | debug | stateChange sonoff.0.DVES_BK7231_N1.ENERGY_ConsumptionLastHour: {"val":0,"ack":true,"ts":1674758080421,"q":0,"from":"system.adapter.sonoff.0","user":"system.user.admin","lc":1674664114742}
sonoff.0 | 2023-01-26 19:34:40.421 | debug | stateChange sonoff.0.DVES_BK7231_N1.ENERGY_ConsumptionTotal: {"val":270.451538,"ack":true,"ts":1674758080420,"q":0,"from":"system.adapter.sonoff.0","user":"system.user.admin","lc":1674757975536}
sonoff.0 | 2023-01-26 19:34:40.421 | debug | Set State after task: sonoff.0.DVES_BK7231_N1.ENERGY_ConsumptionLastHour
sonoff.0 | 2023-01-26 19:34:40.421 | debug | process task: {"type":"addObject","id":"sonoff.0.DVES_BK7231_N1.ENERGY_ConsumptionLastHour","data":{"_id":"sonoff.0.DVES_BK7231_N1.ENERGY_ConsumptionLastHour","common":{"role":"value","read":true,"write":false,"type":"number","name":"DVES_BK7231_N1 ENERGY ConsumptionLastHour"},"native":{},"type":"state"},"cbArg":0}
sonoff.0 | 2023-01-26 19:34:40.420 | debug | stateChange sonoff.0.DVES_BK7231_N1.ENERGY_Current: {"val":0,"ack":true,"ts":1674758080419,"q":0,"from":"system.adapter.sonoff.0","user":"system.user.admin","lc":1674758071416}
sonoff.0 | 2023-01-26 19:34:40.419 | debug | Set State after task: sonoff.0.DVES_BK7231_N1.ENERGY_ConsumptionTotal
sonoff.0 | 2023-01-26 19:34:40.419 | debug | process task: {"type":"addObject","id":"sonoff.0.DVES_BK7231_N1.ENERGY_ConsumptionTotal","data":{"_id":"sonoff.0.DVES_BK7231_N1.ENERGY_ConsumptionTotal","common":{"role":"value","read":true,"write":false,"type":"number","name":"DVES_BK7231_N1 ENERGY ConsumptionTotal"},"native":{},"type":"state"},"cbArg":270.451538}
[... cut ...]
sonoff.0 | 2023-01-26 19:34:40.410 | debug | updateState sonoff.0.DVES_BK7231_N1.ENERGY_Factor - 5
sonoff.0 | 2023-01-26 19:34:40.410 | debug | updateState sonoff.0.DVES_BK7231_N1.ENERGY_ReactivePower - 4
sonoff.0 | 2023-01-26 19:34:40.410 | debug | updateState sonoff.0.DVES_BK7231_N1.ENERGY_ApparentPower - 3
sonoff.0 | 2023-01-26 19:34:40.410 | debug | updateState sonoff.0.DVES_BK7231_N1.ENERGY_Power - 2
sonoff.0 | 2023-01-26 19:34:40.410 | debug | Set State after task: sonoff.0.DVES_BK7231_N1.Time

</body>
</html>

SteffenB71, 2023-01-31:
Hi
Are there any suggested solutions in the meantime, because I can't use the Sonoff integration like I can now.
Thanks Steffen

christi-ko, 2023-02-01:
Hi Stefan, you can reduce the log level for sonoff in the iobroker, not to show that messages. Anyhow it should work as intendet, at least it does for me ...

spayrosam, 2023-08-02:
Thanks for this great Job.
Can you please add Energy_Today and Energy_Yeseterday.

Here are the Iobroker tasmota data points.
[image]

coltc50, 2023-12-13:
Hello, I also had the warn messages in iobroker log... and the powerstatus is not send just in time when its switched.

What can I do to change this?

Technology89, 2023-12-31:
The warn messages still appear and changes on the openbeken device won't be shown in iobroker. Log shows the MQTT message was send but sonoff adapter on iobroker shows the message delete after XX retires messages. randomly it seems to update the states from time to time but couldn't find a pattern when it is doing it.

pafade89, 2024-01-18:
@openshwprojects

Hello, I'd like to continue helping here, as I've found the software to be really good so far. Currently, it seems there are issues with the communication for the Sonoff adapter (ioBroker). How can I assist to ensure the implementation works well?

PS: The same issue as described here.

[image]

Cheers P

TPPTim, 2024-02-03:
Hey,
i can confirm that problem.
The communication with the sonoff adapter works well! But i got also this warn messages.

Is there any solution or does somebody know why this messages appear?

[image]

openshwprojects, 2024-02-04:
I would love to help with that but I need more information. Can someone tell me specifically what should OBK include more in the MQTT publishes so this warning does not appear?

Related thread:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20944572#20944572

pafade89, 2024-02-04:
I'm sorry, I don't know if that helps, but are there any information on the GitHub page of the adapter that could be helpful?

https://github.com/ioBroker/ioBroker.sonoff

pafade89, 2024-05-30:
@openshwprojects Issue was fixed in Sonoff adapter of IoBroker. You can close this issue #634

              fixed in next version

_Originally posted by @Apollon77 in https://github.com/ioBroker/ioBroker.sonoff/issues/350#issuecomment-2131345063_
            
Next Version is [v3.1.0](https://github.com/ioBroker/ioBroker.sonoff/tree/v3.1.0)