openshwprojects/OpenBK7231T_App #635: WiFi connection fails with  a longer access point password

josephperry, 2023-01-26:
**Describe the bug**
I have two WiFi access points.  My BK7231N can connect to the access point with a 9 character password however there is a problem with the access point which has a 26 character password.  The WiFi led on the BK7231N will stop blinking as though it has connected but the BK7231N doesn't appear on the network.

**Firmware:**
- Version; 1.15.384
- Device: Mini Universal IR Remote model S18
- Chip/model: BK7231N
- Device config: WifiLED on the BK7231's LED

**To Reproduce**
Steps to reproduce the behavior:

1. Configure the access point with a 9 character WiFi password
2. Configure BK7231N to connect with the password
3. Reboot the BK7231N 
4. Connection is successful and the BK7231N is available on the network and the WiFI LED stops blinking
5. Configure the same access point with a 26 character WiFi password but no other changes
6. Configure BK7231N to connect with that new password
7. Reboot the BK7231N 
8. Device WiFi LED on the BK7231N stops blinking but it never shows up on the network

openshwprojects, 2023-01-26:
Hello, can you please try some older version and tell me if it happens there as well?
But in general, that's strange, my WiFi pass is 24 letters and seem to work.... but I will check a longer one, just need to have some time when family doesn't need to access router. They don't like being cut off from the internet.
Futhermore... we had user with 64 characters password and he said that it was tested and works, look here:
https://github.com/openshwprojects/OpenBK7231N/pull/13

Another strange thing...
" The WiFi led on the BK7231N will stop blinking as though it has connected but the BK7231N doesn't appear on the network."
this would mean that device has connected so password worked but dhcp failed?

Can you get an UART log from TXD2?

josephperry, 2023-01-26:
I'll try to get the UART log from TXD2.

Don't know if this helps, but the 9 character password that works is all letters, the 26 character password that fails is all digits.

openshwprojects, 2023-01-26:
I never say things with full confidence but it seems almost like a fluke. Password content should have NOTHING to do with your device connection.

josephperry, 2023-01-26:
Installed version 1.15.391 and WiFi now seems okay with the 26 digit password.  I can't explain it.

Tried to get the UART log by connecting to TX2 but got nothing.  Also disabled IRSend on P7 (TX2) and IRRecv on P8 (it's a CBU module) but still no output to PuTTY.  Tried connecting at 921000 baud and a few other rates but nothing worked.  What else should I have tried?  

I wasn't aware of your great project until a few days ago!  I intended to swap the Tuya module for an ESP8266 module until I found my device had a CBU module.

closed 2023-01-26 by josephperry