openshwprojects/OpenBK7231T_App #638: Rename project, as it has grown out of Beken and now a fully fledged multiplatform app, supporting many IoT platforms?

openshwprojects, 2023-01-29:
It was brought to my attention that our project name is obsolete and may make some people confused or unhappy, as Beken is just a BK7231T and BK7231N manufacturer.

We are now currently supporting BK7231T and N, but also W600/W601, W800/W801, BL602 (possible 604 and 700-s as well), and XR809, and of course also Windows (with simulator).

I think we should consider getting a more "multiplatform" new, that does not contain the manufacturer name of a certain platform, to make everything clear and keep certain people happy.

So my question is - do you have any ideas for a possible future name for out multiplatform IoT firmware?

DiegoJp, 2023-02-02:
Since they are all based on Tuya, how about OpenTuyaBreak? (added break as a wow factor lol)

Wolfieeewolf, 2023-02-02:
I ran you question through ChatGPT a few times and it spat out these 

Sure, here are a few suggestions for a new name for your multi-platform IoT firmware:

    IoT Hub
    CrossConnect
    OmniCore
    OneOS
    SmartBridge
    UniFirm
    NexusNet
    FusionCore
    FlexiIoT
    GlobalNet

    IoT Fusion
    PlatformHub
    Universal IoT
    InterConnect
    IoT Nexus
    The Connector
    IoT Synergy
    MultiIoT
    All-in-One IoT

    ConnectCore
    OneIoT
    AllNodes
    IoT Hub
    SmartNet
    MultiMesh
    EdgeFuse
    PlatformPlus

These names aim to convey the idea of a unified, all-encompassing, and flexible platform that can connect and support multiple IoT devices

I like Crossconnect or maybe some kind of mashup of something else on the list.

openshwprojects, 2023-02-03:
We can't obviously use Tuya trademark, that would be a change for worse. Futhermore, W600, W800, etc are non-Tuya.
BL602 as well. 
I will consult that our internal Discord as well.

closed 2023-02-03 by openshwprojects

kuba2k2, 2023-02-03:
Internal Discord? What a shame it's not public. I remember you saying you wanted to keep everything public, and indexable by search engines, for example on the Elektroda forum. I wonder why did you change your mind.