openshwprojects/OpenBK7231T_App #639: NTP not working

Klaas61, 2023-01-29:
The NTP server is not contacted. All ntp related commands are not working. time is incorrect

- Version 1.15.398 (also lower firmware versions)
- Chip/model: BK7231N
- Device config Standard after flash

Flashed device. After connected to home network and configure mqtt in mqtt exploerer the following can be seen 
{"Time":"1970-01-01T00:00:00","Uptime":"30T02:59:30","UptimeSec":11386,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":23,"POWER":"OFF","Wifi":{"AP":1,"SSId":"*******","BSSId":"************","Channel":11,"Mode":"11n","RSSI":78,"Signal":-55,"LinkCount":21,"Downtime":"0T06:13:34"}}

The time\data is not correct and the uptime is in this case less then a day
command ntp_info returns an error.

openshwprojects, 2023-01-29:
Hello, have you started driver?

Klaas61, 2023-01-29:
That was not the case.
Now I get normal responses.
Thanks for pointing me the right direction.

closed 2023-02-13 by openshwprojects