openshwprojects/OpenBK7231T_App #640: Question: OpenBK7231T vs Localtuya

chrispazz, 2023-01-30:
Hello,
i am using a bounch of GX53 WIfi lamps connected to HA using Localtuya.

WIll I get any benefit in speed and connection reliability using this firmware?

Thank you
Chris

wu7zh, 2023-01-30:
From my experience Localtuya and OpenBeken via MQTT are comparable in speed, but for me OpenBeken never had a connection issue, while Localtuya had some connection issues (same device).

The real benefit here is disconnecting yourself from Tuya. While localtuya works locally, when at some point you reconnect the device (new router or whatever) and you need a new localkey, you need Tuyas API, and seeing how they restrict tuya-convert etc. I wouldn't bet on this being possible forever.

With OpenBeken you own the device and it will be usable forever.

chrispazz, 2023-01-30:
I agree with your answer....thank you.
Now I have to understand how to check if my device is compatible. It is not in the repository. It is a GY GX53 sold on Amazon....
From SmartLife app I can see:

Main Module: V1.3.21
MCU: V.1.3.21

so I think it has a BK7231xx chipset...

ferbulous, 2023-01-31:
> From my experience Localtuya and OpenBeken via MQTT are comparable in speed, but for me OpenBeken never had a connection issue, while Localtuya had some connection issues (same device).

Agreed, had same issues with Localtuya integration
Another thing to note is getting rid of the blinking lights if they accidently get reset because eventually it will happen with the stock firmware

chrispazz, 2023-02-01:
Currently having same issues and I hate when lamps start blinking...:)