openshwprojects/OpenBK7231T_App #641: Unable to send Panasonic IR codes from a BK7231N

josephperry, 2023-01-30:
**Describe the bug**
Unable to send Panasonic IR codes from a BK7231N

**Firmware:**
- Version: 1.15.398
- Device: MiniUniversal IR Remote model S18 
- Chip/model: BK7231N
- Device config: IRSend P7 on IRRecv on P8

I have a BK7231N, it's a MiniUniversal IR Remote model S18 with version 1.15.398

Log when transmitting IR from a Panasonic TV Remote:
Info:MQTT:Publishing val IR_Panasonic 0x8 0x20 0 to ir_blaster_1/ir/get retain=0
Info:IR:IR MQTT publish IR_Panasonic 0x8 0x20 0 took 5ms
Info:MQTT:Publishing val {"IrReceived":{"Protocol":"Panasonic","Bits":48,"Data":"0xA0200080"}} to ir_blaster_1/RESULT retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic ir_blaster_1/ir/get
Info:MQTT:MQTT in topic ir_blaster_1/ir/get
Info:MQTT:MQTT topic not handled: ir_blaster_1/ir/get
Info:MQTT:channelSet part topic ir/get
Info:MQTT:channelSet channel 0
Info:MQTT:channelSet NOT 'set'

Log when sending the Command "**IRSend Panasonic 0x8 0x20 0**":
Info:IR:Delay 100ms
Info:IR:IR send Panasonic 0x8 0x20  protocol 10 addr 0x8 cmd 0x20 repeats 0

but my phone camera **does not detect IR output from the BK7231N**

However the Log when sending the Command "**IRSend NEC 0x1000 0x0 1**":
Info:IR:Delay 110ms
Info:IR:Delay 100ms
Info:IR:IR send NEC 0x1000 0x0 1 protocol 7 addr 0x1000 cmd 0x0 repeats 1

and my phone camera **does detect IR output from the BK7231N**

It looks like the Panasonic IR signals are correctly detected but the Panasonic IR codes can't be sent even though the NEC codes can be sent.

I don't see the relationshop from the IrReceived JSON output to the required IRSend command.  At some point it would be helpful if the BK7231N would accept IR commands from MQTT eg:
Topic: **cmnd/ir_blaster_1/IRSend**
Payload: **{"Protocol":"Panasonic","Bits":48,"Data":"0xA0200080"}**

openshwprojects, 2023-01-31:
Hello, json format can be added as soon as tomorrow, no problems here. I can can do that quickly as long as our it lib supports describing send in that format.
Regarding sending issue... well, it would be better if @btsimonh looked into it, he has more IR experience.

avdmr, 2023-01-31:
Seems like the same issue I reported. The behavior is the same for Panasonic and Sony codes. 
https://github.com/openshwprojects/OpenBK7231T_App/issues/442

I would be great if btsimonh could have a look at this.

josephperry, 2023-01-31:
Saw the previous issue with Panasonic and Sony codes listed back in November.  Looks like @btsimonh  hadn't gotten a chance to look into it.  I'll help with this in any way I can.

I tried earlier to get the UART log by connecting to TX2 but saw no output.

Also disabled IRSend on P7 (TX2) and IRRecv on P8 (it's a Tuya CBU module) but still had no output to PuTTY. Tried connecting at 921000 baud and a few other rates but nothing worked.  Don't know if the problem is because TX2 is also connected to the IR transmit hardware. 

Is there anything else should I have tried to get the UART log from TX2 or is the log I posted sufficient?

openshwprojects, 2023-01-31:
@josephperry the log should be always available at TX2, how did you manage to get it not working? it sends debug log there even before loading obk firmware, take a look, here is bootloader printing uart data:
[image]

josephperry, 2023-01-31:
@openshwprojects thanks for your quick reply.  I'll try again with the UART settings you've shown.  It's difficult to solder a lead to TX2 on my device due to tight spacing with one of the IR LEDs.  It may have had been a bad solder joint.  

I'll report back after I attempt again.

josephperry, 2023-01-31:
@openshwprojects  I don't like to admit it, but I had connected to P15 (TX1) rather than P7 (TX2) on the CBU module.  :unamused:

Connecting to TX2 at 115200 baud does display the UART debug log in PuTTY as expected.  Didn't mean to complain about a non-problem.

So if the UART debug log is needed to help track down the problem with transmitting Panasonic IR codes I can now produce it.

ferbulous, 2023-01-31:
> Seems like the same issue I reported. The behavior is the same for Panasonic and Sony codes.

Having the same problem as well with Sony codes

EvgeniyBlinov, 2023-12-02:
+1
OpenBK7231N_QIO_1.17.308.bin
Tuya s18 IR Blaster

artin961, 2024-02-25:
Hello @openshwprojects,

I see here is discussed parsing json payloads for irsend. Ho you have further plans to implement it?
I'm interested in that so my tasmota and beken devices can use the same smartir codes or HA scripts. Json style commands are much easyer to configure just by copying from the log. 

I have some spare IR-blasters JM_YK01_BK that can use for testing if user feedback is required. 
Best regards,
artin961