openshwprojects/OpenBK7231T_App #642: Not issues. Request Tuya WBR2 W701

AdrianGomez2023, 2023-01-31:
Hi, first of all thanks for this great project and sorry for my english, i'm from Argentina.

I don't know if this is the right place to ask, but i cannot find another place to do.
Do you plan add compatibility with Tuya WBR2 chip? i attach a photo of chip.

You can find this in most of smart switch and i have a lot of them.
Thank for all,
Best regards.
[image]

openshwprojects, 2023-01-31:
Hey, is there SDK available somewhere?

AdrianGomez2023, 2023-01-31:
> 

Hi, thank for answer. After several tries to download the SDK from Tuya website, i cannot find a way to download. Then i don't know where i can get the SDK =(

Blackspell01, 2023-07-27:
Are there any news yet? :)

Blackspell01, 2023-11-01:
@AdrianGomez2023 are you planning on adding support for this chip any soon?

lulol, 2023-12-30:
Mine is in a WiFi-BT Tuya cloud gateway and uses a [CR3L](https://fcc.report/FCC-ID/2ANDL-CR3L/5058202.pdf) [module](https://web.archive.org/web/20230518090314/https://developer.tuya.com/en/docs/iot/cr3l-module-datasheet?id=Ka3gl6ria8f1t) with W701M chip [RTL8720CM Ameba Zii](https://www.realtek.com/en/products/communications-network-ics/item/rtl8720cm).

WBR2 is W701-VA2-CG
CR3L is W701M-VT2-CG

[Realtek provides](https://www.amebaiot.com/en/ameba-sdk-getting-started-8720/) an [SDK](https://github.com/ambiot/ambz2_sdk) but I'm not sure it would be practical to include this family chip in this project.
The SDK is based on IAR Embedded Workbench. [Realtek Ameba ZII DEV User Manual](https://files.seeedstudio.com/wiki/Ameba/Realtek_Ameba-ZII_DEV_User_Manual_1v0_191021.pdf)

[image]
[image]

bootlog (◼️=censored ids, keys and mac)
<details>
<pre>
$ picocom -b 115200 /dev/ttyUSB0
picocom v3.1
 
port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no
 
Type [C-a] [C-h] to see available commands
Terminal ready
 
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0
 
== Boot Loader ==
Nov 12 2020:10:54:11
 
fwx SELE[fffffffc]
fw SELE Bitidx 2, fw1 valid 1, sn 0, fw2 valid 1, sn 1
fw1 USE, return sn 0
 
Boot Loader <==
 
== RAM Start ==
Build @ 19:49:10, Mar 11 2022
[MEM DBG] heap init-------size:178272 addr:->100131a0---------
[MEM DBG] heap init-------size:4194304 addr:->60000000---------
interface 0 is initialized
interface 1 is initialized
 
Initializing [01-01 18:1WIFI ...2:15 TUYA I][mqc_app.c:350] mqc app init ...
[01-01 18:12:15 TUYA I][uni_thread.c:220] thread_create name:sys_timer,stackDepth:4096,totalstackDepth:4096,priority:5
[01-01 18:12:15 TUYA I][uni_thread.c:220] thread_create name:cmmod,stackDepth:4096,totalstackDepth:8192,priority:4
[01-01 18:12:15 TUYA D][mqc_app.c:144] mq_pro:5 cnt:1
[01-01 18:12:15 TUYA D][mqc_app.c:144] mq_pro:31 cnt:2
[01-01 18:12:15 TUYA D][svc_online_log.c:295] svc online log init success
[01-01 18:12:15 TUYA D][log_seq.c:894] read m:1 f:0 l:0
[01-01 18:12:15 TUYA I][uni_thread.c:220] thread_create name:wk_th-0,stackDepth:5120,totalstackDepth:13312,priority:3
[... 223 lines cut ...]
[01-01 18:14:00 TUYA E][tuya_gw_mesh_mgmt.c:129] tuya_iot_sigmesh_get_net_info op_ret:-1
[01-01 18:14:00 TUYA E][tuya_gw_mesh_mgmt.c:630] get net info err:-1
[01-01 18:14:01 TUYA N][tuya_device.c:615] __event_ctl_reboot
tuya_hal_system_reset
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0
 
== Boot Loader ==
Nov 12 2020:10:54:11
 
fwx SELE[fffffffc]
fw SELE Bitidx 2, fw1 valid 1, sn 0, fw2 valid 1, sn 1
fw1 USE, return sn 0
 
Boot Loader <==
 
== RAM Start ==
Build @ 19:49:10, Mar 11 2022
[MEM DBG] heap init-------size:178272 addr:->100131a0---------
[MEM DBG] heap init-------size:4194304 addr:->60000000---------
interface 0 is initialized
interface 1 is initialized
</pre>
</details>

Chasbrot, 2024-01-20:
Hi!
I recently bought a smart water relay for irrigation from amazon. https://www.amazon.de/dp/B0BRK97X51

It also has a W701 chip inside, on a CR3L module. It connects to WiFi and communicates via bluetooth to the outdoor relay.

[image]

If this chip would be supported, we could add support for this device and probably others. 

Many thanks.

divadiow, 2025-01-28:
RTL87X0C support is now available: https://www.elektroda.com/rtvforum/topic4097185.html

also. please send me any RTL87xx firmware backups you may have made in the past or future :)