openshwprojects/OpenBK7231T_App #647: Build on linux (arm-none-eabi-gcc)

evlo, 2023-02-02:
How to build on linux? As tuya-cloudcutter supports primarily ubuntu 20 (22 does have issue with dns and needs modifying one file to run)

I'm following this https://github.com/openshwprojects/OpenBK7231T_App/wiki/Building guide

```
mkdir -p Debug/bin
build beken378/app/app_bk.o
echo prerequirement2
prerequirement2
/bin/sh: 1: ../toolchain/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: not found
make[2]: *** [[application.mk:1176](http://application.mk:1176/): beken378/app/app_bk.o] Error 127
make[2]: *** Waiting for unfinished jobs....
```

For anyone wanting to build on linux too, here is changes I did so far:

- changed T to N in b.sh `./build_app.sh apps/OpenBK7231T_App OpenBK7231T_App 1.0.0`
- sudo apt install gcc-arm-none-eabi
- sudo apt install make
- backslash to forward slash `slash git clone https://github.com/openshwprojects/OpenBK7231T_App apps**/**OpenBK7231T_App`

openshwprojects, 2023-02-03:
Isn't it here? 
https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q1-update

valeklubomir, 2023-02-04:
I am running building on Debian linux, works fine. 
Tools were downloaded with SDK for chipset. OpenBL7231N/T. 
And it was pretty straight forward. 
Example for BK7231N toolchain should be installed in 'platforms/bk7231N/toolchain/gcc-arm.......'
Path is mentioned : 
/bin/sh: 1: ../toolchain/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: not found
Download:
wget https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q1-update/+download/gcc-arm-none-eabi-4_9-2015q1-20150306-linux.tar.bz2

Extracted contents of tar.bz2 to 'platforms/bk7231n/toolchain/'

And each time build failed I googled some solution until build was successful.

Dheenhasty, 2023-02-07:
Same here, i'll use a Xubuntu ... so debian based distro for compiling .... no issue.

evlo, 2023-02-08:
I used ubuntu as it is mentioned in guide, issue.

openshwprojects, 2023-02-10:
Path to toolchain is wrong? Please check it, directory by directory.

Xinayder, 2023-02-18:
I use Arch Linux and I could build it just fine. All I did was fetch the submodules `git submodule init --recursive`, then I executed the same `make` task present in the GitHub Actions workflow file:

`make APP_VERSION=${{ needs.refs.outputs.version }} APP_NAME=${{ matrix.platform }} ${{ matrix.platform }}`

replacing the placeholder variables with the MCU I was compiling for (in this case BK7231N).

As far as I'm aware you don't need a custom version of gcc. 

What system are you trying to compile on (distro, device)?