openshwprojects/OpenBK7231T_App #648: accidental relay turns off

Stalker-4x4, 2023-02-02:
**Describe the bug**
Sometimes Digma DiPlug 200S turns off (without any mqt command) - about twice a day. But it only turns off under load, I didn't get any turns off without load. With the original Tuya firmware with the same load, I have not encountered turns off

**Firmware:**
- Version 1.15.405
- Device: Digma DiPlug 200S (looks like as clone of Aubess Smart Socket EU 16A)
- Chip/model: BK7231T
- Device config: PWM0 - LED_n, PWM1 - BL0937CF, PWM2 - BL0937CF1, RXD1 - Btn, PWM4 - BL0937SEL, PWM5 - Rel

openshwprojects, 2023-02-02:
Does the device reboots as well?
Maybe it has a low quality power supply and it resets, do you have "PowerSave" in short startup command or autoexec.bat? 
What's in the log?

Stalker-4x4, 2023-02-02:
The device did not reboots and there is nothing special in the logs, or I failed to notice the power-off moment in the logs.
I tried to write to a file all the mqtt messages from the device topics, but there were no commands to turn off the relay. 
Now I tried to enable the use of UART2 in the settings, because the button is connected to RXD1 - maybe something is trying to read from UART1 and causes a false button trigger. 
If this not help then I'll try to disable MQTT (change MQTT server to the invalid one, because I can't configure empty server address - it resets back to the configured before)

Stalker-4x4, 2023-02-06:
I tried with changed settings for UART2 and also with disabled MQTT (change the address of a mqtt server to the non-existent one) - no luck. Relay switch's off in a few hours (but without load in works longer until power-off). Maybe can be related with power metering

openshwprojects, 2023-02-10:
"Now I tried to enable the use of UART2 in the settings, because the button is connected to RXD1 - maybe something is trying to read from UART1 and causes a false button trigger." it won't change anything because UART1 is not active by default.

Please disable button role (set to none) or set button channel to other channel and observe if your relay still goes off.

Stalker-4x4, 2023-02-12:
Thanks. It seems to have helped. In any case, it would be nice to get support for the button if it is connected to the TX pin.