openshwprojects/OpenBK7231T_App #652: Exponential brightness "feature"

662KDV, 2023-02-05:
**Describe the bug**
Non programmer here. Using LSC Smart Ceiling light (RGBCW, T7231N), unable to disable "exponential brightness" feature.

Controls in web interface and HA are not accurate.

To achieve 10% warm/cold light, the slider must go up to 35%

**Firmware:**
- Version  All above 1.15.301 
- Device - LSC Smart Ceiling Light (RGBCW)
- Chip/model: BK7231N
- Device config - default config. Using autodiscovery.

**To Reproduce**
Steps to reproduce the behavior:

Controls in web interface and HA are not accurate.

To achieve 10% warm/cold light, the slider must go up to 35%

**Screenshots**
If applicable, add screenshots to help explain your problem.

 
[image]

**Additional context**
Add any other context about the problem here.

openshwprojects, 2023-02-05:
Hello, I can add an option to disable it, but you are the first person requesting it. Are you sure that you'd get better results without it? I've had reports that without it it's not possible to get down to very dark brightness levels and it improves overall brightness control quality.

662KDV, 2023-02-05:
Hi openshwprojects,

With old firmware up .1xx the ceiling light I'm using worked without issues. I have automation in HA to turn on lights to 10% in the morning. Newer firmware up to .301 I could use the command "led_expoMode 0" and the behaviour of the lights was the same. (brightness slider worked without issues.

With firmware .302 and newer I can't use "led_expoMode 0 " any more. From what I understood it got replaced by "brtMinCW"

I tried using the last command without success. The console would report "unknown command" so for the moment I'm stuck with that "unintuitive slider"

The effect is that at 10% slider...the led's go down to 1% or close and you can barely see the light. 

One workaround would be to redo the automation in HA and instead of 10% light, ask for 35% (to get the old 10%)

I might be doing something wrong. As I said I'm not a programmer so I'm using default generated config.
(big THANK YOU for this feature, as looking for good/working configs was a pain for a noob like me)

There might be others with similar LSC devices who would need this "feature to the feature" :)

I have another LSC CW bulb, with BK7231T, I did not test the firmware on that device yet, but it might behave the same.

So please, if possible add some possibility to disable exponential brightness for both.

openshwprojects, 2023-02-05:
Well, ok, if you think so... I added flag 32 for you, it disables gamma ramp totally, can you check if it works?

662KDV, 2023-02-05:
I just flashed .412

The light stays at 100% no matter where I move the slider.

Just to make sure I explained correctly, look at the picture attached in first post.

With version .412 the value of Channel 5 stays at 100% all the time.

662KDV, 2023-02-05:
This is the config I copied from HA.

device:
  identifiers:
    - LSC_Aplica_1
  name: Aplica_1
  sw_version: 1.15.412
  manufacturer: Beken Corporation
  model: BK7231N
  configuration_url: http://x.x.x.x/index
name: Aplica_1
availability_topic: Aplica_1/connected
payload_on: '1'
payload_off: '0'
unique_id: LSC_Aplica_1_light
qos: 1
rgb_command_template: '{{''#%02x%02x%02x0000''|format(red,green,blue)}}'
rgb_value_template: >-
  {{ value[0:2]|int(base=16) }},{{ value[2:4]|int(base=16) }},{{
  value[4:6]|int(base=16) }}
rgb_state_topic: Aplica_1/led_basecolor_rgb/get
rgb_command_topic: cmnd/Aplica_1/led_basecolor_rgb
color_temp_command_topic: cmnd/Aplica_1/led_temperature
color_temp_state_topic: Aplica_1/led_temperature/get
state_topic: Aplica_1/led_enableAll/get
command_topic: cmnd/Aplica_1/led_enableAll
brightness_state_topic: Aplica_1/led_dimmer/get
brightness_command_topic: cmnd/Aplica_1/led_dimmer
brightness_scale: 100
platform: mqtt

openshwprojects, 2023-02-05:
Oops, I forget to multiply by brightness. Can you check now?

662KDV, 2023-02-05:
Don't see new version.

rogere66, 2023-02-06:
With the new flag LED Gamma and RGB Calibration can be disabled with a flag, but it can also be done using the led_gammaCtrl command. Correct command sequence to disable is (command + sub-command + value(s)):

led_gammaCtrl gamma 1.0
led_gammaCtrl brtMinRGB 0.0
led_gammaCtrl brtMinCW 0.0
led_gammaCtrl cal 1.0 1.0 1.0

For full details check the doc pages at: https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/led_gamma_control.md

662KDV, 2023-09-24:
Update to the reported bug.

It seems that after a (accidental or not) power off, the ceiling light will not work. I have to disable the added flag - save - reboot light - enable the flag - save.

I'm not sure if this was happening from the beginning as I didn't think it was necessary to also test this scenario.

Thank you,