openshwprojects/OpenBK7231T_App #654: can not upgrade to newer version

thinhvuq, 2023-02-07:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version: OpenW600_1.15.375
- Device: 3 Gang Wall Switch
- Chip/model: W600 - TW-03
- Device config?

Both OTA (update sofware by Wifi) and OTA (Web Application) can not upgrade to any newer version from version OpenW600_1.15.375.
Regards,
Thinh Vu

iprak, 2023-02-07:
It seems that builds 1.15.376 onwards give checksum error. 1.15.375 seems to be last good build which passes checksum.

```
magic_no=2701131679, img_type=0, zip_type=1
Error:OTA:Image header check failed
```

It seems that the image from 376 onwards overlaps with the run area. Perhaps due to new features being added to the image.

openshwprojects, 2023-02-07:
So we need to reduce the size of built binary for W600?

iprak, 2023-02-09:
I checked the make file to dump sections from the intermediate .out file using `arm-none-eabi-size` and that gave

	GNU-SIZE .output/w600/image/w600.out
	   text    data     bss     dec     hex filename
	 516884   10248  122220  649352   9e888 .output/w600/image/w600.out

c1cbcb6cabacc5f0eac8f8e3374d08a0d86e4857 (the last good version 1.15.375):

	GNU-SIZE .output/w600/image/w600.out
	   text    data     bss     dec     hex filename
	 513804   10240  122204  646248   9dc68 .output/w600/image/w600.out

After commenting out json_interface.c I got this and OTA was again operational.

	GNU-SIZE .output/w600/image/w600.out
	   text    data     bss     dec     hex filename
	 501044   10232  121948  633224   9a988 .output/w600/image/w600.out

So, yes, it seems that new code has been slowly pushing W600 image to its limit. I don't know if this was it or something else. In long term, maybe the build features should be opt-in.

iprak, 2023-02-09:
So then I commented out the content in `http_fn_ha_cfg` and the helper functions which gave this along with successful OTA. This was with the latest code.

	GNU-SIZE .output/w600/image/w600.out
	   text    data     bss     dec     hex filename
	 500740   10240  121964  632944   9a870 .output/w600/image/w600.out

openshwprojects, 2023-02-09:
What is taking so much memory? Can we disable it for W600?

I am also planning to shorten the strings and use a single function to check arguments count for commands.

iprak, 2023-02-09:
I don't know, this is just trial an error.

I think the help texts, flag settings, labels all might be contributing.

I some more experimentation to move this string in `http_fns.cs` into a static and use it at 8 places.

    static char SUBMIT_AND_END_FORM[] = "<br><input type=\"submit\" value=\"Submit\"></form>";

This dropped the text section by 192 

from
	GNU-SIZE .output/w600/image/w600.out
	   text    data     bss     dec     hex filename
	 517036   10248  122220  649504   9e920 .output/w600/image/w600.out
	 
to

	GNU-SIZE .output/w600/image/w600.out
	   text    data     bss     dec     hex filename
	 516844   10296  122220  649360   9e890 .output/w600/image/w600.out

openshwprojects, 2023-02-09:
@iprak  if you have any ideas like that for improvement, please submit.

I will today make the arguments count checking for console command consistent and into a single function so the strings like "command xyz takes 5 arguments" do not use up the memoryu

openshwprojects, 2023-02-09:
Also merged I2C codes. More coming soon.

iprak, 2023-02-10:
I can confirm that this is mostly due to the new file `json_interface.c`. I pushed in a PR to help reduce the image size a bit though.

iprak, 2023-02-10:
But this separate PR to W600 SDK should reduce the overall image for now - https://github.com/openshwprojects/OpenW600/pull/5

openshwprojects, 2023-02-10:
Very good job @iprak, I am also doing some generic optimizations now

iprak, 2023-02-11:
@thinhvuq Please try out the latest build. I was able to install it successfully.

thinhvuq, 2023-02-11:
Thank you.
 It's updated with the latest build OpenW600_1.15.435_gz.

openshwprojects, 2023-02-11:
@iprak  would you be able to add some kind of alert/forced error to the build process of W600 so we know when the OTA binary is too large?

iprak, 2023-02-11:
@openshwprojects Double checking if the build using latest W600 SDK? I think the SDK was updated buy maybe the linked submodule was not. If that the case then our net src changes did reduce the image enough to allow OTA but SDK changes should reduce it more.

iprak, 2023-02-11:
> @iprak would you be able to add some kind of alert/forced error to the build process of W600 so we know when the OTA binary is too large?

Yes that should be possible but maybe that is not needed. I pushed some more changes to W600 which take out a couple more unused libraries.