openshwprojects/OpenBK7231T_App #657: IR - Single Remote not working

glichten1, 2023-02-09:
**Describe the bug**
IR remote with device ID "IR_NEC 0xEF00" not working. Every other IR remote control with same configuration (for Example IR_NEC 0x2) is working.

**Firmware:**
- Version 1.15.417
- Device casalux smart wlan led strip
- Chip/model: BK7231N
- Device config?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.
Autoexec.bat:
[image]
Logs
[image]

**Additional context**
Add any other context about the problem here.

openshwprojects, 2023-02-09:
Hello. It looks like it might be caused by our wrong assumption that arguments are signed 16 bit integers. 
I've added a fix:
https://github.com/openshwprojects/OpenBK7231T_App/commit/c4e179229c2db26223099ac7b4729b21a912bcc5
Can you check if it works for you with this commit - get latest update when it's built.

glichten1, 2023-02-10:
That was really quick.
It is working!
Thx

closed 2023-02-13 by openshwprojects