openshwprojects/OpenBK7231T_App #658: Anyone with LED drivers like SM* or BP* or I2C sensors?

openshwprojects, 2023-02-10:
I have merged I2C codes because they were just copied in a nasty, non-optimal way.
I have SM2135 and BP* bulbs, and I will test new system well, but it would be still useful to know that there is someone else out there ready to test today or tomorrow and make sure that everything is correct.

So, anyone here with SM* or BP* I2C drivers? I will need testing of next release, today or tomorrow.

zain1144, 2023-02-10:
I'm here if you want any testing. I have both SM2135 and BP5758 bulbs.

openshwprojects, 2023-02-10:
@zain1144  I will push final commands clearup in 15 minutes, can you check then if those bulbs still work?

zain1144, 2023-02-10:
Yes, I can

zain1144, 2023-02-10:
I updated both bulbs, I didn't notice anything diffrent

openshwprojects, 2023-02-10:
That's good news!

ByteMangler, 2023-02-15:
great work. May I suggest you spin the soft i2c code out into its own file?  Makes the cht8305 driver look skinny! The delay in drv_cht8305.h is now superfluous. Same in the sht3x driver as well so these files can be deleted.

There may be an opportunity to combine the cht8305 and sht3x drivers as the sht driver implements more functionality that could benefit the cht8305 - calibration and heater control. If I get a free moment I might have a look at this.

openshwprojects, 2023-02-18:
@ByteMangler  it is obviously the plan, it should be in separate file, it's just I am super super busy because now I also started doing calendar system and I am also making YT videos with my assistant:
https://www.youtube.com/watch?v=L6d42IMGhHw