openshwprojects/OpenBK7231T_App #660: Help configuring MI-BW932-999W

kogeki, 2023-02-10:
Hello,

I have a few of these MI-BW932-999W lights. They are RGBW. I am able to individually control the colors via the web interface, however when I use the RGB color picker, it all breaks.

Here is what I was able to figure out:
```
pwm0 - red
pwm1 - ???
pwm2 - white
pwm3 - ???
pwm4 - blue 
pwm5 - green
```

What is the correct configuration for this? I am new to this, so I am not even sure if I am doing it correctly. Thanks!

openshwprojects, 2023-02-11:
Hello, so you have all 4 PWMs figured out correctly, right?
What exactly breaks, what happens?

kogeki, 2023-02-11:
Sorry for not being more clear. 

Basically, I can't get the RGB color picker to map to the correct colors on the bulb. So, when I choose a color with the color picker, it will change a different value.

openshwprojects, 2023-02-11:
No, it's RGBCW, or RGBW, white is after RGB.

Sooo basically you didn't set channel indexes? Okay, try to set them correctly and we will see how it goes.

I am not sure about white-only device, as we have mostly RGBCW devices.

kogeki, 2023-02-11:
> No, it's RGBCW, or RGBW, white is after RGB.
Ah. This was the key to making it work. Thanks! I have one more question though. 

When trying to start discovery for home assistant, I get a message that says "No relay, PWM, sensor or power driver running". How do I get that to work?

Thank you so much for this amazing project!

openshwprojects, 2023-02-11:
What is your current configuration? I mean pins settings?

Or... @iprak , do we have discovery for RGBW? I think we may need to add, 99% people so far had RGBCW bulbs.

kogeki, 2023-02-11:
My pin settings are here:
[image]

I am not entirely sure that it is RGBW but the local tuya addon only pulls these data points: 
[image]
This leads me to believe that there is no color temperature, unless of course it is using RGB values to do that.