openshwprojects/OpenBK7231T_App #664: Nous Smart WiFi Bulb P8 RGBCW, GU10, EB01-B + BP5758 issue

lungucosmin02, 2023-02-13:
I have successfully flashed OTA the latest version of OpenBK7231N and completed the configuration from [here](https://www.elektroda.pl/rtvforum/topic3934503.html).
The setup that I have them in is the following: Smart Switch DS-101 2 Gang and 6 Nous P8 Smart Wifi Bulbs.
The problem is that whenever I switch them off directly from the switch (I disconnect them from the electricity) or restart them individually, whenever I turn them on, the color switches to the opposite one (e.g. if I have them on cold white, when I turn them back on they switch to warm white) and this happens to colours as well (e.g. if I have them set on green, when I switch them back on they come back up as magnetta / if I have them on red, they switch back on blue). Also, the slider inside the light's webpage shows up as active with the correct color.
Also, if I toggle the lights on and off (not from the switch but from the device) they switch to the correct color. 

**Firmware:**
- Version 1.15.441
- Device Nous Smart WiFi Bulb P8 RGBCW, GU10
- Chip/model: BK7231N
- Device config: the only startup command that I have is  *BP5758D_Map 2 0 1 4 3*

**To Reproduce**
Steps to reproduce the behaviour:
1. Turn off the main switch for the lights / restart the lightbulb while it is on a specified color;
2. Turn on the main switch for the lights / wait for reboot to complete;
3. The light will become available with the opposite light color, rather than the one it was set on, as per the description above. 

**Additional context**
I have tried disconnecting  MQTT as well, but the same behaviour was seen.

openshwprojects, 2023-02-13:
I know what's wrong. I just realized...
short startup command is applied after the led colour is restored...

openshwprojects, 2023-02-14:
https://github.com/openshwprojects/OpenBK7231T_App/commit/9095b2f0b962ca10012ff547af8b72bb3793c38a

lungucosmin02, 2023-02-14:
I have flashed the latest version and I can confirm that the issue has been resolved. The lightbulbs now retain the last state they were in, before turning them off. The issue can be closed.

closed 2023-02-15 by lungucosmin02