openshwprojects/OpenBK7231T_App #665: Pin config limit to only 32 pins.

valeklubomir, 2023-02-13:
W800 chip allow up to 44 PINs. Configuration struct can work only with up to 32 pins.
Upgrading to 64 limit is not an issue, done it and it works. 
Unfortunately this change will cause complete loss of configuration and reset device to default state.

Any idea how to avoid mass loss of configuration? 

I think not every user will be happy to reconfigure device after such firmware upgrade.

openshwprojects, 2023-02-13:
Very easy to do without a loss but time consuming. Basically, we would have to take the advantage of version field, and do something like that:
https://github.com/openshwprojects/OpenBK7231T_App/commit/858dc075d3d9aafc46f92cc89ab0d4a95b630b48
You could do that in PR and I can try help with testing (on Windows, etc, Windows can load flash image and simulate device), but please wait, because I am currently adding a field to config, so I need a day or two to finish.

I have BK7231 and BL602 at hand to test it, I also have W600/W800 somewhere, but i would have to look for that.

To be more precise, we would have to change the config structure to totally new one, but keep old struct in place with different name, something like, config_v3_t.
At config load time, we would need to check for older version and in conditional block just convert old config to new one on the fly.

We could also use this opportunity to improve the config structure itself, because it had some little padding issues.

valeklubomir, 2023-02-13:
Or We could limit it only to PLATFORM_W800 other device types do not need so many io pins. Would be just waste of space.
32 is enough.

Configuration struct is quite mess.

openshwprojects, 2023-02-13:
That might be the best idea. In worst case, only W800 users would lose config..

but config size is now 2016, and sector size is 1000 and flash wear is per erase cycle which is per sector, what means that.... we could easily bump struct size to 3000 and lose nothing. Because 2016 does erase for 3 sectors anyway.

valeklubomir, 2023-02-13:
Really 1000. Usually flash organization is 2's complement. 256, 512, 1024, 2048 or 4096.
PLATFORM_W800 specific changes did not affect W600 and BK7231 where tested changes also.

closed 2023-02-15 by valeklubomir