openshwprojects/OpenBK7231T_App #667: Missing BiStable relay

kodiakastu, 2023-02-14:
Hi! I just flashed a MCB238 (bk7231T) and the relay is a bistable kind. 
Set is on PWM3 and reset is on PWM2. Is there a way to byspass the fact that there's no relay_b in openbeken ?

openshwprojects, 2023-02-14:
Hello, maybe you could just use scripting to simulate it, but I can also add a Relay_b support for you directly.
Can you just provide me basic description how those two pins should operate for you? I haven't any devices with bistable relays yet so I am not sure how they work.

kodiakastu, 2023-02-14:
I don't mind doing it in scripting; I just don't know how to read a relay in script and how to run a script at boot. I'm willing to share a teardown and setup if it helps.

A bistable relay work like a flip-flop. You pull high a pin for a short moment to set/reset, then you pull it low again. ( set meaning the result is closing the relay, reset onpening it again).

I can feed you something from ESPhome if you wish.

Thanks for your quick reply.

openshwprojects, 2023-02-14:
So basically when channel becomes 1 you want to set PIN_SET to 1 for a moment , and when channel becomes 0 you want to set PIN_RESET to 1 for a moment?
Okay, I can do that for you in about 12 hours or so, it's a late night at my country now.
Regarding scripting, you can just do something like:
```
alias tick_setter backlog setChannel 2 1; addRepeatingEvent 1 1 setChannel 2 0;
addChangeHandler Channel1 == 0 tick_setter
```
but that's not needed, I will add a hardcoded option for that, but I will need you to help me with testing.

openshwprojects, 2023-02-14:
"tick_setter" could be futher modified to avoid using channel and just first set role for pin AlwaysHigh and then AlwaysLow...

openshwprojects, 2023-02-14:
Better version:

```
alias tick_setter backlog SetPinRole 15 AlwaysHigh; addRepeatingEvent 1 1 SetPinRole 15 AlwaysLow
alias tick_zeroer backlog SetPinRole 16 AlwaysHigh; addRepeatingEvent 1 1 SetPinRole 16 AlwaysLow
addChangeHandler Channel1 == 1 tick_setter
addChangeHandler Channel1 == 0 tick_zeroer
```

kodiakastu, 2023-02-14:
From what I understand from this platform's scripting, you seem to understand what I need. I'm gonna try to get something going with this script. 

It will be a pleasure to test anything 😄

kodiakastu, 2023-02-14:
I tried the first version but the "trigger" is way to long. Tried the second and nothing happens (the logs mentions nothing about changing something but setChannel does).

This code does exactly what I need:
```
alias tick_zeroer backlog setChannel 2 1; delay_ms 50; setChannel 2 0;
alias tick_setter backlog setChannel 3 1; delay_ms 50; setChannel 3 0;
addChangeHandler Channel0 == 1 tick_setter
addChangeHandler Channel0 == 0 tick_zeroer
```

openshwprojects, 2023-02-14:
Did you enter correct pin indexes in the second version?

Are you sure that button presses do not fire change handler? I think they do, but I will check this in few hours.

valekCN, 2023-02-14:
Recently added Bridge driver is handling BiStable relays or I call it polarized relays. 
[image]

kodiakastu, 2023-02-14:
> Did you enter correct pin indexes in the second version?
You are right. I forgot to change it. Tried again this morning and I have the same behavior with both version.

> Are you sure that button presses do not fire change handler? I think they do, but I will check this in few hours.
It does fire the handlers, but nothing happens. Same from HA via 

> Also, maybe your button is linked to channel 1 and not 0?
Button and LED are on Channel 0, but i revised it in the "always" version.

kodiakastu, 2023-02-14:
> Recently added Bridge driver is handling BiStable relays or I call it polarized relays. [image] FWD - pulse for switch relay ON REV - pulse for switch relay OFF Pulse length is 200ms

I cannot find the documentation about how to load drivers into the wiki; could you give me the command to run ?

openshwprojects, 2023-02-14:
"startDriver Bridge" in short startup command.

Btw, your script with "delay_ms" within alias will not work, delay_ms is only allowed within script thread. Maybe this is because you think that it doesn't work from HA?

kodiakastu, 2023-02-14:
Loaded the Bridge driver and it works perfectly (even with HA and the button). Is there a way to shorten the pulse ?

> Btw, your script with "delay_ms" within alias will not work, delay_ms is only allowed within script thread. Maybe this is because you think that it doesn't work from HA?

I don't know. Logs would still say that it was "flipping" the pins, but nothing would happen if it vas not triggered via the web page.

valekCN, 2023-02-14:
> Loaded the Bridge driver and it works perfectly (even with HA and the button). Is there a way to shorten the pulse ?
> 
pulse length is constant, I will create command to setup length.

kodiakastu, 2023-02-14:
>Different relays needs different length.

That's why I'm asking. 
Just for my information: why is it called bridge (and not bistable relay or 2 coil relay)? I'm just curious.

Secondly: where is all the documentation about this. I'm trying to find those thing in the wiki but I can't.

Sidenote: you guys are very quick to answer and it's amazing! How can I support you guys ?

openshwprojects, 2023-02-14:
Bridge driver is very new, if you can Valek, please submit some kind of information to Readme about how and when to use it. Or maybe to forum, or both.

@kodiakastu if you have any device that hasn't been yet documented, please submit a teardown/template here:
https://openbekeniot.github.io/webapp/devicesList.html
you could also write few words about how to configure that driver, so further users can just google information and find answers more easily

kodiakastu, 2023-02-14:
I've written a quick template for my device on the forum.

Since I've loaded de Bridge driver, I'm having lots of errors from it:
```
Info:DRV:ERRORBridge Driver OnChannelChanged: CH:15 VAL:2375
```
CH15 being configured as ADC.

I'm so sorry to bother you guys again. Do you want me to open a new thread about this and close the original as solved ?

valekCN, 2023-02-14:
That's strange. Can you list pin configuration.

kodiakastu, 2023-02-14:
[image]

openshwprojects, 2023-02-14:
It looks like a debug log within channel change callback that is catching every ADC change. I think this debug log line could be commented out or at least moved to the if block.

valekCN, 2023-02-14:
Will change it to debug level.

kodiakastu, 2023-02-14:
...And I don't know if this is wantend that way, but ADC value is only published if forced to do so

closed 2023-02-15 by kodiakastu