openshwprojects/OpenBK7231T_App #668: Is it able to support the mqtt connection via TLS (https connection) on MQTT remote server?

dinhchinh82, 2023-02-14:
First of all, thank you guys so much for supporting many new beken devices.
I've tried to flash some new Tuya wifi devices and they work well with Home Assistant.

By the way, is it able to connect the device direct to the remote MQTT server via TLS (https) such as Blynk.io ?

That would be good solution to make the wifi device can work independently without Home Assistant.

valeklubomir, 2023-02-14:
Secure MQTT is supported by LWIP, but requirres a lot of RAM. I think it is worth to test it.

openshwprojects, 2023-02-14:
It should be possible to enable it and do multiple builds per platform, for example one with secure connection enabled and one without it. It will be necessary over time as features grow and grow and the amount of flash memory we have is finite.

valekCN, 2023-02-14:
I will check LWIP if TLS is not already compiled. Normally during creating conenction to MQTT type is selected plain/secure and certificate must be loaded into RAM for TLS. From using MQTT SSL/TLS in ESP32 where LWIP is also used, I know that it requirres a lot of RAM. I think RAM shortage will be more in an issue than FLASH shortage.

dinhchinh82, 2023-02-15:
> Secure MQTT is supported by LWIP, but requirres a lot of RAM. I think it is worth to test it. EDIT: Changes would requirre:
> 
> * add configration for using secure connection
> * confugration to store/change certificate

could you please check the whether orginal Tuya firmware which might be already supported MQTT over TLS?
we hope to have testing version soon in near future.

Thank you so much.