openshwprojects/OpenBK7231T_App #673: Bk7231N ota not working

sowmimalu, 2023-02-15:
I am working on Bk7231N  - sdk , in defaultt code wifi is succesfully connecting and tried OTA getting the following error.

FOR OTA : 
after the connection of wifi Calling otarequest below function.
otarequest("http://xx.xx.x.x//Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl");

in OTa.h file changed the address for N varient.
#if PLATFORM_BK7231T
#define START_ADR_OF_BK_PARTITION_OTA 0x132000
#elif PLATFORM_BK7231N
#define START_ADR_OF_BK_PARTITION_OTA 0x12A000
#else
// TODO
#define START_ADR_OF_BK_PARTITION_OTA 0x12A000   //0x132000
#endif

ERROR LOG:
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
ssid:twtest2, 1
ht in scan
scan_start_req_handler
Info:HTTP_CLIENT:Parse url http://ovitag.s3.ap-south-1.amazonaws.com/TW_GWR/Smart_light/OpenBK7231T_App_1.0.0_19_1_23.rbl
Info:HTTP_CLIENT:host: 'ovitag.s3.ap-south-1.amazonaws.com', port: 80
Error:HTTP_CLIENT:getaddrinfo error
Error:HTTP_CLIENT:establish connection failed, error -1
Error:HTTP_CLIENT:httpclient_connect is error,ret = -8
Info:OTA:
Info:OTA:close OTA, addr 0xff000
Info:OTA:

note: i have checked the link that is working fine in browser.
help me out to find the solution.
Thanks.

valekCN, 2023-02-15:
Looks like DNS did not resolved host name into IP address. 
Or strange looking domain name was scrambled during parsing.
"xx.xx.xx.x" -> "xxxxx.com"

Or did you just due to privacy replaced it?
Can replaced domain name with IP and try again?

sowmimalu, 2023-02-16:
ovitag.s3.ap-south-1.amazonaws.com - host name and ota file is placed in server and same time i am able to download the file in browser.

valekCN, 2023-02-16:
First I expected buffer overrun. but length id only 87 bytes.

Trying from SBC Pi I receive this:
root@pi3:~# wget http://ovitag.s3.ap-south-1.amazonaws.com/Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl
--2023-02-16 11:32:36--  http://ovitag.s3.ap-south-1.amazonaws.com/Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl
Resolving ovitag.s3.ap-south-1.amazonaws.com (ovitag.s3.ap-south-1.amazonaws.com)... 3.5.211.148
Connecting to ovitag.s3.ap-south-1.amazonaws.com (ovitag.s3.ap-south-1.amazonaws.com)|3.5.211.148|:80... connected.
HTTP request sent, awaiting response... 403 Forbidden
2023-02-16 11:32:36 ERROR 403: Forbidden.

Similar for web browser.
<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>FC7A12CAWMSJP195</RequestId>
<HostId>XSPKcPUXYAWi3JSGXcCUexpk2+oz82cIE+Hju5sfeLDHNkVBsczCnlB7hH9GuEePR1X2/GVWgHc=</HostId>
</Error>

amazonaws usually requirres authetificated connection. 
I don't think that unsecured access is possible. 
I think your browser does all the secure connection for your on background. 

Once I have used AWS for download firmware for OTA on ESP and it requirred TLS/SSL with correct CA.

sowmimalu, 2023-02-16:
http://ovitag.s3.ap-south-1.amazonaws.com/TW_GWR/Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl   this my exact url where i place by rbl ota file .

valekCN, 2023-02-16:
Sorry now it works. I do some tests.

sowmimalu, 2023-02-16:
yes , otarequest("http://ovitag.s3.ap-south-1.amazonaws.com/TW_GWR/Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl");  i am passing this same url to the ota function  but i am getting the same error
Error:HTTP_CLIENT:getaddrinfo error
Error:HTTP_CLIENT:establish connection failed, error -1
Error:HTTP_CLIENT:httpclient_connect is error,ret = -8
Info:OTA:
Info:OTA:close OTA, addr 0xff000
Info:OTA:

valekCN, 2023-02-16:
My test:
12:30:32.819> Info:HTTP:http_fn_ota_exec: will try to do OTA for http://ovitag.s3.ap-south-1.amazonaws.com/TW_GWR/Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl 
12:30:32.819> Info:HTTP_CLIENT:Parse url http://ovitag.s3.ap-south-1.amazonaws.com/TW_GWR/Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl
12:30:32.819> Info:HTTP_CLIENT:host: 'ovitag.s3.ap-south-1.amazonaws.com', port: 80
12:30:32.819> Info:HTTP_CLIENT:HAL_TCP_Establish: created socket 2
12:30:33.169> Error:HTTP_CLIENT:success to establish tcp, fd=2
12:30:33.169> Info:OTA:init OTA, startaddr 0x132000
12:30:33.169> Info:OTA:
12:30:33.169> myhttpclientcallback state 0 total 0/0
12:30:33.169> Info:MQTT:OTA started MQTT will be closed
12:30:33.169> Info:MAIN:Time 56, idle 239265/s, sleep 117284/s(728426/s) free 79768, MQTT 1(1), bWifi 1, secWithNoPing -1, socks 3/38 
12:30:33.169> Info:HTTP_CLIENT:httpclient_recv 255 bytes has been read
12:30:33.169> Info:HTTP_CLIENT:httpclient_recv 226 bytes has been read
12:30:33.169> Info:HTTP_CLIENT:httpclient_recv 255 bytes has been read

It works.

Second: I think this file ins not for BK7231T. File size: 603248. OTA firmware for BK7231T has usually size 480000 bytes.

My BK7231T module flashed it, but afterwards it is bricked.

sowmimalu, 2023-02-16:
this file is for BK7231N  and trying to do Ota in same BK7231N

valekCN, 2023-02-16:
Sorry. I don't why I used T instead of N. 
I have used just command in Web UI:
[image]

Before I started to directly write to code. Switching to N version.

sowmimalu, 2023-02-16:
okay  , you have any idea on this why i am getting the following error 
same thing parsing the url into my code.
Error:HTTP_CLIENT:getaddrinfo error
Error:HTTP_CLIENT:establish connection failed, error -1
Error:HTTP_CLIENT:httpclient_connect is error,ret = -8
Info:OTA:
Info:OTA:close OTA, addr 0xff000
Info:OTA:

valekCN, 2023-02-16:
My code:
`case WIFI_STA_CONNECTED:
            g_bHasWiFiConnected = 1;
            ADDLOGF_INFO("Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - %i\r\n", code);
            if(bSafeMode == 0){
                if(strlen(CFG_DeviceGroups_GetName())>0){
                    ScheduleDriverStart("DGR",5);
                }
                // if SSDP should be active,
                // restart it now.
                if (DRV_SSDP_Active){
                    ScheduleDriverStart("SSDP",5);
                    //DRV_SSDP_Restart(); // this kills things
                }
                if (ota_progress() == -1)
                {
                    otarequest("http://ovitag.s3.ap-south-1.amazonaws.com/TW_GWR/Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl");
                }
            }

            break;
`

Log: 
13:06:30.814> configuring interface mlan (with DHCP client)WARN: TCPIP mutex is NOT locked (1) caller 580AB
13:06:30.814> WARN: TCPIP mutex is NOT locked (1) caller 580AB
13:06:30.814> Info:HTTP_CLIENT:Parse url http://ovitag.s3.ap-south-1.amazonaws.com/TW_GWR/Smart_light/OpenBK7231T_App-main_1.0.0_V11_A.rbl
13:06:30.814> Info:MAIN:wl_status 11
13:06:30.814> Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - 4
13:06:30.814> Info:HTTP_CLIENT:host: 'ovitag.s3.ap-south-1.amazonaws.com', port: 80
13:06:30.814> Info:HTTP_CLIENT:HAL_TCP_Establish: created socket 2
13:06:30.814> Info:MAIN:Time 11, idle 61394/s, free 71056, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 3/38 
13:06:30.814> Error:HTTP_CLIENT:success to establish tcp, fd=2
13:06:30.814> [Flash]id:0x1c7015
13:06:30.814> Info:OTA:init OTA, startaddr 0x12a000
13:06:30.814> Info:OTA:
13:06:30.814> myhttpclientcallback state 0 total 0/0
13:06:30.814> Info:HTTP_CLIENT:httpclient_recv 255 bytes has been read
13:06:30.814> Info:HTTP_CLIENT:httpclient_recv 226 bytes has been read

Unfortunatelly image is not flashed correctly device loops always trying to get OTA

valekCN, 2023-02-16:
Calling it from code is not the best idea.