openshwprojects/OpenBK7231T_App #674: BL602 - Device Configuration

biggoof1, 2023-02-15:
Hi all,

this is more a question..
I have a LED controller like this..
[Magic Home Controller](https://www.elektroda.com/rtvforum/topic3889041-270.html) 

Only difference is, that it is RGBW instead of RGB as in the article.
From the webapp I can not find a corresponding device.

Therefore the channels have different names and choosing colors does only work with manually switching every channel instead using the color picker from the webapp.

I want to control the stripe with mqtt by sending a dimmer value and one color code.

Is there anything I can adjust?

Thanks for your help,
Christian

openshwprojects, 2023-02-15:
Hello.
Well, first, do you know which pins are used for Red, Green, Blue and White?

biggoof1, 2023-02-15:
I followed the guide and picture from the linked forum thread. There is the wireing.
I have all colors available. :-)

Webapp says cwhite, blue, green, no color from Top to bottom on the status page. 
At the moment white is red and nothing is white.
I changed it,but the color balance only switches on blue and green.
So I reverted as in the thread.

openshwprojects, 2023-02-16:
Red is channel 0, Green is Channel 1, Blue is channel 2... and then white, but we might need to work on white.

Does the colour picker on main HTTP page (not app) works ok for RGB?

If not, then reorder channels

biggoof1, 2023-02-16:
Hi,

I have attached some screens..
Version 1.15.450

Does this help?

Christian

[image]

[image]

[image]

udovic2100, 2023-02-18:
> Hi,
> 
> 
> 
> I have attached some screens..
> 
> Version 1.15.450
> 
> 
> 
> Does this help?
> 
> 
> 
> Christian
> 
> 
> 
> [image]
> 
> 
> 
> [image]
> 
> 
> 
> [image]
> 
> 

This configuration worked for me. Thanks.