openshwprojects/OpenBK7231T_App #678: PowerSave command - i really need feedback from You guys

openshwprojects, 2023-02-18:
Thanks to the @btsimonh  hard work, we have added a PowerSave command some time ago.

This enables lower power mode on T and N, and it seems to work well, as long as you don't have the Web App console log open (it does HTTP request every 2 seconds or so and breaks the sleep).

Apart from that, the device seem to work 100% normal with PowerSave on, there is no noticeable extra MQTT lag, no input lag, etc. It's not a deep sleep!

My question is - **should we enable PowerSave by default?**

There are some devices with weak power supplies that can get premature death of caps by too high power usage:
https://www.elektroda.pl/rtvforum/topic3898805.html

**Can anyone reading (and willing to do some testing and take some potential risks) put PowerSave in short startup command and/or autoexec.bat and tell me if device still behaves well?**

avassdal, 2023-02-18:
I can do some testing if needed

openshwprojects, 2023-02-18:
just put PowerSave in short startup command or autoexec and reboot and tell me if it still works okay.

Check if device works with HA, if it's stable, if it reacts to WWW panel, etc etc etc. Does it keep connection with wifi in long term, is it still useable

lungucosmin02, 2023-02-18:
I've enabled it on Nous Smart WiFi Bulb P8 RGBCW - version 1.15.460  BK7231N chipset. 
There's no signs of noticeable slowdown as of now. The lightbulb performs operations as the other ones that do not have the feature enabled. 
Also works fine with HA. Will see how this works on the long run.

openshwprojects, 2023-02-18:
If working, this command could even reduce your electricity bills.

valekCN, 2023-02-18:
After using following autoexec.bat

```
addClockEvent 17:30:00 0xff 1 ToggleChannel 0
addClockEvent 17:35:00 0xff 2 ToggleChannel 0
PowerSave 1

```
Device was no longer accessible by WebGUI.
I had to restart it into SAFE. 
WebApp was not working in safe mode.
Command 'lfs_remove autoexec.bat' was not working also LFS not initialized under SAFE mode ?????
Finally adding ;lfs_remove autoexec.bat; to startup command killed autoexec.bat

I try it again without powersave.

openshwprojects, 2023-02-18:
While I don't think that PowerSave can be that bad, I still think you have very good point with LFS.

It seems that lfs_format command should be also available in unsafe mode, so we can clear littlefs scripts in case of emergency.

openshwprojects, 2023-02-18:
I think I can move "LFSAddCmds" from Unsafe init to safe init....

valekCN, 2023-02-18:
My BK7231N completely frozen after PowerSave 1;
This time I executed it from webgui command. 
LOG is dead.  (I know that active log is against power saving) 
Web gui is dead. WebApp is also dead.
Button is not working also. Device is frozen.

I am using it on DINRAIL power meter. Probably not the ideal device.
I change it to device which has wired UART log. To see it exactly.

openshwprojects, 2023-02-18:
Hmm my N devices seems to work with PowerSave:
[image]
so the question would be where is the difference.....

valekCN, 2023-02-18:
I have running:
[image]

cplagz, 2023-02-19:
12 hours uptime no issues yet

biggoof1, 2023-02-19:
BL602 - Power save enabled.
Up to now (20 minutes) everything seems to be normal.
Will post, if something seems to be weird.

openshwprojects, 2023-02-19:
As said in first post - it's for T and N, not BL602, but... you gave me an idea, maybe it's possible also on BL602?
I will update PowerSave to do something on BL602, wait, but don't try it yet.

biggoof1, 2023-02-19:
Sorry.. Did not map T and N to the model ;-) But at least it did not disturb anything on the BL602 ;-)

openshwprojects, 2023-02-19:
Yea, it seems it should be possible on BL602... thanks to your comment, of course! @biggoof1  I didn't check that earlier.
https://github.com/openshwprojects/OpenBK7231T_App/commit/4eff71a4ae8f409264a4ab4196ade613dd16c2ab
let me try...

biggoof1, 2023-02-21:
Hi,

updated to 1.15.476 as I understood that PowerSave for 602 has been enabled?

PowerSave 1 executed as Command and it states, that is has been set to 1.

Can I check somehow that it is really in PowerSave mode?
For now (just a few tests) BL602 seems to behave as usual.