openshwprojects/OpenBK7231T_App #680: BL602 Factory firmware backup before flashing OpenBK

ataa, 2023-02-19:
I have some BL602 based RGBCW bulb, flashed some of them successfully but a some their features (Bluetooth remote control, ...) lost as yet to be implemented in this firmware. Is there a way to dump/backup the factory firmware before flashing this firmware. I have BLDevCube but couldn't find any way to backup the firmware.

openshwprojects, 2023-02-19:
Hello, accordinng to my knowledge there is no way to do this in BLDevCube, but I may be wrong.

ataa, 2023-02-19:
After many unsuccessful attempts, I managed to dump the flash using blflash using command below, but I am not sure if I did it correctly, The part I am not sure is the ending address.

`blflash-windows-amd64.exe dump BL602-Lamp.bin -b 230400 --port COM3 0 0x200000`

openshwprojects, 2023-02-19:
Wow, nice, this is very useful. Can you submit it in form of a short tutorial/topic to our Tutorials section?
https://www.elektroda.com/rtvforum/forum517.html
Give a proper title to topic, write short description how to do that, etc, where to get blflash
Thanks!

ataa, 2023-02-19:
Sure, I'll do that as soon as I make sure I did the ending address right and the dumped firmware is valid.
Do you know the BL602 flash ending address? or a way to check the dumped flash validity? I found 0x200000 end address in blfash repo.

openshwprojects, 2023-02-19:
I don't know, but... if you launch flasher from our SDK:
[image]
You get reference to: 
W:\GIT\OpenBL602\tools\flash_tool\bl602\partition\partition_cfg_2M.toml
and that file contains:
```
[pt_table]
#partition table is 4K in size
address0 = 0xE000
address1 = 0xF000

[[pt_entry]]
type = 0
name = "FW"
device = 0
address0 = 0x10000
size0 = 0xC8000
address1 = 0xD8000
size1 = 0x88000
# compressed image must set len,normal image can left it to 0
len = 0

[[pt_entry]]
type = 2
name = "mfg"
device = 0
address0 = 0x160000
size0 = 0x32000
address1 = 0
size1 = 0
# compressed image must set len,normal image can left it to 0
len = 0

[[pt_entry]]
type = 3
name = "media"
device = 0
address0 = 0x192000
size0 = 0x57000
address1 = 0
size1 = 0
# compressed image must set len,normal image can left it to 0
len = 0

[[pt_entry]]
type = 4
name = "PSM"
device = 0
address0 = 0x1E9000
size0 = 0x8000
address1 = 0
size1 = 0
# compressed image must set len,normal image can left it to 0
len = 0

[[pt_entry]]
type = 5
name = "KEY"
device = 0
address0 = 0x1F1000
size0 = 0x2000
address1 = 0
size1 = 0
# compressed image must set len,normal image can left it to 0
len = 0

[[pt_entry]]
type = 6
name = "DATA"
device = 0
address0 = 0x1F3000
size0 = 0x5000
address1 = 0
size1 = 0
# compressed image must set len,normal image can left it to 0
len = 0

[[pt_entry]]
type = 7
name = "factory"
device = 0
address0 = 0x1F8000
size0 = 0x7000
address1 = 0
size1 = 0
# compressed image must set len,normal image can left it to 0
len = 0

#if user want to put RF calibration data on flash, uncomment following pt entry
#[[pt_entry]]
#type = 8
#name = "rf_para"
#device = 0
#address0 = 0x1FF000
#size0 = 0x1000
#address1 = 0
#size1 = 0
## compressed image must set len,normal image can left it to 0
#len = 0
```

ataa, 2023-02-19:
Latest BLDevCube is 1.8.2 which looks different than your screenshot and there's a change in filenames and partition table. After running it and selecting that partition cfg, I get `Warning: This v1 partition table is deprecated`. 

To test if the dumped flash is valid, I flashed your firmware on BL602, then flash the factory one again to check if the dumped flash is valid and working as expected. I'll report back.

openshwprojects, 2023-02-20:
But does OBK boot correctly?

Sso far I've been using the older version of BLDevCube. I am not aware yet how to use the newest version, it seems that Bouffalo Lab has changed something in their partitions standard. It requires more investigation.

If you flash device with OBK, and then read back binary, are the files same? 

Maybe binary for flashing needs to be prefixed with a certain header?

ataa, 2023-02-20:
I tested it and it does work on mine, but it may not be the case for recent chips with new partition table. Also please use newer version in the upcoming video tutorials as the newer version of BLDevCube has new UI and different tabs (IoT, MCU, MFG).

[image]

openshwprojects, 2023-02-20:
Wait, my BL602 knowledge is very small, so the new BLDevCube works on older chips if you use older TOML (older partition table)? 

That's good to know. We already have recorded the BL602 Magic Strip tutorial on older DevCube, but I will try to check out the new version soon.

ataa, 2023-02-20:
Yes, I tested it for BL-M5 V1.2 Board on ZJ-PWH-RGBWW-L2 V1.1 Bulb and flashed the latest released of your firmware successfully.

Flash Log:
`Warning: This v1 partition table is deprecated
[05:29:43.857] - serial type is general
[05:29:43.989] - Open COM3 Success
[05:29:48.020] - Close COM3 Success
[05:29:48.020] - ========= Interface is Uart =========
[05:29:48.036] - eflash loader bin is eflash_loader_40m.bin
[05:29:48.036] - ========= chip flash id: ef4015 =========
[05:29:48.051] - create partition.bin, pt_new is True
[05:29:48.067] - fw_boot_head_gen xtal: 40M
[05:29:48.067] - Create bootheader using C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\img_create_iot/efuse_bootheader_cfg.ini
[05:29:48.067] - Updating data according to <C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\img_create_iot/efuse_bootheader_cfg.ini[BOOTHEADER_CFG]>
[05:29:48.083] - Created file len:176
[05:29:48.098] - Create efuse using C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\img_create_iot/efuse_bootheader_cfg.ini
[05:29:48.098] - Updating data according to <C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\img_create_iot/efuse_bootheader_cfg.ini[EFUSE_CFG]>
[05:29:48.098] - Created file len:128
[05:29:48.114] - ========= sp image create =========
[05:29:48.114] - Image hash is b'2b6e636f8817b35bc922846d7a82f3c9b3947bb7c19fad1078859fa8b9692e10'
[05:29:48.114] - Hash ignored
[05:29:48.130] - Header crc ignored
[05:29:48.130] - Write flash img
[05:29:48.145] - fw_boot_head_gen xtal: 40M
[05:29:48.145] - Create bootheader using C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\img_create_iot/efuse_bootheader_cfg.ini
[05:29:48.145] - Updating data according to <C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\img_create_iot/efuse_bootheader_cfg.ini[BOOTHEADER_CFG]>
[05:29:48.161] - Created file len:176
[05:29:48.161] - Create efuse using C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\img_create_iot/efuse_bootheader_cfg.ini
[05:29:48.161] - Updating data according to <C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\img_create_iot/efuse_bootheader_cfg.ini[EFUSE_CFG]>
[05:29:48.177] - Created file len:128
[05:29:48.177] - ========= sp image create =========
[05:29:48.192] - Image hash is b'a1cb783954e80fe98f263e509b5da2080fb40caa55f1c4cb2a7e910c91aa45ce'
[05:29:48.192] - Header crc: b'0d97d3cf'
[05:29:48.192] - Write flash img
[05:29:48.208] - FW Header is 176, 3920 still needed
[05:29:48.208] - FW OTA bin header is Done. Len is 4096
[05:29:48.599] - BL60X_OTA_Ver1.0
[05:29:48.958] - FW OTA bin is Done. Len is 793120
[05:29:50.239] - OTA XZ file len = 423872
[05:29:50.239] - Partiton len = 557056
[05:29:50.239] - BL60X_OTA_Ver1.0
[05:29:50.458] - FW OTA xz is Done
[05:29:50.458] - ========= eflash loader config =========
[05:29:51.005] - Version: eflash_loader_v2.4.3
[05:29:51.020] - Program Start
[05:29:51.020] - ========= eflash loader cmd arguments =========
[05:29:51.020] - C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\eflash_loader/eflash_loader_cfg.ini
[05:29:51.020] - Config file: C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\eflash_loader/eflash_loader_cfg.ini
[05:29:51.020] - serial port is COM3
[05:29:51.020] - cpu_reset=False
[05:29:51.020] - chiptype: bl602
[05:29:51.020] - ========= Interface is uart =========
[05:29:51.020] - com speed: 2000000
[05:29:51.020] - Eflash load helper file: C:\Users\Ataa\Downloads\BouffaloLabDevCube-v1.8.2\chips\bl602\eflash_loader/eflash_loader_40m.bin
[05:29:51.020] - ========= load eflash_loader.bin =========
[05:29:51.020] - Load eflash_loader.bin via uart
[05:29:51.020] - ========= image load =========
[05:29:51.409] - tx rx and power off, press the machine!
[05:29:51.409] - cutoff time is 0.1
[05:29:51.518] - power on tx and rx
[05:29:52.347] - reset cnt: 0, reset hold: 0.005, shake hand delay: 0.1
[05:29:52.347] - clean buf
[05:29:52.377] - send sync
[05:29:52.595] - ack is b'4f4b'
[05:29:52.643] - shake hand success
[05:29:52.659] - get_boot_info
[05:29:54.674] - ack is b''
[05:29:54.674] - Not ack OK
[05:29:54.674] - result:
[05:29:54.674] - fail
[... 390 lines cut ...]
[05:30:46.460] - Load 405504/423872 {"progress":95}
[05:30:46.475] - Load 407552/423872 {"progress":96}
[05:30:46.492] - Load 409600/423872 {"progress":96}
[05:30:46.522] - Load 411648/423872 {"progress":97}
[05:30:46.538] - Load 413696/423872 {"progress":97}
[05:30:46.570] - Load 415744/423872 {"progress":98}
[05:30:46.585] - Load 417792/423872 {"progress":98}
[05:30:46.616] - Load 419840/423872 {"progress":99}
[05:30:46.631] - Load 421888/423872 {"progress":99}
[05:30:46.647] - Load 423872/423872 {"progress":100}
[05:30:46.647] - Load 423872/423872 {"progress":100}
[05:30:46.663] - Write check
[05:30:46.694] - Flash load time cost(ms): 5078.30029296875
[05:30:46.694] - Finished
[05:30:46.710] - Sha caled by host: e3ffb219e63d7141f09d5ba0335a7a1ed1faa0043517e7e193d6769c54e9d962
[05:30:46.710] - xip mode Verify
[05:30:46.929] - Read Sha256/793120
[05:30:46.929] - Flash xip readsha time cost(ms): 218.76708984375
[05:30:46.929] - Finished
[05:30:46.929] - Sha caled by dev: e3ffb219e63d7141f09d5ba0335a7a1ed1faa0043517e7e193d6769c54e9d962
[05:30:46.929] - Verify success
[05:30:46.945] - Program Finished
[05:30:46.945] - All time cost(ms): 12462.46826171875
[05:30:47.069] - close interface
[05:30:47.069] - [All Success]`

openshwprojects, 2023-02-21:
Hello, is the _[WARN  blflash::flasher] sha256 not match_ issue resolved?

Btw, do you know how to reset BL602 efuse to factory settings?

ataa, 2023-02-21:
- Still testing different methods, rates, UARTs. None of them was successful. I am using SIL CP2102 UART and what I found is that by default high baudrates (1500000+) are not reliable on this chip, but there's a way to reprogram the chip, So I am going to reprogram it and test again.
- I have no idea, Is that even possible? Isn't efuse a "one time write" flash?

ataa, 2023-02-21:
Good News!
Reprogramed CP2102 for 2000000 baudrate and reflashed factory firmware using Latest BLDevCube and it was a success. Everything works as expected even the Bluetooth remote for BL602 Bulbs.
Now I have to figure out a way to make an OTA flash out of it.
It would be nice if you can add a section to your forum for sharing factory firmwares for (for research purposes OFC)

openshwprojects, 2023-02-22:
That's great, what was exactly the steps you've taken to do a read and then backup restore? 

Hmmm I think you can just attach factory firmware to the teardown article, we have our own attachment (and images) hosting on Elektroda.

Also, I am currently having an issue with my BL602 dev board - I tried to change mac and supposedly I overwrote efuse and it does not connect to wifi even after full erase of flash. Maybe your dump could help me recover my device.

ataa, 2023-02-22:
- To dump the current firmware, use [BLflash](https://github.com/spacemeowx2/blflash):  `blflash-windows-amd64.exe dump factory.bin -b 230400 --port COM3 0 0x200000`

- To flash the factory image, use BLDevCube 1.8.2, These settings worked for me:

[image]

I have two factory images, try them out:

[dump.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/10801829/dump.zip)
[Bl602Lamp-4th.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/10801834/Bl602Lamp-4th.zip)

openshwprojects, 2023-02-22:
From which exactly devices are those firmwares and how to pair them with (which app?) for testing purposes?

ataa, 2023-02-22:
- Magichome Wifi LED controller
- 10w RGBCW bulb (surplife / magichue)

openshwprojects, 2023-02-22:
After flashing [Bl602Lamp-4th.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/10801834/Bl602Lamp-4th.zip),  should I get anything on UART, like a log output?

openshwprojects, 2023-02-22:
Well, it's not working for me. Not in 1.4.0 BldevCube and not in the 1.8.0.

Console doesn't print anything after I repower the module into the run mode.
With Chip Erase = True:
[image]

Witch Chip Erase = False;
[image]

nothing:
[image]

ataa, 2023-02-22:
does it activate Wifi AP after removing boot bin connection and power cycle? AP SSID should start with LEDnet00

openshwprojects, 2023-02-22:
No AP at all.

It only shows AP when I flash OpenBL602 build by the old method with old dev cube like in my guide, but the AP is not functional after I supposedly broke refuse (AP shows up but my PC is unable to connect to it). I was playing with MAC address change and it's like that since I used EF_Ctrl_Write_MAC_Address_Opt function. The device on OpenBL602 works, but my PC fails to connect to it's AP, and when it's in client mode, the device fails to get DHCP IP from my Router (even though it has unique MAC).

@ataa btw didn't you read flash from offset 0x0 and ain't I flashing it now to 0x2000 offset? Just a hunch...

Log:
```

[17:46:51.075] - EFUSE_CFG
[17:46:51.076] - BOOTHEADER_CFG
[17:46:51.092] - Create bootheader using W:\GIT\bl_iot_sdk\tools\flash_tool\chips\bl602\img_create_mcu/efuse_bootheader_cfg.ini
[17:46:51.093] - Updating data according to <W:\GIT\bl_iot_sdk\tools\flash_tool\chips\bl602\img_create_mcu/efuse_bootheader_cfg.ini[BOOTHEADER_CFG]>
[17:46:51.100] - Created file len:176
[17:46:51.111] - Image create path: W:\GIT\bl_iot_sdk\tools\flash_tool\chips\bl602\img_create_mcu
[17:46:51.113] - Config file: W:\GIT\bl_iot_sdk\tools\flash_tool\chips\bl602\img_create_mcu/img_create_cfg.ini
[17:46:51.152] - Image hash is b'4d0927172cb417c9a8c3dba45fd7a2ca5f38ebe10586f37e99517d832bc90e7d'
[17:46:51.154] - Header crc: b'9567167c'
[17:46:51.156] - Write flash img
[17:46:51.201] - ========= eflash loader config =========
[17:46:51.667] - Version: eflash_loader_v2.4.2
[17:46:51.669] - Program Start
[17:46:51.669] - ========= eflash loader cmd arguments =========
[17:46:51.670] - None
[17:46:51.670] - Config file: W:\GIT\bl_iot_sdk\tools\flash_tool\chips\bl602\eflash_loader\eflash_loader_cfg.ini
[17:46:51.676] - serial port is COM15
[17:46:51.677] - cpu_reset=False
[17:46:51.683] - chiptype: bl602
[17:46:51.683] - ========= Interface is uart =========
[17:46:51.683] - com speed: 2000000
[17:46:51.684] - Eflash load helper file: W:\GIT\bl_iot_sdk\tools\flash_tool\chips\bl602\eflash_loader/eflash_loader_40m.bin
[17:46:51.685] - ========= load eflash_loader.bin =========
[17:46:51.685] - Load eflash_loader.bin via uart
[17:46:51.685] - ========= image load =========
[17:46:51.951] - tx rx and power off, press the machine!
[17:46:51.953] - cutoff time is 0.1
[17:46:52.055] - power on tx and rx
[17:46:52.785] - reset cnt: 0, reset hold: 0.005, shake hand delay: 0.1
[17:46:52.786] - clean buf
[17:46:52.787] - send sync
[17:46:53.003] - ack is b'4f4b'
[17:46:53.035] - shake hand success
[17:46:53.047] - get_boot_info
[17:46:53.050] - data read is b'010000000000000003000000fd1fc305b9181700'
[17:46:53.051] - ========= ChipID: 18b905c31ffd =========
[17:46:53.051] - last boot info: None
[17:46:53.051] - sign is 0 encrypt is 0
[17:46:53.052] - segcnt is 1
[17:46:53.062] - segdata_len is 37152
[17:46:53.147] - 4080/37152
[17:46:53.232] - 8160/37152
[17:46:53.318] - 12240/37152
[17:46:53.404] - 16320/37152
[17:46:53.488] - 20400/37152
[17:46:53.574] - 24480/37152
[17:46:53.659] - 28560/37152
[17:46:53.745] - 32640/37152
[17:46:53.830] - 36720/37152
[17:46:53.841] - 37152/37152
[17:46:53.844] - Run img
[17:46:53.947] - Load helper bin time cost(ms): 2261.14111328125
[17:46:54.048] - Flash load shake hand
[17:46:54.060] - default set DTR high
[17:46:54.162] - clean buf
[17:46:54.164] - send sync
[17:46:54.381] - ack is b'4f4b'
[17:46:54.414] - Read mac addr
[17:46:54.417] - flash set para
[17:46:54.418] - ========= flash read jedec ID =========
[17:46:54.421] - Read flash jedec ID
[... 279 lines cut ...]
[17:47:05.819] - Load 430080/441016 {"progress":97}
[17:47:05.837] - Load 432128/441016 {"progress":97}
[17:47:05.854] - Load 434176/441016 {"progress":98}
[17:47:05.877] - Load 436224/441016 {"progress":98}
[17:47:05.894] - Load 438272/441016 {"progress":99}
[17:47:05.918] - Load 440320/441016 {"progress":99}
[17:47:07.230] - Load 441016/441016 {"progress":100}
[17:47:07.231] - Load 441016/441016 {"progress":100}
[17:47:07.231] - Write check
[17:47:07.271] - Flash load time cost(ms): 6034.134521484375
[17:47:07.274] - Finished
[17:47:07.296] - Sha caled by host: 4d0927172cb417c9a8c3dba45fd7a2ca5f38ebe10586f37e99517d832bc90e7d
[17:47:07.298] - xip mode Verify
[17:47:07.869] - Read Sha256/2097152
[17:47:07.870] - Flash xip readsha time cost(ms): 569.56005859375
[17:47:07.870] - Finished
[17:47:07.872] - Sha caled by dev: 4d0927172cb417c9a8c3dba45fd7a2ca5f38ebe10586f37e99517d832bc90e7d
[17:47:07.872] - Verify success
[17:47:07.874] - Program Finished
[17:47:07.874] - All time cost(ms): 16204.5615234375
[17:47:07.977] - close interface
[17:47:07.978] - [All Success]
```

I don't have another BL602 dev board currently and I don't know if I can recover my one and if the issue with your flash is related to my board having broken mac efuse or to something else. BUt judging from OpenBL602 behaviour, I would at least expect log output to work and AP to show up.

ataa, 2023-02-22:
I have 10s of the same board at the moment and no access to other BL602 boards, If I flash OpenBK and then reflash the factory bin file using the above method, after removing boot pin connection and power cycle, factory firmware works fine, AP shows up and I get model number on the terminal. 

So I am not sure if there's security feature in place to prevent you from using it on your board or not. even in that case we should at least get some error messages