openshwprojects/OpenBK7231T_App #686: BP1658CJ max power limit

kogeki, 2023-02-20:
Hello,

I am reading the documentation for ESPHome and I see that the [BP1658CJ](https://esphome.io/components/output/bp1658cj.html) driver has the option to increase the max power draw. How can I set this setting with OpenBK? Thanks!

openshwprojects, 2023-02-21:
Hello, it seems that the command has to be yet added. I can try to do that today, can you help with testing?

kogeki, 2023-02-21:
Yes I can.

kogeki, 2023-03-04:
Not to pressure you (I'm sure you have other things to worry about), but is there any chance you can look at this? Thanks!

openshwprojects, 2023-03-04:
I don't have this device at hand, all I can do is add command and let you test, but I will need multiple tests...
I added BP1658CJ_Current that takes two arguments and need light change after that to apply (to refresh),
try BP1658CJ_Current 2 2 or something like that.
I am not sure if it will work, there is also option that I might to do one more change. Please check and let me know how it behaves.

kogeki, 2023-03-04:
I appreciate the response. After trying it out with various values, I have noticed that while it does make the light brighter (sometimes), it also makes the LED Temp slider behave very erratically. 
[image]
^this makes the light turn off (`BP1658CJ_Current 2 2`)
[image]
^this is really dim warm white (`BP1658CJ_Current 2 2`)
Using other values just makes them have the same behavior but brighter or dimmer.

Any other things I can try? Thanks!

openshwprojects, 2023-03-04:
We need to try things but I don't have this device. Do you know how to download build from a pull request?
Try that: https://github.com/openshwprojects/OpenBK7231T_App/pull/708
here, at the bottom:
https://github.com/openshwprojects/OpenBK7231T_App/actions/runs/4332532167
[image]

download this build and try again

kogeki, 2023-03-04:
This version works perfectly. Thanks a lot! I am wondering though, what does each value mean? I am running with `15 15` and it seems to be equal to the real light.

openshwprojects, 2023-03-04:
wow, that's great, so I managed to get it without device on the second try.

	//cmddetail:{"name":"BP1658CJ_Current","args":"[MaxCurrentRGB][MaxCurrentCW]",

i will merge this into main tree tomorrow

kogeki, 2023-03-04:
Thanks so much! Do you happen to have a crypto currency donation address? I would be glad to send some money. Thanks for this great project!

openshwprojects, 2023-03-05:
No, only Paypal, do you prefer another donation method? If so, what kind of method and why? I always wanted to get into that crypto thing, but never had time to look into that well.
https://www.paypal.com/paypalme/openshwprojects

kogeki, 2023-03-05:
I am underage and my PayPal account is limited. I can't send money. Bitcoin would be the best option, if you can.