openshwprojects/OpenBK7231T_App #688: MQTT "get" channels status is not picked up by Home Assistant upon reboot

p1ne, 2023-02-21:
**Describe the bug**
Only "set" mqtt channels status is surviving MQTT broker/Home Assistant restart, "get" channels are initially empty. Thus default value for all the switches in Home Assistant is OFF despite actual state.

**Firmware:**
- Version 1.15.4.39
- Device AOFO Power Strip
- Chip/model: CB2S/BK7231N
- Device config
**autoexec.bat**
```
startDriver TuyaMCU
setChannelType 11 toggle
setChannelType 12 toggle
setChannelType 13 toggle
setChannelType 14 toggle
setChannelType 15 toggle
setChannelType 16 toggle
linkTuyaMCUOutputToChannel 38 bool 11
linkTuyaMCUOutputToChannel 1 bool 12
linkTuyaMCUOutputToChannel 2 bool 13
linkTuyaMCUOutputToChannel 3 bool 14
linkTuyaMCUOutputToChannel 4 bool 15
linkTuyaMCUOutputToChannel 7 bool 16

```
**On flags:**
Flag 2 - [MQTT] Broadcast self state every N (def: 60) seconds
Flag 21 - [MQTT] Retain power channels (Relay channels, etc)

**HA switch config (for switch 1, other 4 are similar)**
```
- name: "Switch Top Right 1"
  unique_id: power_top_right_1
  state_topic: "power_top_right/12/get"
  command_topic: "power_top_right/12/set"
  qos: 1
  payload_on: 1
  payload_off: 0
  retain: true
```

**To Reproduce**
Steps to reproduce the behavior:
1. Setup HA and OpenBeken according to configuration
2. Click on some switches in OpenBeken Web UI, observe status changed to On both in OpenBeken and Home Assistant and both set and get topics set in MQTT
3. Reboot Home assistant
4. Observe all switches off and only "set" values in MQTT

openshwprojects, 2023-02-22:
Hello, do you have a "Broadcast self state on mqtt connect"?

p1ne, 2023-02-22:
Tried it as well - the only difference is that I get ip/sockets/uptime and other informational topics. all get topics are still absent

[image]

as a temporary workaround I'm using 'set' as state_topic, but it is wrong by conception - just because 'get' topic value is confirmed by actual relay state

p1ne, 2023-04-14:
Any chance to get it fixed soon?

konradmb, 2023-10-07:
Quick workaround: enable `Flag 21 - [MQTT] Retain power channels (Relay channels, etc)`