openshwprojects/OpenBK7231T_App #689: Support RF Control

TriForMine, 2023-02-22:
I have successfully flashed this product:
https://fr.aliexpress.com/item/1005004558905919.html

Is there any way the RF part would be supported?

# Pins

I have the IR Receiver/Transmitter working perfectly with the following pins
P7 - IRSend
P8 - IRRecv
P9 - WifiLED

# Photos
[image]
[image]
[image]
[image]

ataa, 2023-02-28:
I have similar one with model S11 label, it has BK7231N and SH4 modules. PCB Version S11-SH4-CBU-V2.4
[SH4 Module Datasheet](https://developer.tuya.com/en/docs/iot/sh4-module-datasheet?id=Ka04qyuydvubw)

uzlonewolf, 2023-05-04:
I traced most of the other pins out:

| CBU Pin       | BK7231N       | Function      | Pad           |
| ------------- | ------------- | ------------- | ------------- |
|1|P14 (SPI_SCK)|SH4 CLK|SCK
|2|P16 (SPI_MOSI)|SH4 SDIO via R9||
|3|P20|SH4 GPIO1|INT1|
|4|P22|SH4 GPIO2|INT2|
|5|ADC|Button (S1) via RC network|ADC|
|6|RX2|||
|7|TX2|||
|8|P8|IR receiver||
|9|P7|IR tx||
|10|P6|||
|11|P26|||
|12|P24|||
|13|GND|GND|GND|
|14|3V3|3V3|3V3|
|15|TX1||TXD|
|16|RX1||RXD|
|17|P28|||
|18|CEN|CEN||
|19|P9|Status LED||
|20|P17 (SPI_MISO)|~~SH4 SDIO via (not populated) R18~~||
|21|P15 (SPI_CS)||CSN|

Tuya's info page on the SH4 says it uses the CMTOV30-EQR chip.  While I couldn't find that exact chip, CMOSTEK's CMT2300 has a suspiciously similar part number (CMT2300A-EQR) and the same pinout.

Marcos-Reckers, 2024-06-18:
I have the same module. I used GPT help to identifying and configuring the pins for both IR and RF functionalities.

### Pin Mapping Table

Here's a consolidated table mapping the IR and RF pins between the CBU (BK7231N) and SH4 modules based on the datasheets:

| CBU Pin (BK7231N)   | Function            | SH4 Pin        | Function                  |
|---------------------|---------------------|----------------|---------------------------|
| P7                  | IRSend              | N/A            | IR Transmission           |
| P8                  | IRRecv              | N/A            | IR Reception              |
| P14                 | SPI_SCK             | SPI_CLK        | SPI Clock Signal          |
| P15                 | SPI_CS              | SPI_CS         | SPI Chip Select           |
| P16                 | SPI_MOSI            | SDIO           | SPI Data (MOSI)           |
| P17                 | SPI_MISO            | SDIO           | SPI Data (MISO)           |
| P20                 | GPIO/Interrupt      | GPIO1/INT1     | Digital I/O               |
| P22                 | GPIO/Interrupt      | GPIO2/INT2     | Digital I/O               |
| P9                  | GPIO (Status LED)   | N/A            | Status LED                |

### Steps to Connect and Configure

1. **Identify and Connect Pins on PCB**:
   - Verify the corresponding pins on your device's PCB and make connections as per the table above.

2. **Configure OpenBK7231T Firmware**:
   - After connecting the pins correctly, access the OpenBK7231T web interface to configure the pins. Example YAML configuration:
     ```yaml
     pins:
       - pin: P7
         function: IRSend
       - pin: P8
         function: IRRecv
       - pin: P14
         function: SPI_SCK
       - pin: P15
         function: SPI_CS
       - pin: P16
         function: SPI_MOSI
       - pin: P17
         function: SPI_MISO
       - pin: P20
         function: GPIO  # or INT1, as needed
       - pin: P22
         function: GPIO  # or INT2, as needed
       - pin: P9
         function: StatusLED
     ```

3. **Test Functionality**:
   - Use the OpenBK7231T web interface or MQTT commands to test IR and RF functionalities. Monitor the logs to verify signals are processed correctly.

### Useful Resources and Documentation

For more details on IR and RF configuration support in OpenBK7231T, check out these sources:

- [OpenBK7231T on GitHub](https://github.com/openshwprojects/OpenBK7231T_App)
- [CBU Module Datasheet](https://developer.tuya.com/en/docs/iot/cbu-module-datasheet?id=Ka07pykl5dk4u)
- [SH4 Module Datasheet](https://developer.tuya.com/en/docs/iot/sh4-module-datasheet?id=Ka04qyuydvubw)