openshwprojects/OpenBK7231T_App #690: [Feature Request] DDP Support for BL602

ataa, 2023-02-23:
Using DDP, We can control LED lights and effects over the network using WLED, LEDFx and XLight.

More information:
https://github.com/arendst/Tasmota/pull/14017
https://cloudfree.shop/blog/2022/02/17/pair-tasmota-with-wled-for-lighting-effects/

I can see a DDP driver in the source code but not sure if it's active in BL602 branch.

openshwprojects, 2023-02-23:
Hey, nice idea. My BL602 dev board is not functional right now but I also have BL602 bulb and LED strip, so I will check this out tomorrow.

openshwprojects, 2023-02-24:
https://github.com/openshwprojects/OpenBK7231T_App/commit/c21611bda21799d173d290c7eb85d79ecd03fc57

ataa, 2023-02-24:
Thanks,
Does it work on your bulb?
I flashed it and WLED is complaining "Connection to light failed" and there's no log entry for DDP on BL602.
Should I change any setting on BL602?

openshwprojects, 2023-02-24:
Sorry, I didn't have time to test it yet, I am doing lot's of stuff for this project and it's hard while you're just a one person : D
But have you done "startDriver SSDP"? It must be done on startup, use autoexec.bat or short startup command.

ataa, 2023-02-24:
I understand and I am thankful for what you do. 
- Yes, DDP driver is active, I have this line in the log:
`Info:MAIN:Started DDP.`
- Not sure about the connection failed message, I set WLED DDP like this (based on [this guide](https://cloudfree.shop/blog/2022/02/17/pair-tasmota-with-wled-for-lighting-effects/):

[image]

- On Tasmota, additional command is needed to activate DDP (`Scheme 5`), but it does not work on BL602. 
- I'm not sure how to get the packet from WLED, I'll try to capture using wireshark.

openshwprojects, 2023-02-24:
We are currently using a very simple assumption that I also saw in Tasmota, and just get raw bytes:
[image]
If you need anything else, or something to be parsed, just tell me.

According to linked guide, i would need ESP to run WLED... is there a Windows tool for WLED as well, so I could test myself? I don't have ESP board at hand...

ataa, 2023-02-24:
- Does it listen to device IP or broadcast IP? Does the logging feature for DDP is active on BL602, there is no log entries for DDP other than driver start message.
- Yes, You can use [XLights](https://xlights.org) for windows.
- Here is a video showing Kauf bulb with DDP feature, WLED + DDP and XLights + DDP
- DDP Source file from Kauf Repo : https://github.com/KaufHA/kauf-rgbww-bulbs/blob/867eb5ea4440407d499e312d7946b2e3a3a9aa55/components/light/light_state.cpp

dg98, 2023-02-24:
Is this specific to BL602? Are the drivers not shared across branches?
I just tried on my newly flashed bulbs with BK7231N and BP5758 and after running "startDriver DDP" I  can control them using WLED just fine

ataa, 2023-02-24:
I don't have any bk lights to check. but tried on two different bl602 lights without success. So it might be and issue on my end.
- Have you changed any setting on the light other than starting the driver? Could you please share your WLED setting for DDP?
- Do you get DDP entries in the log other than driver start message?

dg98, 2023-02-24:
To get the lights to listen to WLED, I only had to start the driver. While controlled by WLED, I'm getting these entries in the log: `Info:CFG:####### Flash Save LED #######`
It's notable that when I'm running a WLED effect, the light bulb's web interface goes unresponsive until I turn it off.

These are my WLED settings:
[image]

Also for some reason the lights don't respond to DDP when I put `startDriver DDP` in my startup command, the log says `Info:DDP:no sock` until I start and stop the driver.

openshwprojects, 2023-02-24:
_Also for some reason the lights don't respond to DDP when I put startDriver DDP in my startup command, the log says Info:DDP:no sock until I start and stop the driver._
Thanks! I've added a retry. No sockets can be at startup. It will retry after 5 seconds automatically now.
https://github.com/openshwprojects/OpenBK7231T_App/commit/e139202103b8d186507b323344d236b5682f2a3e

_It's notable that when I'm running a WLED effect, the light bulb's web interface goes unresponsive until I turn it off._
I will need to try that first and see. How many packets it sends per second?

_I don't have any bk lights to check. but tried on two different bl602 lights without success. So it might be an issue on my end._
I need to check, maybe something else is not enabled in BL602 SDK.

I will try to install xligths and test

openshwprojects, 2023-02-24:
Ok ok, I know what's going on.
It works well and page is responsive as long as you set channels count (IN XLIGHT) to 3 or smth like that. Our LED is not a LED strip, it doesn't have 500 channels or so:
[image]
If you have a value like 500 here:
[image]
you will DDOS your light...

dg98, 2023-02-24:
As you can see in my screenshot, I set length to 1 so I have only one LED per DDP device. But I was also thinking that I was DDOSing my light because an effect like blend or breathe sends a lot of packets per second.

openshwprojects, 2023-02-24:
@dg98  how to set that effect? I am totally lost in that interface...

@ataa I will add stat counters to DDP driver

dg98, 2023-02-24:
I'm not sure, I never used xLight. I'm using the WLED web interface.

I just tested your last commit and starting the driver on startup is working now. I just have to turn WLED on and off for the effect to restart after cutting power to the bulbs.

openshwprojects, 2023-02-24:
" I just have to turn WLED on and off for the effect to restart after cutting power to the bulbs." Can you describe more precisely what's happening?

I am working on next commit, should be up in 15 minutes or so, with DDP stats and possibly first attempt to fix page not responsive issue.

openshwprojects, 2023-02-24:
btw that one effect is nice:
[image]

dg98, 2023-02-24:
Sorry, my previous response wasn't accurate
When I start an effect like breathe and then turn the bulbs on and off, they restore to the last color they were set to before I cut power. After about 10 seconds, the effect starts again without making any changes in WLED.
I think this is the expected behaviour

openshwprojects, 2023-02-24:
"After about 10 seconds, the effect starts again without making any changes in WLED." - Well, the WLED is still running all the time, right? 

The OpenBeken device itself has no DDP memory, it just parses DDP packet to get first 3 bytes for RGB and forgets the packet forever.

Anyway, I'm on it, let me know if there are any other things that needs to be done with DDP in OBK.

Hey, BL602 works as well!
[image]

dg98, 2023-02-24:
> "After about 10 seconds, the effect starts again without making any changes in WLED." - Well, the WLED is still running all the time, right?
> 
> The OpenBeken device itself has no DDP memory, it just parses DDP packet to get first 3 bytes for RGB and forgets the packet forever.
> 
> Anyway, I'm on it, let me know if there are any other things that needs to be done with DDP in OBK.
> 
> "After about 10 seconds, the effect starts again without making any changes in WLED." - Well, the WLED is still running all the time, right?

Yes, I was too impatient and thought I needed to restart the effect after reboot but everything works fine.

openshwprojects, 2023-02-24:
[image]
strangely enough, it seems to receive something when i tick off the test in xlight

ataa, 2023-02-24:
> It works well and page is responsive as long as you set channels count (IN XLIGHT) to 3 or smth like that

Mine was set to 3 as the video above suggested, but didn't work, then I updated the firmware to the release 487 and both xlights and wled started to work. For me web interface is responsive even while running fast color fades. Even made them sound reactive using LEDFx with acceptable frame rate.

Thanks a lot,
Let me know how to buy you a beer or two.

openshwprojects, 2023-02-24:
Thanks, the donation link is at the bottom of the readme:
https://www.paypal.com/paypalme/openshwprojects
but there are also other ways to support, for example you can just post a detailed teardown of not yet listed device on our forums, it always helps:
https://openbekeniot.github.io/webapp/devicesList.html
or post a generic tutorial on our forums, about something IoT related.
you can also consider subscribing our YT channel, but that's of course assuming you are interested in the videos:
https://www.youtube.com/@elektrodacom
I don't think that YT algorithm "likes" ghost subscribes, so you'd have watch a bit what I've recorded with my assistant.

ataa, 2023-02-24:
- Sure, I will do that.
- I have RGBWW with warm and cool white, while the WLED DDP driver supports extra white channels, I am not able to use cool white and warm white channels. Is this supported? White is emulated RGB white

openshwprojects, 2023-02-24:
thanks,
currently we just take three bytes as RGB values. 
[image]
Can you tell us either:
a) how the packet looks like in case of RGBCW DDP?
b) how to send RGBCW from xLights?

By looking at the documentation, I can tell that byte 2 seems to contain a data type, which can be RGB or HSL or RGBW, and this might be something we are looking for, but it's a bit confusing, because I'd rather expect a CW value so we can control LED temperature easily. I don't even have any "emulated white" LEDs, all my bulbs are RGBCW, so they have two channels for white: cool and warm.
http://www.3waylabs.com/ddp/#Packet%20Format

I will investigate more tomorrow.

ataa, 2023-02-24:
DDP network bus RGBW support: https://github.com/Aircoookie/WLED/commit/3270605b4f401acc16d42262c81ecffe89647df3

openshwprojects, 2023-02-25:
I've setup a simple DDP listen socket on my PC and:
[image]
[image]
No matter what I sent, the byte 2 is zero:
[image]
So it's different from linked example, hmm....

How should I go about supporting white or CW? 
Maybe just... check the number of channels in packet? It's adjustable in xLights...

dg98, 2023-03-11:
I just got some bulbs that use BK7231N with an SM2235 led driver and if I set my startup command to `startDriver DDP; SM2235_Map 1 0 2 3 4` the DDP driver doesn't start after rebooting. I didn't see anything in the logs about that and DDP works fine if I start it after boot.

On another note, I noticed that while the webinterface now works fine even when receiving DDP packets, OTA updates don't work. It reboots the device but the old version remains. Once I stopped WLED, OTA worked again.

openshwprojects, 2023-03-18:
IT seems that I have to stop drivers when OTA starts. Thanks. I will keep it in mind.

For a command with ;, you need to use backlog.

dg98, 2023-06-30:
> How should I go about supporting white or CW? Maybe just... check the number of channels in packet? It's adjustable in xLights...

Have you made any progress on supporting white channels?