openshwprojects/OpenBK7231T_App #693: New Tuya module FL-M61 appeared

efal, 2023-02-25:
In an ordinary wifi switch module from AliExpress (looking like sonoff basic) a new wifi module appeard.
The modul looks similar to a tuya chip like CB2S but the SoC is  no Beken bu called TR6260S1 from 'is Transa semi'.
Works like other devices with smart life app.
Any idea?
Desoldered one and replaced with ESP01 but reprogamming would be much easier....
Thanx for OpenBeken.

efal, 2025-07-21:
Thanks, divadiow. Will try

divadiow, 2025-07-21:
Please take a backup of existing firmware beforehand - we could do with more factory TR6260 dumps.

Also note the supported features in https://github.com/openshwprojects/OpenBK7231T_App/blob/afdf4b5cb54c2e82ca966d7198a0c110debcb43f/docs/platforms.md

Recently updated by @insmod

divadiow, 2025-08-12:
did you have any luck @efal ?

efal, 2025-08-12:
Not tried yet. Thanx for asking. Will post any Progress.--Gesendet mit der mail.com Mail App