openshwprojects/OpenBK7231T_App #699: Power Toggle not working Merkury MI-BW210-999W BK7231T WB2L SM2135 LED Driver

SawKyrom, 2023-02-26:
Issue: 
Toggle light only changes bulb from color mode to white mode, not on or off.  
Color selector only works on HA interface, not web interface, (web interface will update color status appropriately)

Managed to successfully UART flash OpenBK7231T on device, yet MQTT 'cmnd/MerkB58/power : msg.payload : 0' or 'cmnd/MerkB58/led_enableAll' only changes bulb from color mode to white mode, not on/off.  Must drop brightness scale to 0 to turn off light.  Results are the same with either HA or web page control.

Only the HA color selector works.  When attempting to use the LED RGB Color selector on web, no response generated.  MQTT cmnd led_basecolor_rgb works with HA.  

Information: 
Build 1.15.487
Chipset BK7231T

Pin Assignment:
P7 (PWM1) SM2135DAT
P8 (PWM2) SM2135CLK

Flag 4, 9, 10, 34

Startup Command Line: SM2135_map 2 1 0 3 4

SawKyrom, 2023-02-26:
Additional Information:
```
  - name: "MerkB58"
    rgb_command_template: "{{ '#%02x%02x%02x0000' | format(red, green, blue)}}"
    rgb_value_template: "{{ value[0:2]|int(base=16) }},{{ value[2:4]|int(base=16) }},{{ value[4:6]|int(base=16) }}"
    rgb_state_topic: "MerkB58/led_basecolor_rgb/get"
    rgb_command_topic: "cmnd/MerkB58/led_basecolor_rgb"
    command_topic: "cmnd/MerkB58/led_enableAll"
    state_topic: "MerkB58/led_enableAll/get"
    availability_topic: "MerkB58/connected"
    payload_on: 1
    payload_off: 0
    brightness_command_topic: "cmnd/MerkB58/led_dimmer"
    brightness_state_topic: "MerkB58/led_dimmer/get"
    brightness_scale: 100
    #brightness_value_template: "{{ value }}"
    color_temp_command_topic: "cmnd/MerkB58/led_temperature"
    color_temp_state_topic: "MerkB58/led_temperature/get"
    #color_temp_value_template: "{{ value }}"
```

MQTT Debug Information:

```
cmnd/MerkB58/power : msg.payload : string[3]
"OFF"

stat/MerkB58/RESULT : msg.payload : Object
{ Dimmer: 100, Fade: "OFF", Speed: 1, LedTable: "ON", Color: "0,0,0,0,0" … }
```

openshwprojects, 2023-02-26:
Hello, have you tried with the SM2135 flag?
[image]
Or does the problem happen is older versions as well?

SawKyrom, 2023-02-26:
@openshwprojects Thank you for the reply and suggestions.

> Do you mean that color selector is not working neither on PC or mobile, or that it does not work only on mobile app?

The color selector does not work when accessing web control (see image red arrow) using IP address.  I can change color using MQTT commands via HA or NodeRed.  When color is changed using MQTT/HA, the web control interface correctly updates color.  (see image black arrows)
[image]

The color issues is not of high importance since I don't plan on using the web address interface for control and control is possible via MQTT and HA.  I only bring it up in event that it might be related to on/off toggle issue.  Any feedback to resolve the toggle button (turn on/off light) would be much appreciated.  Am I missing a PWM function on WB2L chipset that is not coded for LED on/off?  Or is there a setting to make the MQTT power toggle `cmnd/MerkB58/led_enableAll : msg.payload : number
0` to change "Dimmer:" value to "0" versus the "100" posted in the MQTT status response shown below:

`stat/MerkB58/RESULT : msg.payload : Object
{ Dimmer: 100, Fade: "OFF", Speed: 1, LedTable: "ON", Color: "0,0,0,0,0" … }`

openshwprojects, 2023-02-26:
Which browser are you using to access www page? Can you try on latest chrome?
I am using it with chrome daily and it always works.

Regarding power... so when you click "Toggle LED power", it stays on?

openshwprojects, 2023-02-26:
I have tested on my family phones. It seems color picker was not working on Android, but it worked ok on iPhone and chrome on wdinows.
I have added a fix, can you update your obk to latest version and check now if color pickers on www page works?

ataa, 2023-02-27:
Color picker does not work on Firefox / Firefox Android even after updating to 1.15.508, If I select a color and check the console I get `GEThttp://192.168.20.116/index?rgb=#ffffff&rgbIndex=131` and it stays white.

openshwprojects, 2023-02-27:
@ataa  huh, that's strange, but thanks for telling me. Well, at least color picker works on my Xiaomi S11...
Well, you are right, it does not work on firefox:
[image]

Let me check if I can fix it... btw, do you have any ideas what else I could do there to catch any change events?
[image]

We are very lucky that I can run OBK on Windows and check things fast

openshwprojects, 2023-02-27:
update, it seems that one of the issues is that for Firefox, the page refresh seems to disconnect the color picker from the control

openshwprojects, 2023-02-27:
This condition for active element is not working on firefox:
[image]

openshwprojects, 2023-02-27:
UPDATE 3 : it seems that I resolved the disjointed update problem. Can anyone check latest commit, once it compiles?

I tested it on:
- PC Firefox and Chrome
- iPhone and android

ataa, 2023-02-27:
Thanks, I tried 509, It works on Firefox now, if you open color picker and select a color once and close it, if you select a color and then select another, second one does not apply, not sure if it's bug or this is default behavior for firefox color picker.

openshwprojects, 2023-02-27:
You are correct. This is what I mentioned earlier before modifying my message. When you apply the color, the page is reloaded and on chrome and all normal browsers, the color picker is also closed/reloaded, but for firefox the old color picker stays detached somehow on the screen, while the new input color field appears on the HTML.

You can see that on chrome, when you pick a color, the picker will disappear, right?

I am not sure yet what to do about it, maybe there is a better way to handle color, or I will close that picker in code, if possible.

In general, it's a mess, every browser is different.

ataa, 2023-02-27:
Thank you for a quick fix. It works, so it's good enough for me.

openshwprojects, 2023-02-27:
Are there any other browswers that are currently probletmaic?