openshwprojects/OpenBK7231T_App #704: WiFi fast connect (flag 37) testers needed!

openshwprojects, 2023-02-28:
In the past we had some little issues with connecting to WiFi too early, most likely because of not thread-safe LWIP....
We have fixed that by adding an overly large connect delay, but now that LWIP is updated, I am trying to reduce the time it takes for device to connect to network.
I have added fast connect flag, which is not enabled by default:
[image]
**If anyone want to help - please enable this flag and report here if it works and are there any problems.**
If you get problems with that flag, just do 5 power on/off and in ap mode (safe mode) disable that flag again.

ataa, 2023-02-28:
Does this flag active on BL602?

openshwprojects, 2023-02-28:
Yes, on every platform. There was an artificial delay introduced by us to fight some stability errors in the past... this flag allows you to skip that delay. Also for MQTT.

ataa, 2023-03-01:
I just tested this new flag and it works, I get connection in 4 to 5 seconds.

openshwprojects, 2023-03-01:
and without it, how long was that?

ataa, 2023-03-01:
14-15s on BL602 Bulb:
```
1 drivers active (BP5758D), total 18
Channel 0 = 0.00, Channel 1 = 0.00
Cfg size: 2016, change counter: 30, ota counter: 0, boot incompletes 0 (might change to 0 if you wait to 30 sec)!
Ping watchdog (192.168.20.1) - will start in 29!
Wifi RSSI: Excellent (-1dBm)
MQTT State: not configured
```

ekobres, 2023-03-09:
Seems to work for me. BK7231N Device reconnects after restart in about 5 seconds with no issues. Same device with flag 37 turned off takes about 15 seconds to start responding.

GravityRZ, 2023-04-12:
i tested this on an BK7231T
with the flag to on it seems that the mqtt status shows disconnected after a rebootr and it take s a lot of time to reconnect.
with the flag disabled the mqtt status goes to disconnecte and after a 2 seconds to connected again.

without this flag it rougly takes 7 seconds to connect to wifi so i find this flag useless

openshwprojects, 2023-04-17:
RTE is fixed now. @GravityRZ  please recheck.

We also added static IP support, which increases connection speed.

sillyfrog, 2023-04-22:
I have tested on "Mirabella Genio 9W 750Lumen, 2700K~6500K RGB + CCT Downlight - Part I004544" devices, and found connection time from restart to re-connect goes from ~15s to ~8s. So far things are looking stable, but not yet extensive testing.

ostat, 2023-04-25:
Running on BK7231T. Wifi time went from 11s to about 6s. Otherwise no change, seems to be working well.

dskw, 2023-05-18:
Running 1.17.115 on a BK7231N door sensor withOUT tuyaMCU.
Connection takes ~18 seconds. The flag makes no difference for me.
Sometimes it does connect in ~6 seconds though (maybe 1/10 times)

belveder79, 2024-01-15:
Running on BK7231N water sensor withOUT tuyaMCU
Connection takes between 8 and 15 seconds (similar to what @dskw reported)
I have to say that this is a HUGE improvement to running without the Flag37, but arguably the faster, the better...