openshwprojects/OpenBK7231T_App #709: The device stops responding

KaMyKaSii, 2023-03-05:
**Describe the bug**
After entering the "Config module" page, the device no longer responds to the page or ping. The router says that the device is still connected to Wi-Fi, basic operation like turning on/off via the physical button still works.

**Firmware:**
- Version: 1.15.529
- Device: LSPA3 (16A smart plug with power meter, EU/Brazil plug)
- Chip/model: BK7231T WB2S
- Device config: LSPA7 template

**To Reproduce**
Just open the "Config module" page. The same happens sometimes when submitting deactivate on some "Gerenal/flags" options.

**Screenrecord**
https://user-images.githubusercontent.com/17092335/222978518-358da5ec-76c6-47e5-9891-7286da365d5c.mp4

openshwprojects, 2023-03-05:
Thanks, I will check this soon. Are you able to determine with which version it has started happening?

KaMyKaSii, 2023-03-05:
> Thanks, I will check this soon. Are you able to determine with which version it has started happening?

I flashed this custom firmware on my device for the first time yesterday on version 1.15.524 and every other OTA I've installed has the same problem. Can you reproduce the problem?

openshwprojects, 2023-03-05:
Not yet, my yesterdays build seems ok:
https://imgur.com/a/aGVp57h
I will also test official build now

KaMyKaSii, 2023-03-06:
With a small shell script I wrote I noticed that the device starts to respond and with its web interface functional again after 19 minutes of inactivity, three different tests had this same result. Is this time some default in the code?
```
#!/bin/bash
device_ip=192.168.31.67

echo "Waiting for ${device_ip} response..."

start=${SECONDS}
while ! ping -c 1 ${device_ip} 1>/dev/null; do
	sleep 1
done

elapsed_time=$((SECONDS - start))
echo "${elapsed_time} seconds until a successful ping."
```
```
kamykasii:[~]: ./timetosuccessfulping.sh
Waiting for 192.168.31.67 response...
1144 seconds until a successful ping.
```

openshwprojects, 2023-03-06:
I don't know anything about that. There are no magic delay like that.

Do you have MQTT connected? I presume no, just like on your video?

How many devices do you have?

Is the free memory size in Logs going down gradually?

KaMyKaSii, 2023-03-06:
This is my only device, I bought it mainly for the power meter function at a low price, I'm not into automation. So I basically just installed OpenBK, looked for some working template and did the basic calibration. Looking at the log the value "free 88040" seems to be pretty constant.

<details>

```
fo:MAIN:Time 9206, idle 448604/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9207, idle 222520/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9208, idle 207142/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9209, idle 196181/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9210, idle 224186/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:GEN:dhcp=0 ip=192.168.31.67 gate=192.168.31.1 mask=255.255.255.0 mac=7c:f6:66:20:41:ff 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-52,ssid=casa,bssid=28:d1:27:80:2b:0a ,channel=13,cipher_type:CCMP
Info:MAIN:Time 9211, idle 202839/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9212, idle 230348/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9213, idle 196101/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9214, idle 208080/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9215, idle 220900/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9216, idle 225145/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9217, idle 235153/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9218, idle 216301/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9219, idle 480472/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9220, idle 216084/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:GEN:dhcp=0 ip=192.168.31.67 gate=192.168.31.1 mask=255.255.255.0 mac=7c:f6:66:20:41:ff 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-54,ssid=casa,bssid=28:d1:27:80:2b:0a ,channel=13,cipher_type:CCMP
Info:MAIN:Time 9221, idle 196471/s, free 88040, MQTT 0(573), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:mqtt_host empty, not starting mqtt
Info:MAIN:Time 9222, idle 252225/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9223, idle 222808/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9224, idle 218598/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9225, idle 205635/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9226, idle 222402/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9227, idle 213631/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9228, idle 235212/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9229, idle 224357/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9230, idle 203516/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:GEN:dhcp=0 ip=192.168.31.67 gate=192.168.31.1 mask=255.255.255.0 mac=7c:f6:66:20:41:ff 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-54,ssid=casa,bssid=28:d1:27:80:2b:0a ,channel=13,cipher_type:CCMP
Info:MAIN:Time 9231, idle 212201/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9232, idle 225666/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9233, idle 230712/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9234, idle 240966/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:CMD:[WebApp Cmd 'loglevel 3' Result] OK
Info:MAIN:Time 9235, idle 210751/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9236, idle 224559/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9237, idle 202143/s, free 88040, MQTT 0(574), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MQTT:mqtt_host empty, not starting mqtt
Info:MAIN:Time 9238, idle 228690/s, free 88040, MQTT 0(575), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9239, idle 214658/s, free 88040, MQTT 0(575), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9240, idle 246225/s, free 88040, MQTT 0(575), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:GEN:dhcp=0 ip=192.168.31.67 gate=192.168.31.1 mask=255.255.255.0 mac=7c:f6:66:20:41:ff 
Info:GEN:sta: 1, softap: 0, b/g/n
Info:GEN:sta:rssi=-51,ssid=casa,bssid=28:d1:27:80:2b:0a ,channel=13,cipher_type:CCMP
Info:MAIN:Time 9241, idle 212673/s, free 88040, MQTT 0(575), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9242, idle 224387/s, free 88040, MQTT 0(575), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9243, idle 209911/s, free 88040, MQTT 0(575), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9244, idle 268012/s, free 88040, MQTT 0(575), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9245, idle 198457/s, free 88040, MQTT 0(575), bWifi 1, secondsWithNoPing -1, socks 2/38 
[... cut ...]
Info:MAIN:Time 9425, idle 213500/s, free 88040, MQTT 0(586), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9426, idle 218079/s, free 87824, MQTT 0(586), bWifi 1, secondsWithNoPing -1, socks 3/38 
Info:MAIN:Time 9427, idle 244296/s, free 88040, MQTT 0(586), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9428, idle 235204/s, free 88040, MQTT 0(586), bWifi 1, secondsWithNoPing -1, socks 2/38 
Info:MAIN:Time 9429, idle 213359/s, free 88040, MQTT 0(586), bWifi 1, secondsWithNoPing -1, socks 2/38 
```
</details>

duculete, 2023-03-21:
fixed: power on/off multiple times (10) so that the device will start in AP mode; once in AP go and remove all PIN configuration (I used the one suggested in the app and that seemed to be the problem):
  "pins": {
    "3": "PWM;2",
    "4": "PWM;0",
    "17": "PWM;3",
    "20": "PWM;4",
    "21": "PWM;1"
  },

now I have to figure out what are the pins config for PWM RGBW.

lulol, 2023-09-16:
The internal watchdog also seems to stop receiving ping replies, even if it was pointing to an unused ip address (192.168.0.1) that wouldn't ever reply.
```
OpenBL602_DEADBEEF
0 drivers active, total 12
Cfg size: 3584, change counter: 2, ota counter: 0, boot incompletes 1 (might change to 0 if you wait to 30 sec)!
Wifi RSSI: Excellent (-1dBm)
MQTT State: not configured

Build on Sep 15 2023 08:05:30 version 1.17.245
Online for 5 minutes and 30 seconds
Device MAC: DE:AD:DE:AD:BE:EF              <<< censored
Short name: oblDEADBEEF, Chipset BL602
```
This is what I tested:
1. Factory reset thru OpenBK7231T_App: _"Return device to factory settings (this will erase everything, even your Wifi config)"_
2. Start pinging 192.168.1.xx
3. Configure AP via WiFi.
4. Ping replies start normally.
5. Open http://192.168.1.xx/app? in browser.
6. After about 70s ping replies stop.

<details>
<pre>
Info:MAIN:Main_Init_Before_Delay
Warn:CFG:CFG_InitAndLoad: Correct config has been loaded with 2 changes count.
Info:GEN:PIN_SetupPins pins have been set up.
Info:MAIN:Main_Init_Before_Delay done
Info:MAIN:Main_Init_Delay
Info:MAIN:Main_Init_Delay done
Info:MAIN:Main_Init_After_Delay
Info:MAIN:Using SSID [xxxxxxxxxx]
Info:MAIN:Using Pass [xxxxxxxxxx]
Info:MQTT:MQTT_RegisterCallback called for bT oblDEADBEEF/ subT oblDEADBEEF/+/set
Info:MQTT:MQTT_RegisterCallback called for bT bl602s/ subT bl602s/+/set
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/oblDEADBEEF/ subT cmnd/oblDEADBEEF/+
Info:MQTT:MQTT_RegisterCallback called for bT cmnd/bl602s/ subT cmnd/bl602s/+
Info:MQTT:MQTT_RegisterCallback called for bT oblDEADBEEF/ subT oblDEADBEEF/+/get
Error:CMD:cmd startScript NOT found (args autoexec.bat)
Info:MAIN:Main_Init_After_Delay done
Info:MAIN:Time 1, idle 0/s, free 109304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 2, idle 0/s, free 109304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 3, idle 0/s, free 109304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 4, idle 0/s, free 109304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 5, idle 0/s, free 109304, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
Info:MAIN:Time 6, idle 0/s, free 107264, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Boot complete time reached (5 seconds)
Info:MAIN:Time 7, idle 0/s, free 107264, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 8, idle 0/s, free 107264, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 9, idle 0/s, free 109264, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTED - 4
Info:MAIN:Time 10, idle 0/s, free 103640, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 3/21 
Info:MAIN:Time 11, idle 0/s, free 109264, MQTT 0(0), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:mqtt_host empty, not starting mqtt
Info:MAIN:Time 12, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 13, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 14, idle 0/s, free 106728, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/21 
Info:MAIN:Time 15, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 16, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 17, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 18, idle 0/s, free 108976, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/21 
Info:MAIN:Time 19, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 20, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 21, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 22, idle 0/s, free 106728, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/21 
Info:MAIN:Time 23, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 24, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 25, idle 0/s, free 109264, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 26, idle 0/s, free 108976, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 3/21 
Info:MAIN:Time 27, idle 0/s, free 90232, MQTT 0(1), bWifi 1, secondsWithNoPing -1, socks 5/21 
[... 57 lines cut ...]
Info:MAIN:Time 79, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 10, socks 2/21 
Info:MAIN:Time 80, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 11, socks 2/21 
Info:MAIN:Time 81, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 12, socks 2/21 
Info:MAIN:Time 82, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 13, socks 2/21 
Info:MAIN:Time 83, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 14, socks 2/21 
Info:MAIN:Time 84, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 15, socks 2/21 
Info:MAIN:Time 85, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 16, socks 2/21 
Info:MAIN:Time 86, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 17, socks 2/21 
Info:MAIN:Time 87, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 18, socks 2/21 
[... cut ...]
Info:MAIN:Time 91, idle 0/s, free 109264, MQTT 0(5), bWifi 1, secondsWithNoPing 22, socks 2/21 
Info:MQTT:mqtt_host empty, not starting mqtt
Info:MAIN:Time 92, idle 0/s, free 109264, MQTT 0(6), bWifi 1, secondsWithNoPing 23, socks 2/21 
Info:MAIN:Time 93, idle 0/s, free 109264, MQTT 0(6), bWifi 1, secondsWithNoPing 24, socks 2/21
...
</pre>
</details>

----

Have other device running v1.17.98 with a BK7231N chipset that doesn't show this issue.

Tried downgrading to `OpenBL602_1.17.98_OTA.bin.xz.ota` but didn't help.

coltc50, 2023-12-21:
I think have the same problem. I'm using 3 blubs with BK7231T. Wifi is good. When I go to config Flags the site loads half and the bulb isn't reach anymore for the next minutes. I'm on 1.17.343

lulol, 2023-12-28:
>_... isn't reach anymore for the next minutes_

Do they start responding again after those minutes?

openshwprojects, 2023-12-28:
Can you configure them via web app?

coltc50, 2023-12-28:
> > _... isn't reach anymore for the next minutes_
> 
> Do they start responding again after those minutes?

Yes they respond after some time... I didn't look on time how long it took... When they doesn't respond, the devices are also marked as offline in mqtt.

lulol, 2023-12-28:
@openshwprojects Have updated and configured via web app this same BL602 board via OTA from version 1.17.245 to the latest version 1.17.376. All went smoothly and without any issue.

But I still see a similar behavior as previously: It stops responding to pings about 70 seconds after boot, even without accessing the web interface.

@coltc50 I've been pinging mine for over an 1 hour and never responded again.
Although in my case mqtt and the web interface keep working just fine.
[MQTT Explorer](https://github.com/thomasnordquist/MQTT-Explorer)  and nodered show the associated traffic as expected.
Even Alexa still is able to handle it without any problem.
The only symptom I see is the lack of ICMP ping replies, which is just a minor issue, only with this board.
After pressing the restart button in the web interface it replies again until it stops 70 seconda later.

Have you tried to do a factory reset and full reconfigure as I did show above?

<details>
<pre>
Current Device:
UpTime: 581s
Build: Build on Dec 25 2023 20:45:48 version 1.17.376
IP Address: 192.168.1.XXX
MAC Address: XXXXXXXXXXXX
MQTT Server: 192.168.1.XXX:1883
MQTT Topic: oblXXXXXXXX
Device Short Name: XXX
WEBAPP Url root: https://openbekeniot.github.io/webapp/
Chipset: BL602
Flags: 1024
Version: 1.17.376 
</pre><pre>
Export Current Template
Please fill all fields before submitting.
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BL602",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "2": "Rel;0",
    "4": "Btn;0",
    "20": "LED_n;0",
    "22": "Btn;0"
  },
  "command": "backlog startDriver WEMO; startDriver SSDP",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}
</pre>
</details>